istoreos/istoreos #1349: 大佬帮我看看，重启三次的系统错误，直接打不开系统了

AlanWang2333, 2024-02-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述：我在那调那个nps，卸载重装了一遍，重启了三次系统，他就报错了，进不去系统了

### (1) 具体问题

A：进不去系统

### (2) 路由器型号和固件版本

A：x86 J3160，用的最新版本固件

### (3) 详细日志和/或截图

A：
[image]

问了一下gpt
#我的openwrt，打不开系统，显示 987675] procd:Cannot set group dialout for /dev/tty38（getgrnam:0)
#这个错误表明系统无法将 `/dev/tty38` 设置为 `dialout` 组。通常情况下，`dialout` 组是用于串口设备的组。尝试以下解决方法：

1. **检查设备节点权限：** 确保 `/dev/tty38` 的权限正确设置，并且拥有 `dialout` 组的访问权限。你可以运行以下命令来确认权限：
   ```
   ls -l /dev/tty38
   ```

2. **手动设置权限：** 如果权限不正确，你可以手动设置：
   ```
   sudo chown root:dialout /dev/tty38
   sudo chmod g+rw /dev/tty38
   ```

3. **检查组是否存在：** 确保 `dialout` 组存在于系统中：
   ```
   cat /etc/group | grep dialout
   ```

4. **重启系统：** 在应用更改后，尝试重新启动系统以查看是否解决了问题。

如果以上方法都没有解决问题，可能需要更深入地检查系统日志以获取更多信息，或者考虑是否有其他因素导致系统无法正常启动。
#uninitialized urandom read (4bytes read) Failed to connect to ubus
#这些错误信息通常涉及系统启动过程中的随机数生成和进程间通信问题。

1. **uninitialized urandom read (4bytes read):** 这个错误表明系统在尝试从 `/dev/urandom` 设备中读取随机数时遇到了问题。这可能是由于系统启动过程中的某些环境或配置问题引起的。你可以尝试以下方法解决：
   - 确保系统磁盘没有损坏或者磁盘空间不足。
   - 尝试在系统启动时检查文件系统并修复错误。

2. **Failed to connect to ubus:** 这个错误表明系统在连接到 `ubus` 时遇到了问题。`ubus` 是 OpenWrt 上的一个进程间通信机制。可能的解决方法包括：
   - 检查 `ubus` 进程是否正在运行，并且没有被防火墙或其他安全设置阻止。
   - 检查系统日志以获取更多信息，确定具体的连接问题。

综上所述，建议先检查文件系统和磁盘状态，然后检查 `ubus` 进程和相关的网络或安全设置。如果问题仍然存在，可能需要更深入地检查系统日志以找出根本原因。

jjm2473, 2024-02-27:
可能中毒了？你没改管理员密码吗？

AlanWang2333, 2024-02-27:
> 可能中毒了？你没改管理员密码吗？

改了呀，我还以为是装的东西太多了。哎呦，(╥ω╥`)我都崩溃了，砖了第二次了，这玩意可真不好搞

jjm2473, 2024-03-04:
检查是不是硬盘有故障，换个硬盘，或者u盘启动

AlanWang2333, 2024-03-05:
重装了系统，选了个旧版的，你知道哪个版本最稳定吗