istoreos/istoreos #1350: 最新版 iStoreOS 22.03.6 2024012613 无法重设IP

carsonrz, 2024-02-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：如题所示，通过命令行界面的quickstart->chang lan IP， 或者通过web界面的【内网配置】修改保存后，重启软路由，设置也没生效。但如果通过web界面【网络向导】->【配置为旁路由】界面修改保存就可以立刻生效。

### (2) 路由器型号和固件版本

A：最新稳定版本 iStoreOS 22.03.6 2024012613

### (3) 详细日志和/或截图

A：

cnkmmk, 2024-02-23:
碰到同样的问题，无法通过quickstart进行LAN口的IP配置

cnkmmk, 2024-02-26:
查看 /etc/config/network 文件，里面的IP获取方式还是DHCP。手动改成 static 。

hugonj, 2024-03-04:
我也是这样的问题，我是用在X86旁路由上，在istoreos 内升级后出现无法给设备自动分配IP, 一开始不知道什么问题，就重新装了系统，发现无法通过quickstart进行LAN口的IP配置，直到用回22.03.5版才可用quickstart进行LAN口的IP配置，之后DHCP也能自动给设备分配IP.

marine19811006, 2024-03-07:
一样的问题，vi etc/config/network 改的静态ip就好了

carsonrz, 2024-03-07:
这些算是基本功能，解决也不难，但很多小白会懵逼，还是希望官方能尽快修复

CWWK-TechnicalSupport, 2024-03-27:
2024012613这个不对吧，最新的是2024031514，最新的修复了这个问题

closed 2024-04-03 by carsonrz