istoreos/istoreos #1351: R5S只能识别两个接口

lxdai7610, 2024-02-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? 
 没有类似的issue

2. 详细叙述

(1) 具体问题

R5S从r5s-20231231-openwrt-rockchip-armv8-friendlyarm刷到 iStoreOS 22.03.6 系统识别城R5C 而且只识别到两个接口，换了好几个版本都是这样

 (2) 路由器型号和固件版本

机器型号：R5S 4G版
固件版本：iStoreOS 22.03.6 2024011910

 (3) 详细日志和/或截图

A：
型号识别错误，应该是R5S
[image]

网口只识别到两个
[image]

lxdai7610, 2024-02-19:
update：
使用 2023052616之前的版本，比如2023052311就会识别正确，网口也可以全部识别
[image]
[image]

jjm2473, 2024-02-21:
直接刷最新固件，不要从老版本升级

lxdai7610, 2024-02-21:
试过了，直接线刷高版本还是一样，反应到设备厂商那，说说春节后的R5S升级了一下，多了一个红外接收器，不知道是否是这个原因导致

jjm2473, 2024-02-22:
你的R5S是新买的？

jjm2473, 2024-02-22:
你能拍个R5S的正面照吗，看看是不是 [R5S-LTS](https://www.friendlyelec.com/index.php?route=product/product&product_id=287) 。还有，问下厂商，“R5S-LTS” 跟 “R5S C1” 是不是同一款机型，我看他们最近添加的是“R5S C1”。

jjm2473, 2024-02-22:
试试这个固件，加上了 R5S LTS 和 R5S C1 的支持 https://we.tl/t-OZpxEc8gZC

lxdai7610, 2024-02-22:
> 尝试一下这个固件，加上 R5S LTS 和 R5S C1 的支持https://we.tl/t-OZpxEc8gZC

这个固件没问题，刷上后识别型号为：R5S LTS，网口也可以识别三个
[image]
[image]

jjm2473, 2024-02-22:
OK，那之后的固件都会支持

lxdai7610, 2024-02-22:
厂商反馈：春节后的R5S 升级了红外接收接口，LTS 就是加了红外的，所以加了新的型号实际就是R5S LTS，感谢大佬@jjm2473，等待官方更新了...

closed 2024-02-23 by jjm2473