istoreos/istoreos #1361: docker 打开防火墙

pili-book, 2024-02-24:
1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时包括已关闭的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)

 没有类似的issue
2. 详细叙述
(1) 具体问题
我发现主路由docker建立容器后会自动打开防火墙，这样我用lucky大吉或者scoat想要端口转发的时候，这个端口就会被占用无法打开，还需要额外打开另外的端口，这样会不会增加外网端口暴露的风险呢
(2) 路由器型号和固件版本
A：FriendlyElec NanoPi R6S
iStoreOS 22.03.5 2023121510 / LuCI istoreos-22.03 branch git-23.342.39264-37c783a

jjm2473, 2024-02-25:
docker的防火墙已经禁用，不会自动开放，你要开放的话，用系统防火墙配置就行了。

closed 2024-02-25 by jjm2473

pili-book, 2024-02-25:
但是我某个容器映射了端口5700，使用lucky大吉反向代理，那个端口就已经占用了。。。。。

pili-book, 2024-02-25:
> docker的防火墙已经禁用，不会自动开放，你要开放的话，用系统防火墙配置就行了。

[image]

jjm2473, 2024-02-25:
那不是废话吗，端口映射不就已经占用掉端口了吗？你是不是不了解端口的概念啊

jjm2473, 2024-02-25:
端口已经被docker自动映射了，也就是已经有端口转发，你要外网访问就在防火墙允许这个端口输入就行了，不需要再配置端口转发。这种问题不要再来istoreos反馈，这是普通的网络知识，跟固件没任何关系。

pili-book, 2024-02-25:
那这种情况luck大吉的反向代理就只能是映射别的端口吗？各种尝试好像没办法，都是显示端口占用了