istoreos/istoreos #1362: 请问cnvi AC9560无线网卡如何驱动呢

SOYOJOE, 2024-02-24:
Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
istore不能驱动cnvi AC9560无线网卡

### (2) 路由器型号和固件版本

A：
x86,
iStoreOS 22.03.6 2024022310

### (3) 详细日志和/或截图
[image]
[image]
[image]
[image]

jjm2473, 2024-02-27:
搜索内核日志，关键词“iwlwifi”

SOYOJOE, 2024-02-29:
我获取到了wifi日志,请帮忙分析呢，谢谢你了~
```
root@iStoreOS:~# dmesg|grep iwlwifi
[    7.205416] iwlwifi 0000:07:00.0: Direct firmware load for iwlwifi-so-a0-jf-b0-66.ucode failed with error -2
[    7.205919] iwlwifi 0000:07:00.0: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-66.ucode
[    8.690965] iwlwifi 0000:07:00.0: Direct firmware load for iwlwifi-so-a0-jf-b0-65.ucode failed with error -2
[    8.691418] iwlwifi 0000:07:00.0: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-65.ucode
[    8.693242] iwlwifi 0000:07:00.0: Direct firmware load for iwlwifi-so-a0-jf-b0-64.ucode failed with error -2
[    8.693692] iwlwifi 0000:07:00.0: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-64.ucode
[    8.695306] iwlwifi 0000:07:00.0: Direct firmware load for iwlwifi-so-a0-jf-b0-63.ucode failed with error -2
[    8.695751] iwlwifi 0000:07:00.0: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-63.ucode
[    8.697543] iwlwifi 0000:07:00.0: Direct firmware load for iwlwifi-so-a0-jf-b0-62.ucode failed with error -2
[    8.698014] iwlwifi 0000:07:00.0: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-62.ucode
[    8.699645] iwlwifi 0000:07:00.0: Direct firmware load for iwlwifi-so-a0-jf-b0-61.ucode failed with error -2
[    8.700106] iwlwifi 0000:07:00.0: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-61.ucode
[    8.701685] iwlwifi 0000:07:00.0: Direct firmware load for iwlwifi-so-a0-jf-b0-60.ucode failed with error -2
[    8.702145] iwlwifi 0000:07:00.0: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-60.ucode
[    8.703939] iwlwifi 0000:07:00.0: Direct firmware load for iwlwifi-so-a0-jf-b0-59.ucode failed with error -2
[    8.704402] iwlwifi 0000:07:00.0: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-59.ucode
[    8.705976] iwlwifi 0000:07:00.0: Direct firmware load for iwlwifi-so-a0-jf-b0-58.ucode failed with error -2
[    8.706439] iwlwifi 0000:07:00.0: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-58.ucode
[    8.708002] iwlwifi 0000:07:00.0: Direct firmware load for iwlwifi-so-a0-jf-b0-57.ucode failed with error -2
[    8.708449] iwlwifi 0000:07:00.0: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-57.ucode
[    8.710007] iwlwifi 0000:07:00.0: Direct firmware load for iwlwifi-so-a0-jf-b0-56.ucode failed with error -2
[    8.710782] iwlwifi 0000:07:00.0: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-56.ucode
[    8.712322] iwlwifi 0000:07:00.0: Direct firmware load for iwlwifi-so-a0-jf-b0-55.ucode failed with error -2
[    8.712780] iwlwifi 0000:07:00.0: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-55.ucode
[    8.714694] iwlwifi 0000:07:00.0: Direct firmware load for iwlwifi-so-a0-jf-b0-54.ucode failed with error -2
[    8.715157] iwlwifi 0000:07:00.0: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-54.ucode
[    8.716712] iwlwifi 0000:07:00.0: Direct firmware load for iwlwifi-so-a0-jf-b0-53.ucode failed with error -2
[    8.717174] iwlwifi 0000:07:00.0: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-53.ucode
[    8.718773] iwlwifi 0000:07:00.0: Direct firmware load for iwlwifi-so-a0-jf-b0-52.ucode failed with error -2
[    8.719240] iwlwifi 0000:07:00.0: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-52.ucode
[    8.720796] iwlwifi 0000:07:00.0: Direct firmware load for iwlwifi-so-a0-jf-b0-51.ucode failed with error -2
[    8.721258] iwlwifi 0000:07:00.0: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-51.ucode
[    8.722800] iwlwifi 0000:07:00.0: Direct firmware load for iwlwifi-so-a0-jf-b0-50.ucode failed with error -2
[    8.723261] iwlwifi 0000:07:00.0: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-50.ucode
[    8.724744] iwlwifi 0000:07:00.0: Direct firmware load for iwlwifi-so-a0-jf-b0-49.ucode failed with error -2
[    8.725191] iwlwifi 0000:07:00.0: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-49.ucode
[    8.726760] iwlwifi 0000:07:00.0: Direct firmware load for iwlwifi-so-a0-jf-b0-48.ucode failed with error -2
[    8.727213] iwlwifi 0000:07:00.0: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-48.ucode
[    8.728800] iwlwifi 0000:07:00.0: Direct firmware load for iwlwifi-so-a0-jf-b0-47.ucode failed with error -2
[    8.729254] iwlwifi 0000:07:00.0: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-47.ucode
[    8.730802] iwlwifi 0000:07:00.0: Direct firmware load for iwlwifi-so-a0-jf-b0-46.ucode failed with error -2
[    8.731254] iwlwifi 0000:07:00.0: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-46.ucode
[    8.732917] iwlwifi 0000:07:00.0: Direct firmware load for iwlwifi-so-a0-jf-b0-45.ucode failed with error -2
[    8.733374] iwlwifi 0000:07:00.0: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-45.ucode
[    8.734940] iwlwifi 0000:07:00.0: Direct firmware load for iwlwifi-so-a0-jf-b0-44.ucode failed with error -2
[    8.735401] iwlwifi 0000:07:00.0: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-44.ucode
[    8.736943] iwlwifi 0000:07:00.0: Direct firmware load for iwlwifi-so-a0-jf-b0-43.ucode failed with error -2
[    8.737411] iwlwifi 0000:07:00.0: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-43.ucode
[    8.738951] iwlwifi 0000:07:00.0: Direct firmware load for iwlwifi-so-a0-jf-b0-42.ucode failed with error -2
[    8.739410] iwlwifi 0000:07:00.0: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-42.ucode
[... cut ...]
[    8.756054] iwlwifi 0000:07:00.0: Direct firmware load for iwlwifi-so-a0-jf-b0-39.ucode failed with error -2
[    8.756517] iwlwifi 0000:07:00.0: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-39.ucode
[    8.758359] iwlwifi 0000:07:00.0: no suitable firmware found!
[    8.758662] iwlwifi 0000:07:00.0: minimum version required: iwlwifi-so-a0-jf-b0-39
[    8.759042] iwlwifi 0000:07:00.0: maximum version supported: iwlwifi-so-a0-jf-b0-66
[    8.759424] iwlwifi 0000:07:00.0: check git://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git
```

jjm2473, 2024-02-29:
下载这个文件 https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/iwlwifi-so-a0-jf-b0-64.ucode?h=20230117 （大小1.3MB左右）
然后放在软路由的 `/lib/firmware/` 文件夹下

然后重启系统。

或者执行命令：
```shell
curl -L -o /lib/firmware/iwlwifi-so-a0-jf-b0-64.ucode 'https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/iwlwifi-so-a0-jf-b0-64.ucode?h=20230117'
ls -l /lib/firmware/iwlwifi-so-a0-jf-b0-64.ucode
[ -s /lib/firmware/iwlwifi-so-a0-jf-b0-64.ucode ] && reboot
```

SOYOJOE, 2024-03-02:
你好，已经可以了，能够正常搜索到wifi,谢谢帮助

closed 2024-03-02 by SOYOJOE

chenjin958, 2024-11-09:
我的怎么不能驱动