istoreos/istoreos #1363: 重启后docker容器丢失的问题

coolxy, 2024-02-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
会偶然出现在重启路由后docker容器丢失的问题，出现几次了，好在数据文件夹是自定义的，数据都还在。
### (1) 具体问题

A：

### (2) 路由器型号和固件版本
J1900，20240126固件和之前的20231124出现过其他固件没经历。
A：

### (3) 详细日志和/或截图

A：

jjm2473, 2024-02-27:
升级到最新固件，如果还有问题，在开机以后尽快把内核日志和系统日志发出来看看。

jjm2473, 2024-02-27:
参考 https://github.com/istoreos/istoreos/discussions/1281 “系统”第2点，在 系统-挂载点 全局设置中把“关键挂载点超时”设置成10秒或者30秒试试

coolxy, 2024-02-28:
再一次出现此情况，起因：WEB管理界面挂掉，无法进入web后台，查看进程是uhttpd进程挂了，于是service uhttpd restart就出现failed to connect to ubus，无法起启，于是手动重启路由（ssh还能进，用的reboot）
重启后容器丢失。

系统日志：
```
Wed Feb 28 09:02:37 2024 kern.notice kernel: [    0.000000] Linux version 5.10.201 (build@buildserver) (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 2023070614) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP Thu Jan 25 12:52:22 2024
Wed Feb 28 09:02:37 2024 kern.info kernel: [    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=8d98e530-02 rootwait noinitrd
Wed Feb 28 09:02:37 2024 kern.info kernel: [    0.000000] BIOS-provided physical RAM map:
Wed Feb 28 09:02:37 2024 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009d7ff] usable
Wed Feb 28 09:02:37 2024 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x000000000009d800-0x000000000009ffff] reserved
Wed Feb 28 09:02:37 2024 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
Wed Feb 28 09:02:37 2024 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000001effffff] usable
Wed Feb 28 09:02:37 2024 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x000000001f000000-0x000000001f0fffff] reserved
Wed Feb 28 09:02:37 2024 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x000000001f100000-0x000000001fffffff] usable
Wed Feb 28 09:02:37 2024 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x0000000020000000-0x00000000200fffff] reserved
Wed Feb 28 09:02:37 2024 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x0000000020100000-0x00000000b93a2fff] usable
Wed Feb 28 09:02:37 2024 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x00000000b93a3000-0x00000000b93d2fff] reserved
Wed Feb 28 09:02:37 2024 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x00000000b93d3000-0x00000000b93e2fff] ACPI data
Wed Feb 28 09:02:37 2024 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x00000000b93e3000-0x00000000b98d6fff] ACPI NVS
Wed Feb 28 09:02:37 2024 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x00000000b98d7000-0x00000000b9babfff] reserved
Wed Feb 28 09:02:37 2024 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x00000000b9bac000-0x00000000b9bacfff] usable
Wed Feb 28 09:02:37 2024 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x00000000b9bad000-0x00000000b9beefff] reserved
Wed Feb 28 09:02:37 2024 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x00000000b9bef000-0x00000000b9d60fff] usable
Wed Feb 28 09:02:37 2024 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x00000000b9d61000-0x00000000b9ff9fff] reserved
Wed Feb 28 09:02:37 2024 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x00000000b9ffa000-0x00000000b9ffffff] usable
Wed Feb 28 09:02:37 2024 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x00000000e00f8000-0x00000000e00f8fff] reserved
Wed Feb 28 09:02:37 2024 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
Wed Feb 28 09:02:37 2024 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x00000000fed08000-0x00000000fed08fff] reserved
Wed Feb 28 09:02:37 2024 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x00000000ffb00000-0x00000000ffffffff] reserved
Wed Feb 28 09:02:37 2024 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000013fffffff] usable
Wed Feb 28 09:02:37 2024 kern.info kernel: [    0.000000] NX (Execute Disable) protection: active
Wed Feb 28 09:02:37 2024 kern.info kernel: [    0.000000] SMBIOS 2.8 present.
Wed Feb 28 09:02:37 2024 kern.info kernel: [    0.000000] DMI: To be filled by O.E.M. To be filled by O.E.M./Aptio CRB, BIOS 5.6.5 05/25/2015
Wed Feb 28 09:02:37 2024 kern.info kernel: [    0.000000] tsc: Detected 2000.000 MHz processor
Wed Feb 28 09:02:37 2024 kern.debug kernel: [    0.001415] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
Wed Feb 28 09:02:37 2024 kern.debug kernel: [    0.001423] e820: remove [mem 0x000a0000-0x000fffff] usable
Wed Feb 28 09:02:37 2024 kern.info kernel: [    0.001441] last_pfn = 0x140000 max_arch_pfn = 0x400000000
Wed Feb 28 09:02:37 2024 kern.debug kernel: [    0.001447] MTRR default type: uncachable
Wed Feb 28 09:02:37 2024 kern.debug kernel: [    0.001449] MTRR fixed ranges enabled:
Wed Feb 28 09:02:37 2024 kern.debug kernel: [    0.001452]   00000-9FFFF write-back
Wed Feb 28 09:02:37 2024 kern.debug kernel: [    0.001455]   A0000-BFFFF uncachable
Wed Feb 28 09:02:37 2024 kern.debug kernel: [    0.001457]   C0000-FFFFF write-protect
Wed Feb 28 09:02:37 2024 kern.debug kernel: [    0.001459] MTRR variable ranges enabled:
Wed Feb 28 09:02:37 2024 kern.debug kernel: [    0.001462]   0 base 000000000 mask F80000000 write-back
Wed Feb 28 09:02:37 2024 kern.debug kernel: [    0.001465]   1 base 080000000 mask FC0000000 write-back
Wed Feb 28 09:02:37 2024 kern.debug kernel: [    0.001468]   2 base 0BA000000 mask FFE000000 uncachable
Wed Feb 28 09:02:37 2024 kern.debug kernel: [    0.001471]   3 base 0BC000000 mask FFC000000 uncachable
Wed Feb 28 09:02:37 2024 kern.debug kernel: [    0.001473]   4 base 100000000 mask FC0000000 write-back
Wed Feb 28 09:02:37 2024 kern.debug kernel: [    0.001475]   5 disabled
Wed Feb 28 09:02:37 2024 kern.debug kernel: [    0.001477]   6 disabled
Wed Feb 28 09:02:37 2024 kern.debug kernel: [    0.001478]   7 disabled
Wed Feb 28 09:02:37 2024 kern.info kernel: [    0.001641] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT
[... cut ...]
[    0.009864] ACPI: DSDT 0x00000000B93D7180 00ABDE (v02 ALASKA A M I    01072009 INTL 20120913)
[    0.009875] ACPI: FACS 0x00000000B98D6F80 000040
[    0.009884] ACPI: APIC 0x00000000B93E1E70 000084 (v03 ALASKA A M I    01072009 AMI  00010013)
[    0.009895] ACPI: FPDT 0x00000000B93E1EF8 000044 (v01 ALASKA A M I    01072009 AMI  00010013)
[    0.009905] ACPI: FIDT 0x00000000B93E1F40 00009C (v01 ALASKA A M I    01072009 AMI  00010013)
[    0.009915] ACPI: MCFG 0x00000000B93E1FE0 00003C (v01 ALASKA A M I    01072009 MSFT 00000097)
[    0.009926] ACPI: HPET 0x00000000B93E2020 00

coolxy, 2024-02-28:
> 升级到最新固件，如果还有问题，在开机以后尽快把内核日志和系统日志发出来看看。

重启再次出现，如上一楼，容器没有了，原来下载过的镜像也没有了

jjm2473, 2024-02-28:
你的docker路径不是在硬盘上？现在的docker数据目录是什么

coolxy, 2024-02-28:
> 你的docker路径不是在硬盘上？现在的docker数据目录是什么

数据目录是自己建的/home/docker_data/下

上面的：/overly/upper/opt/docker/
是我单独分区出来专门用于docker的，整个系统是在是个SSD硬盘上，结构如下：
```
root@iStoreOS:/home/docker_data# lsblk
NAME   MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda      8:0    0 119.2G  0 disk 
├─sda1   8:1    0   128M  0 part /boot
│                                /boot
├─sda2   8:2    0   256M  0 part /rom
├─sda3   8:3    0    20G  0 part /overlay/upper/opt/docker
│                                /overlay
├─sda4   8:4    0     1K  0 part 
├─sda5   8:5    0    94G  0 part /home
└─sda6   8:6    0   4.9G  0 part [SWAP]
```

jjm2473, 2024-02-28:
`cat /etc/config/dockerd`

jjm2473, 2024-02-28:
还有，你的overlay分区为什么有20GB，你自己扩容过？你这样以后都不用升级固件了。

coolxy, 2024-02-28:
> 还有，你的overlay分区为什么有20GB，你自己扩容过？你这样以后都不用升级固件了。

```
cat /etc/config/dockerd

config globals 'globals'
        option log_level 'warn'
        option iptables '0'
        option data_root '/overlay/upper/opt/docker'
        option remote_endpoint '0'

config firewall 'firewall'
        option device 'docker0'
        list blocked_interfaces 'wan'
```

coolxy, 2024-02-28:
> 还有，你的overlay分区为什么有20GB，你自己扩容过？你这样以后都不用升级固件了。

对，我自己扩容过，在写盘完成后扩容的

coolxy, 2024-02-28:
> 还有，你的overlay分区为什么有20GB，你自己扩容过？你这样以后都不用升级固件了。

不用升级固件 是什么意思，难道是不能再升级了么，可以升级几次都没有问题呢，

jjm2473, 2024-02-28:
扩容以后没法升级。为什么不直接使用其他分区作为docker存储目录？你管overlay干嘛？
执行`mount`看看

coolxy, 2024-02-28:
> 扩容以后没法升级。为什么不直接使用其他分区作为docker存储目录？你管overlay干嘛？ 执行`mount`看看

```
# mount
/dev/sda2 on /rom type squashfs (ro,relatime)
proc on /proc type proc (rw,nosuid,nodev,noexec,noatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,noatime)
cgroup2 on /sys/fs/cgroup type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime)
/dev/sda3 on /overlay type ext4 (rw,relatime)
overlayfs:/overlay on / type overlay (rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work)
tmpfs on /mnt type tmpfs (rw,relatime,size=4096k)
/dev/sda1 on /boot type ext4 (rw,noatime)
/dev/sda1 on /boot type ext4 (rw,noatime)
tmpfs on /dev type tmpfs (rw,nosuid,noexec,noatime,size=512k,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,noatime,mode=600,ptmxmode=000)
debugfs on /sys/kernel/debug type debugfs (rw,noatime)
none on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,noatime,mode=700)
tmpfs on /sys/module/md_mod/parameters/new_array type tmpfs (rw,nosuid,noexec,noatime,size=512k,mode=755)
/dev/sda5 on /home type ext4 (rw,relatime)
/dev/sdb2 on /D type ext4 (rw,relatime)
nfsd on /proc/fs/nfsd type nfsd (rw,relatime)
/dev/sda3 on /overlay/upper/opt/docker type ext4 (rw,relatime)
overlay on /overlay/upper/opt/docker/overlay2/edcbfe021bb03738c47dc8648cc2d776eb62f6d5e0c1d122917e59cff91ca93d/merged type overlay (rw,relatime,lowerdir=/overlay/upper/opt/docker/overlay2/l/GAGQUPHX7GCNLNTSBI5BHYVGKV:/overlay/upper/opt/docker/overlay2/l/RAUKS2HBMIOSXOVELURLTUZQWV:/overlay/upper/opt/docker/overlay2/l/CYARZ7F3SXGJVMZFMIGYKHOKQ7:/overlay/upper/opt/docker/overlay2/l/CRXDY7BB34G7KSQ3IDN6HACPHE,upperdir=/overlay/upper/opt/docker/overlay2/edcbfe021bb03738c47dc8648cc2d776eb62f6d5e0c1d122917e59cff91ca93d/diff,workdir=/overlay/upper/opt/docker/overlay2/edcbfe021bb03738c47dc8648cc2d776eb62f6d5e0c1d122917e59cff91ca93d/work)
overlay on /overlay/upper/opt/docker/overlay2/1c053225a6346355498607e02143690ab0ee0423155787ec1fc955d3beecaf82/merged type overlay (rw,relatime,lowerdir=/overlay/upper/opt/docker/overlay2/l/WRTN6MYZYHA7VCX3JVDBITT7NU:/overlay/upper/opt/docker/overlay2/l/RAUKS2HBMIOSXOVELURLTUZQWV:/overlay/upper/opt/docker/overlay2/l/CYARZ7F3SXGJVMZFMIGYKHOKQ7:/overlay/upper/opt/docker/overlay2/l/CRXDY7BB34G7KSQ3IDN6HACPHE,upperdir=/overlay/upper/opt/docker/overlay2/1c053225a6346355498607e02143690ab0ee0423155787ec1fc955d3beecaf82/diff,workdir=/overlay/upper/opt/docker/overlay2/1c053225a6346355498607e02143690ab0ee0423155787ec1fc955d3beecaf82/work)
overlay on /overlay/upper/opt/docker/overlay2/41bfb418636eb15d0e87b0bb5e997b781f1eeb20a68446d8acbd069dfbbb13d2/merged type overlay (rw,relatime,lowerdir=/overlay/upper/opt/docker/overlay2/l/SBEDV5GWWR4V2C3ER6NQ6YNTQ2:/overlay/upper/opt/docker/overlay2/l/AELSHBY7GS2JEXAGUSQBI4AEOH:/overlay/upper/opt/docker/overlay2/l/HW5R47U4JFMQNUPODZOY7W7JCS:/overlay/upper/opt/docker/overlay2/l/QLKYRO4EK2E2DMWGGRIZ4WSK2V:/overlay/upper/opt/docker/overlay2/l/HFXJX26XBE4IG3L4A54TS37JTZ:/overlay/upper/opt/docker/overlay2/l/FJLZ4TJABMZNSX3CCM73RAW5GN:/overlay/upper/opt/docker/overlay2/l/BEMIJEI2ZVYNAKIIL2EBJPNTGC,upperdir=/overlay/upper/opt/docker/overlay2/41bfb418636eb15d0e87b0bb5e997b781f1eeb20a68446d8acbd069dfbbb13d2/diff,workdir=/overlay/upper/opt/docker/overlay2/41bfb418636eb15d0e87b0bb5e997b781f1eeb20a68446d8acbd069dfbbb13d2/work)
nsfs on /tmp/run/docker/netns/a2884ed99d7e type nsfs (rw)
nsfs on /tmp/run/docker/netns/11a9b22ab563 type nsfs (rw)
nsfs on /tmp/run/docker/netns/5c978f64bc64 type nsfs (rw)
overlay on /overlay/upper/opt/docker/overlay2/f8bdef88f78d8f57b57367b80ba4b502f7b15dbe723a9eaea4813a1875d05f99/merged type overlay (rw,relatime,lowerdir=/overlay/upper/opt/docker/overlay2/l/4M7TLREYZBCM4T3VR6VRLT3HIZ:/overlay/upper/opt/docker/overlay2/l/UYO3GYVIUGG3IOXV2AMNQ7HYHD:/overlay/upper/opt/docker/overlay2/l/HCRFSE7BJ7NUOSHREIN2MT543E:/overlay/upper/opt/docker/overlay2/l/U4EXBCB7LCGA565JWJHNCM7EAU:/overlay/upper/opt/docker/overlay2/l/AOBTPLUD3CB7AGUSGS6TO5BCV6:/overlay/upper/opt/docker/overlay2/l/4ZSANFXTM5XAHSE3YLKZJJ3DQC:/overlay/upper/opt/docker/overlay2/l/VHFEEJFK3R56I3VSSSXOKK6KOG:/overlay/upper/opt/docker/overlay2/l/ODFPV72ODZLAFM2FMLGX4WF452:/overlay/upper/opt/docker/overlay2/l/MJORPHUM5SCANXSXMGIVPGHZPO:/overlay/upper/opt/docker/overlay2/l/LEGS6T55IBGYFMCDH6LXXUI4UT:/overlay/upper/opt/docker/overlay2/l/2GFSAULAJHGAFFVZ6ZGKNWMJIG:/overlay/upper/opt/docker/overlay2/l/NOQZV6Q2PPR5DCH72SC3ZS3YVT:/overlay/upper/opt/docker/overlay2/l/MDSDFKNS4PUNZRF5ORV4MVINFY:/overlay/upper/opt/docker/overlay2/l/TQ4REYBXMLFIBG7VPLRXODRTP2:/overlay/upper/opt/docker/overlay2/l/AEATWRXQOT3IK4J44ECYBTL2GK:/overlay/upper/opt/docker/overlay2/l/IQYTS4ONQW5MDXQQP2XXGCBFWV:/overlay/upper/opt/docker/overlay2/l/ZIAEZYSRE5AMI6MGA2AL2DPARY:/overlay/upper/opt/docker/overlay2/l/JVGJDF7ICYES7GK3W4HTS5OLWJ:/overlay/upper/opt/docker/overlay2/l/XHUO2T2OHVV5OA6UGSKF7DXES2:/overlay/upper/opt/docker/overlay2/l/WP76LFEMERE6TCO3XPYA2RXIME:/overlay/upper/opt/docker/overlay2/l/XO52TDTUJ5MZSL6FMRJSVSM4AC:/overlay/upper/opt/docker/overlay2/l/ATNO27UYM7H7KLNW2ZBNKLXL35:/overlay/upper/opt/docker/overlay2/l/LQR5643GCNLUNJE5JHXK2JGJHX:/overlay/upper/opt/docker/overlay2/l/VD5BMXSYPIZMIVUGGI5N4IOCMY:/overlay/upper/opt/docker/overlay2/l/XTV4RCSKSRYUHYYQDAJJQNYKYM:/overlay/upper/opt/docker/overlay2/l/NEYO55XAB4TIHRU3J7PHQBWNCW:/overlay/upper/opt/docker/overlay2/l/PTCF3TEJURBWGD3CSBLTKPCD2S:/overlay/upper/opt/docker/overlay2/l/3LSFI5KQBGWQTH5UXUEYYAHBZS:/overlay/upper/opt/docker/overlay2/l/J4DL4QS2YZJ3MUOTZZ34QY52NB:/overlay/upper/opt/docker/overlay2/l/BQEGI2M6ZR4O2CIBA2LFXEHU5F:/overlay/upper/opt/docker/overlay2/l/5OROXH4PPJFEWDPZZVBKDTTBOY,upperdir=/overlay/upper/opt/docker/overlay2/f8bdef88f78d8f57b57367b80ba4b502f7b15dbe723a9eaea4813a1875d05f99/diff,workdir=/overlay/upper/opt/docker/overlay2/f8bdef88f78d8f57b57367b80ba4b502f7b15dbe723a9eaea4813a1875d05f99/work)
nsfs on /tmp/run/docker/netns/default type nsfs (rw)
```

jjm2473, 2024-02-28:
执行这些命令看看
```shell
docker run hello-world
sync /overlay/upper/opt/docker
dmesg | tail
```

coolxy, 2024-02-28:
> 执行这些命令看看
> 
> ```shell
> docker run hello-world
> sync /overlay/upper/opt/docker
> dmesg | tail
> ```

执行如下：
```
root@iStoreOS:~# docker run hello-world

Hello from Docker!
This message shows that your installation appears to be working correctly.

To generate this message, Docker took the following steps:
 1. The Docker client contacted the Docker daemon.
 2. The Docker daemon pulled the "hello-world" image from the Docker Hub.
    (amd64)
 3. The Docker daemon created a new container from that image which runs the
    executable that produces the output you are currently reading.
 4. The Docker daemon streamed that output to the Docker client, which sent it
    to your terminal.

To try something more ambitious, you can run an Ubuntu container with:
 $ docker run -it ubuntu bash

Share images, automate workflows, and more with a free Docker ID:
 https://hub.docker.com/

For more examples and ideas, visit:
 https://docs.docker.com/get-started/
```

```
root@iStoreOS:~# sync /overlay/upper/opt/docker
root@iStoreOS:~# 
```

```
root@iStoreOS:~# dmesg | tail
[18571.676162] docker0: port 1(veth8c21598) entered disabled state
[18571.676561] veth52e625a: renamed from eth0
[18571.775459] docker0: port 1(veth8c21598) entered disabled state
[18571.780300] device veth8c21598 left promiscuous mode
[18571.780307] docker0: port 1(veth8c21598) entered disabled state
[18571.865002] docker0: port 2(veth5a447ae) entered disabled state
[18571.865437] vethc297e13: renamed from eth0
[18571.975882] docker0: port 2(veth5a447ae) entered disabled state
[18571.982431] device veth5a447ae left promiscuous mode
[18571.982439] docker0: port 2(veth5a447ae) entered disabled state
```

jjm2473, 2024-02-28:
没看到什么错误，等下次出问题别重启，把日志发出来

coolxy, 2024-02-28:
> 没看到什么错误，等下次出问题别重启，把日志发出来

好的，好像与luci界面挂掉有关，日常重启没问题，uhttpd挂掉后重启就有问题了。

jjm2473, 2024-02-28:
那就不是单独docker的问题啊，应该文件系统坏了

jjm2473, 2024-02-28:
你的系统盘是通过usb连接的吗？

coolxy, 2024-02-29:
> 你的系统盘是通过usb连接的吗？

不是，是msata接口的系统盘。

coolxy, 2024-02-29:
> 那就不是单独docker的问题啊，应该文件系统坏了

嗯，我更新系统后再看看，如果不行就重装系统，不再扩容试试

jjm2473, 2024-02-29:
如果你的硬盘本来就有问题，例如扩容盘，那写到后面可能就损坏数据了，所以还是先检测硬盘是不是正常的，扫一下扩容和坏块。还有确保供电足够

marine19811006, 2024-03-06:
如果是虚拟机，在虚拟机下重启系统就好了，反正不够稳定，主动重启又会丢失，后来没办法重装了就好了

dddd64693, 2024-04-17:
我也遇到和你一样的情况了，还不知道怎么解决

coolxy, 2024-04-28:
> 我也遇到和你一样的情况了，还不知道怎么解决

我又再一次出现问题，重新刷了系统，不再进行扩容，而是直接将剩余的硬盘空间挂载后，将DOCKER移到新挂载的空间内，目前未发现新问题。

要看你是因何情况出的问题

dddd64693, 2024-04-28:
我没扩容过，就是把硬盘的剩余空间挂载，再转移 docker 的，另外挂载了 3 个其它硬盘做其它空间用

发自我的 iPhone