istoreos/istoreos #1365: 路由狗安装卸载后首页进不去

antbuster, 2024-02-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
路由狗安装卸载后首页进不去

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：路由狗安装卸载后首页进不去，报以下错误
```
/usr/lib/lua/luci/template.lua:74: Failed to load template 'routerdog/index'.
Error while parsing template '/usr/lib/lua/luci/view/routerdog/index.htm':
No such file or directory
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:74: in function '__init__'
	/usr/lib/lua/luci/util.lua:65: in function 'Template'
	/usr/lib/lua/luci/template.lua:27: in function 'render'
	/usr/lib/lua/luci/dispatcher.lua:1052: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:482: in function </usr/lib/lua/luci/dispatcher.lua:481>
```
### (2) 路由器型号和固件版本
路由器型号:FriendlyElec NanoPi R5C
固件版本:iStoreOS 22.03.6 2024012613

Carseason, 2024-02-25:
rm -rf /tmp/luci*

jjm2473, 2024-02-26:
你是不是自己升级过luci-base或其他系统软件包？

antbuster, 2024-02-27:
> rm -rf /tmp/luci*

可以了

closed 2024-02-27 by antbuster