istoreos/istoreos #1366: R4S使用RTL8156B的2.5G usb 网卡断流

zjqwhy, 2024-02-25:
希望能添加最新的kmod-usb-net-rtl8152-vendor。

jjm2473, 2024-02-25:
现在就是这个驱动，早就不用内核自带的r8152了，只是因为替换了内核驱动，所以驱动名称是r8152

zjqwhy, 2024-02-26:
> 现在就是这个驱动，早就不用内核自带的r8152了，只是因为替换了内核驱动，所以驱动名称是r8152

谢谢，如果不是驱动的问题的话，能帮我看这个到底是什么造成的。
[image]
[image]

zjqwhy, 2024-02-26:
> 现在就是这个驱动，早就不用内核自带的r8152了，只是因为替换了内核驱动，所以驱动名称是r8152

使用 ethtool -K eth1 tx 
[image]
[image]

off rx off后有所改善，但是下载文件时间一长就卡死了。

jjm2473, 2024-02-27:
可能是usb有问题，反正rk的usb有问题也不奇怪，之前r2s的usb网卡甚至都没法识别，打了rk的补丁才稳定。

jjm2473, 2024-02-27:
你测试下openwrt 22.03.6有没有这个问题，如果有，就反馈给openwrt，r4s是openwrt官方维护的机型

closed 2024-02-27 by jjm2473