istoreos/istoreos #1369: AX210 网卡内核报错，无法驱动

zhangbaiquan, 2024-02-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

AX 210 第6版无线网卡添加对应驱动后内核报错，查看驱动目录已默认有AX210驱动，dmesg|grep -i wifi  查看内核报错，更换不同版本驱动后同样报错，删除驱动后又正常报缺少驱动，添加要求的驱动后又报错。

这三个文件都加进驱动目录里也还是报错。
[image]

[image]

### (1) 具体问题

A：AX 210 第6版无线网卡添加对应驱动后内核报错，参考以下日志。

### (2) 路由器型号和固件版本

A：ESXI 8.0  
      iStoreOS：istoreos-22.03.5-2023122916-x86-64-squashfs-combined-efi

### (3) 详细日志和/或截图

[image]

A：有驱动时报如下错误：
root@iStoreOS:~# dmesg|grep -i wifi
[    9.440203] Intel(R) Wireless WiFi driver for Linux
[    9.440657] iwlwifi 0000:02:03.0: enabling device (0000 -> 0002)
[    9.544988] iwlwifi 0000:02:03.0: api flags index 2 larger than supported by driver
[    9.545473] iwlwifi 0000:02:03.0: TLV_FW_FSEQ_VERSION: FSEQ Version: 0.63.2.2
[    9.546333] iwlwifi 0000:02:03.0: loaded firmware version 66.f1c864e0.0 ty-a0-gf-a0-66.ucode op_mode iwlmvm
[   12.807596] iwlwifi 0000:02:03.0: Detected Intel(R) Wi-Fi 6 AX210 160MHz, REV=0x420
[   13.839675] iwlwifi 0000:02:03.0: SecBoot CPU1 Status: 0x7ea9, CPU2 Status: 0xb03
[   13.843714] iwlwifi 0000:02:03.0: UMAC PC: 0xc0080b7c
[   13.844409] iwlwifi 0000:02:03.0: LMAC PC: 0x4d0cce
[   13.845072] iwlwifi 0000:02:03.0: WRT: Collecting data: ini trigger 13 fired (delay=0ms).
[   14.089993] iwlwifi 0000:02:03.0: Start IWL Error Log Dump:
[   14.092822] iwlwifi 0000:02:03.0: Transport status: 0x00000042, valid: 6
[   14.093338] iwlwifi 0000:02:03.0: Loaded firmware version: 66.f1c864e0.0 ty-a0-gf-a0-66.ucode
[   14.093867] iwlwifi 0000:02:03.0: 0x00000084 | NMI_INTERRUPT_UNKNOWN       
[   14.094391] iwlwifi 0000:02:03.0: 0x002082F0 | trm_hw_status0
[   14.094930] iwlwifi 0000:02:03.0: 0x00000000 | trm_hw_status1
[   14.095435] iwlwifi 0000:02:03.0: 0x004DA722 | branchlink2
[   14.095964] iwlwifi 0000:02:03.0: 0x004D0CCE | interruptlink1
[   14.096472] iwlwifi 0000:02:03.0: 0x004D0CCE | interruptlink2
[   14.096972] iwlwifi 0000:02:03.0: 0x004D94DA | data1
[   14.097453] iwlwifi 0000:02:03.0: 0x01000000 | data2
[   14.097926] iwlwifi 0000:02:03.0: 0x00000000 | data3
[   14.098387] iwlwifi 0000:02:03.0: 0x00000000 | beacon time
[   14.098887] iwlwifi 0000:02:03.0: 0x000E6A38 | tsf low
[   14.099343] iwlwifi 0000:02:03.0: 0x00000000 | tsf hi
[   14.099822] iwlwifi 0000:02:03.0: 0x00000000 | time gp1
[   14.101473] iwlwifi 0000:02:03.0: 0x000FAA47 | time gp2
[   14.103230] iwlwifi 0000:02:03.0: 0x00000001 | uCode revision type
[   14.103709] iwlwifi 0000:02:03.0: 0x00000042 | uCode version major
[   14.104170] iwlwifi 0000:02:03.0: 0xF1C864E0 | uCode version minor
[   14.104624] iwlwifi 0000:02:03.0: 0x00000420 | hw version
[   14.105073] iwlwifi 0000:02:03.0: 0x18C89002 | board version
[   14.105519] iwlwifi 0000:02:03.0: 0x8004FC12 | hcmd
[   14.105976] iwlwifi 0000:02:03.0: 0x20020000 | isr0
[   14.106408] iwlwifi 0000:02:03.0: 0x00000000 | isr1
[   14.106831] iwlwifi 0000:02:03.0: 0x00F00002 | isr2
[   14.107240] iwlwifi 0000:02:03.0: 0x00C0000C | isr3
[   14.107637] iwlwifi 0000:02:03.0: 0x00000000 | isr4
[   14.108021] iwlwifi 0000:02:03.0: 0x00000000 | last cmd Id
[... 245 lines cut ...]
[   16.578934] iwlwifi 0000:02:03.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-48.ucode failed with error -2
[   16.582158] iwlwifi 0000:02:03.0: Falling back to sysfs fallback for: iwlwifi-ty-a0-gf-a0-48.ucode
[   16.587087] iwlwifi 0000:02:03.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-47.ucode failed with error -2
[   16.587941] iwlwifi 0000:02:03.0: Falling back to sysfs fallback for: iwlwifi-ty-a0-gf-a0-47.ucode
[   16.593866] iwlwifi 0000:02:03.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-46.ucode failed with error -2
[   16.597126] iwlwifi 0000:02:03.0: Falling back to sysfs fallback for: iwlwifi-ty-a0-gf-a0-46.ucode
[   16.608201] iwlwifi 0000:02:03.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-45.ucode failed with error -2
[   16.611405] iwlwifi 0000:02:03.0: Falling back to sysfs fallback for: iwlwifi-ty-a0-gf-a0-45.ucode
[   16.615084] iwlwifi 0000:02:03.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-44.ucode failed with error -2
[   16.615968] iwlwifi 0000:02:03.0: Falling back to sysfs fallback for: iwlwifi-ty-a0-gf-a0-44.ucode
[   16.631984] iwlwifi 0000:02:03.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-43.ucode failed with error -2
[   16.635373] iwlwifi 0000:02:03.0: Falling back to sysfs fallback for: iwlwifi-ty-a0-gf-a0-43.ucode
[   16.647762] iwlwifi 0000:02:03.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-42.ucode failed with error -2
[   16.651036] iwlwifi 0000:02:03.0: Falling back to sysfs fallback for: iwlwifi-ty-a0-gf-a0-42.ucode
[   16.659457] iwlwifi 0000:02:03.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-41.ucode failed with error -2
[   16.662842] iwlwifi 0000:02:03.0: Falling back to sysfs fallback for: iwlwifi-ty-a0-gf-a0-41.ucode
[   16.686004] iwlwifi 0000:02:03.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-40.ucode failed with error -2
[   16.689351] iwlwifi 0000:02:03.0: Falling back to sysfs fallback for: iwlwifi-ty-a0-gf-a0-40.ucode
[   16.708414] iwlwifi 0000:02:03.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-39.ucode failed with error -2
[   16.711700] iwlwifi 0000:02:03.0: Falling back to sysfs fallback for: iwlwifi-ty-a0-gf-a0-39.ucode
[   16.719746] iwlwifi 0000:02:03.0: no suitable firmware found!
[   16.723096] iwlwifi 0000:02:03.0: minimum version required: iwlwifi-ty-a0-gf-a0-39
[   16.723990] iwlwifi 0000:02:03.0: maximum version supported: iwlwifi-ty-a0-gf-a0-66
[   16.724841] iwlwifi 0000:02:03.0: check git://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git
root@iStoreOS:~#

jjm2473, 2024-02-27:
好像虚拟机才有问题，试试物理机

zhangbaiquan, 2024-02-27:
就是虚拟机有问题，esxi8下win10也驱动不了。虚拟机有法吗

zhangbaiquan, 2024-02-27:
> 好像虚拟机才有问题 https://github.com/istoreos/istoreos/issues/1304 ，试试物理机

就是虚拟机有问题，esxi8下win10也驱动不了。虚拟机有法吗

jjm2473, 2024-02-27:
这个问题可能是宿主机系统导致的，可以试试pve或其他虚拟化环境；也可能是客户机的内核或者驱动问题，虚拟机试试安装其他系统看看，例如lean，openwrt

zhangbaiquan, 2024-02-27:
> 这个问题可能是宿主机系统导致的，可以试试pve或其他虚拟化环境；也可能是客户机的内核或者驱动问题，虚拟机试试安装其他系统看看，例如lean，openwrt

我尝试了一些其他的openwrt系统，也都是一样的错误。关键错误信息：Failed to start RT ucode: -110 我找到一篇相同报错的帖子，提供下参考。

[image]

https://notes.leconiot.com/intel_ac_9560_driver_excepiton.html

jjm2473, 2024-03-01:
重复issue，关了

closed 2024-03-01 by jjm2473

HernandoR, 2024-03-09:
Is or will AX210 can be usable as AP?

jjm2473, 2024-03-11:
@HernandoR All intel wifi chips do no support 5GHz AP (maybe some support 2.4Ghz AP) https://wireless.wiki.kernel.org/en/users/drivers/iwlwifi