istoreos/istoreos #1372: X86版本端口映射全部失效

xiaobaishangche, 2024-02-28:
之前在老版本里用的一点问题都没有，后来升级到了新版本后，端口全部无法外网访问，恢复备份文件到以前老版本后，就可以正常访问了，主路由是爱快旁路由istoreos，由istoreos分配DHCp，还望大佬解决

jjm2473, 2024-02-28:
你用旁路由端口转发？是不是搞笑啊

xiaobaishangche, 2024-02-28:
> 你用旁路由端口转发？是不是搞笑啊
注意审题，旁路由怎么可能做端口转发，怎么转出去，我拿主路由做的转发

jjm2473, 2024-02-28:
那你反馈给istoreos干嘛？

xiaobaishangche, 2024-02-28:
> 你用旁路由端口转发？是不是搞笑啊

旁路由做的dhcp，不是ddns

jjm2473, 2024-02-28:
dhcp跟端口转发有关系吗？

xiaobaishangche, 2024-02-28:
> 那你反馈给istoreos干嘛？

你是问题一点都不看，把istoreos做备份降级处理以后才正常

jjm2473, 2024-02-28:
dhcp只是给内网的设备分配下ip而已，端口转发不是主路由的事吗

jjm2473, 2024-02-28:
那也是主路由的问题啊，可能NAT不完整。你确定主路由能访问要转发的内网端口吗？

xiaobaishangche, 2024-02-28:
> 那也是主路由的问题啊，可能NAT不完整。你确定主路由能访问要转发的内网端口吗？

全部排查过了，最后实在没办法才降级的，降级后立马正常了，你看这是谁的问题

jjm2473, 2024-02-28:
我觉得是主路由的问题

jjm2473, 2024-02-28:
这跟istoreos的端口映射有个毛线关系

xiaobaishangche, 2024-02-28:
> 这跟istoreos的端口映射有个毛线关系
哎，算了，可能我们两个的拓扑结构不一样，我的我网络结构是爱快到op到终端

jjm2473, 2024-02-28:
你要测试的是主路由能不能访问相应的内网端口，如果可以，那只能是主路由的端口转发有问题了。如果不能访问，那么就在相应的内网设备抓包，看看主路由发过来的包是不是源地址和目的地址都NAT了。

jjm2473, 2024-02-28:
如果是运营商阻断，那怎么测都没用

xiaobaishangche, 2024-02-28:
> 如果是运营商阻断，那怎么测都没用

运营商没有阻断，现在就先这么用着，就是过来反馈一下，遇到这个问题
之前有试过关闭op，由主路由分配dhcp后，也是正常端口转发

jjm2473, 2024-02-28:
是不是只是因为内网ip变了，你的端口转发没更新？

xiaobaishangche, 2024-02-28:
> 是不是只是因为内网ip变了，你的端口转发没更新？

不会的，所有要穿透内网都都设置了静态ip

xiaobaishangche, 2024-02-28:
> 是不是只是因为内网ip变了，你的端口转发没更新？

总之感谢你的解答，但确实好像是旁路由的问题

jjm2473, 2024-02-28:
我估计是主路由的NAT不完整，保留了外网源IP，导致了内网设备返回的数据被旁路由又转发了一次，因为旁路由DHCP的情况下，默认网关是旁路由，而不是主路由。所以，这个问题要解决的话，就是让旁路由的DHCP通告主路由为默认网关，而不是通告自己为默认网关。

具体到openwrt或istoreos，就是打开网络-接口，编辑LAN，DHCP服务器-高级设置，在“DHCP 选项”中输入“3,192.168.100.1”（假设你的主路由是192.168.100.1），点击加号，然后保存应用。然后内网设备都重新联网。

jjm2473, 2024-02-28:
如果需要更精细地控制某些设备通过旁路由翻墙可以参考 https://github.com/istoreos/istoreos/discussions/1281#discussioncomment-8177283

但是，你的主路由的NAT不完整可能还会导致其他问题。还不如看看主路由的NAT或者端口转发中有没有相应的设置，避免转发的时候保留源IP。

xiaobaishangche, 2024-02-28:
> 我估计是主路由的NAT不完整，保留了外网源IP，导致了内网设备返回的数据被旁路由又转发了一次，因为旁路由DHCP的情况下，默认网关是旁路由，而不是主路由。所以，这个问题要解决的话，就是让旁路由的DHCP通告主路由为默认网关，而不是通告自己为默认网关。
> 
> 具体到openwrt或istoreos，就是打开网络-接口，编辑LAN，DHCP服务器-高级设置，在“DHCP 选项”中输入“3,192.168.100.1”（假设你的主路由是192.168.100.1），点击加号，然后保存应用。然后内网设备都重新联网。

好的，我去试试，感谢大佬解答，就是刚开始说话有点冲😂

jjm2473, 2024-02-28:
因为这根本就不istoreos端口转发的问题啊

xiaobaishangche, 2024-02-28:
> 因为这根本就不istoreos端口转发的问题啊

好吧，我回去去操作试一下，从新排查一下

jjm2473, 2024-02-28:
参考这个，配置爱快的NAT规则 https://tech.joyzhanglife.com/index.php/archives/74/#menu_index_4

xiaobaishangche, 2024-02-28:
> 参考这个，配置爱快的NAT规则 https://tech.joyzhanglife.com/index.php/archives/74/#menu_index_4
> 
> “NAT 地址”应该是填主路由的内网IP

好的，谢谢，让您费心了

marine19811006, 2024-03-04:
我升级到了最新版也出了这个问题，所有端口都无法访问了，主路由设置好了端口映射的，没改过，突然就不能访问了

jjm2473, 2024-03-04:
又是爱快是吧

marine19811006, 2024-03-06:
回退到老版本就好了，这个版本确实有点问题

jjm2473, 2024-03-07:
都说了是爱快的NAT问题，如果istoreos老版本固件没问题，反而可能是因为istoreos老版本固件哪里有bug。还有，麻烦提供新版本和老版本的版本号，也许我还能找到到底哪个变更影响的。以及，你们说的刷老版本，是直接保留配置刷的吗，还是重做了系统，然后导入之前的备份文件？

xiaobaishangche, 2024-03-07:
> 都说了是爱快的NAT问题，如果istoreos老版本固件没问题，反而可能是因为istoreos老版本固件哪里有bug。还有，麻烦提供新版本和老版本的版本号，也许我还能找到到底哪个变更影响的。以及，你们说的刷老版本，是直接保留配置刷的吗，还是重做了系统，然后导入之前的备份文件？

我的是pve虚拟机直接回滚备份

xiaobaishangche, 2024-03-07:
> 回退到老版本就好了，这个版本确实有点问题

我现在从双路由改为单路由，然后就没出现问题了，但现在出现，过段时间会有断网现象😂

jjm2473, 2024-03-07:
@xiaobaishangche 你没试过爱快增加NAT规则？

xiaobaishangche, 2024-03-07:
> @xiaobaishangche 你没试过爱快增加NAT规则？

加了，有作用，我现在升级1000m宽带，虚拟网卡已经跑不满了，所以改成单路由了

jjm2473, 2024-03-07:
@xiaobaishangche 试试virtio虚拟网卡，这种网卡不限速

xiaobaishangche, 2024-03-07:
> @xiaobaishangche 试试virtio虚拟网卡，这种网卡不限速

试过了没用，双工不行，双工速度就在70mb左右

jjm2473, 2024-03-07:
那可能分配的cpu配额太少，看看cpu使用率

xiaobaishangche, 2024-03-07:
> 那可能分配的cpu配额太少，看看cpu使用率

4核全分配，i5-7400，不是CPU问题，我在网上查询过，虚拟化情况下双工只能到700m左右

jjm2473, 2024-03-07:
不是吧，virtio不是效率很高的吗，要么再看看有没有vmxnet3，这个是万兆虚拟网卡