istoreos/istoreos #1373: 能否增加RTL8852的驱动

NokiPeng, 2024-02-28:
添加RTL8852驱动支持，期待中……

jjm2473, 2024-02-28:
又是usb无线网卡？

jjm2473, 2024-02-28:
如果你真要搞wifi，建议用MTK的，realtek的驱动依托答辩

NokiPeng, 2024-02-28:
pci-e的，OpenWrt已支持，看一下可不可以移植过来

jjm2473, 2024-02-28:
OpenWrt支持了吗，哪个版本，我看看有没有驱动

jjm2473, 2024-02-28:
目前驱动支持程度咋样，能否作为AP，是否支持双频（双频同时在线或者同一时间只能单频在线）？

NokiPeng, 2024-02-28:
现在用的固件是这位大神编译的，AP基本可以，支持双频
https://www.right.com.cn/forum/thread-8307497-1-1.html
[image]

而最新版本的OpenWRT是加入了RTL8852的支持，在编译菜单中发现
[image]

jjm2473, 2024-02-28:
看一下“无线”页面有几个wifi设备

jjm2473, 2024-02-28:
这是只有一个“radio0”，大概不支持双频同时在线

jjm2473, 2024-02-28:
试试这个固件，如果有问题就把内核日志发出来 https://we.tl/t-0pqECsfMSb

jjm2473, 2024-02-29:
测试了没，结果怎样

jjm2473, 2024-02-29:
无线这里点一下重启，然后再把内核日志发出来看看

NokiPeng, 2024-02-29:
内核日志
[ 4994.822135] br-lan: port 2(wlan0) entered blocking state
[ 4994.822137] br-lan: port 2(wlan0) entered disabled state
[ 4994.822192] device wlan0 entered promiscuous mode
[ 4994.822575] device wlan0 left promiscuous mode
[ 4994.822590] br-lan: port 2(wlan0) entered disabled state

系统日志
Thu Feb 29 20:05:20 2024 daemon.debug ModemManager[30498]: hotplug: event reported: action=remove, name=wlan0, subsystem=net
Thu Feb 29 20:05:20 2024 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0) --> new PHY
Thu Feb 29 20:05:20 2024 daemon.debug ModemManager[30555]: hotplug: event reported: action=add, name=wlan0, subsystem=net
Thu Feb 29 20:05:20 2024 daemon.notice hostapd: Frequency 5300 (primary) not allowed for AP mode, flags: 0x217b NO-IR RADAR
Thu Feb 29 20:05:20 2024 daemon.err hostapd: Primary frequency not allowed
Thu Feb 29 20:05:20 2024 daemon.warn hostapd: wlan0: IEEE 802.11 Configured channel (60) or frequency (5300) (secondary_channel=0) not found from the channel list of the current mode (2) IEEE 802.11a
Thu Feb 29 20:05:20 2024 daemon.warn hostapd: wlan0: IEEE 802.11 Hardware does not support configured channel
Thu Feb 29 20:05:20 2024 daemon.err hostapd: Could not select hw_mode and channel. (-3)
Thu Feb 29 20:05:20 2024 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->DISABLED
Thu Feb 29 20:05:20 2024 daemon.notice hostapd: wlan0: AP-DISABLED
Thu Feb 29 20:05:20 2024 daemon.err hostapd: wlan0: Unable to setup interface.
Thu Feb 29 20:05:20 2024 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Thu Feb 29 20:05:20 2024 kern.info kernel: [ 4912.847330] br-lan: port 2(wlan0) entered blocking state
Thu Feb 29 20:05:20 2024 kern.info kernel: [ 4912.847332] br-lan: port 2(wlan0) entered disabled state
Thu Feb 29 20:05:20 2024 kern.info kernel: [ 4912.847380] device wlan0 entered promiscuous mode
Thu Feb 29 20:05:20 2024 kern.info kernel: [ 4912.847730] device wlan0 left promiscuous mode
Thu Feb 29 20:05:20 2024 kern.info kernel: [ 4912.847750] br-lan: port 2(wlan0) entered disabled state
Thu Feb 29 20:05:20 2024 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Thu Feb 29 20:05:20 2024 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Thu Feb 29 20:05:20 2024 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0 wasn't started
Thu Feb 29 20:05:20 2024 daemon.notice netifd: radio0 (30476): Command failed: ubus call hostapd config_add {"iface":"wlan0", "config":"/var/run/hostapd-phy0.conf"} (Invalid argument)
Thu Feb 29 20:05:20 2024 daemon.notice netifd: radio0 (30476): Usage: ubus [<options>] <command> [arguments...]
Thu Feb 29 20:05:20 2024 daemon.notice netifd: radio0 (30476): Options:
Thu Feb 29 20:05:20 2024 daemon.notice netifd: radio0 (30476):  -s <socket>:		Set the unix domain socket to connect to
Thu Feb 29 20:05:20 2024 daemon.notice netifd: radio0 (30476):  -t <timeout>:		Set the timeout (in seconds) for a command to complete
Thu Feb 29 20:05:20 2024 daemon.notice netifd: radio0 (30476):  -S:			Use simplified output (for scripts)
Thu Feb 29 20:05:20 2024 daemon.notice netifd: radio0 (30476):  -v:			More verbose output
Thu Feb 29 20:05:20 2024 daemon.notice netifd: radio0 (30476):  -m <type>:		(for monitor): include a specific message type
Thu Feb 29 20:05:20 2024 daemon.notice netifd: radio0 (30476): 			(can be used more than once)
Thu Feb 29 20:05:20 2024 daemon.notice netifd: radio0 (30476):  -M <r|t>		(for monitor): only capture received or transmitted traffic
Thu Feb 29 20:05:20 2024 daemon.notice netifd: radio0 (30476):
Thu Feb 29 20:05:20 2024 daemon.notice netifd: radio0 (30476): Commands:
Thu Feb 29 20:05:20 2024 daemon.notice netifd: radio0 (30476):  - list [<path>]			List objects
Thu Feb 29 20:05:20 2024 daemon.notice netifd: radio0 (30476):  - call <path> <method> [<message>]	Call an object method
Thu Feb 29 20:05:20 2024 daemon.notice netifd: radio0 (30476):  - subscribe <path> [<path>...]	Subscribe to object(s) notifications
Thu Feb 29 20:05:20 2024 daemon.notice netifd: radio0 (30476):  - listen [<path>...]			Listen for events
Thu Feb 29 20:05:20 2024 daemon.notice netifd: radio0 (30476):  - send <type> [<message>]		Send an event
Thu Feb 29 20:05:20 2024 daemon.notice netifd: radio0 (30476):  - wait_for <object> [<object>...]	Wait for multiple objects to appear on ubus
Thu Feb 29 20:05:20 2024 daemon.notice netifd: radio0 (30476):  - monitor				Monitor ubus traffic
Thu Feb 29 20:05:20 2024 daemon.notice netifd: radio0 (30476):
Thu Feb 29 20:05:20 2024 daemon.notice netifd: radio0 (30476): Device setup failed: HOSTAPD_START_FAILED
Thu Feb 29 20:05:20 2024 daemon.notice netifd: Wireless device 'radio0' set retry=0
Thu Feb 29 20:05:20 2024 daemon.crit netifd: Wireless device 'radio0' setup failed, retry=0
Thu Feb 29 20:05:20 2024 daemon.notice netifd: Wireless device 'radio0' is now down
Thu Feb 29 20:05:23 2024 daemon.info [7997]: <info>  [base-manager] couldn't check support for device '/sys/devices/pci0000:00/0000:00:10.0': not supported by any plugin

NokiPeng, 2024-02-29:
无线网卡可识别了，但是一连就断
[image]

Thu Feb 29 20:21:31 2024 daemon.notice hostapd: Remove interface 'wlan0'
Thu Feb 29 20:21:31 2024 daemon.notice hostapd: wlan0: interface state ENABLED->DISABLED
Thu Feb 29 20:21:31 2024 daemon.notice hostapd: wlan0: AP-DISABLED
Thu Feb 29 20:21:31 2024 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Thu Feb 29 20:21:31 2024 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Thu Feb 29 20:21:31 2024 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Thu Feb 29 20:21:31 2024 kern.info kernel: [ 5883.995067] device wlan0 left promiscuous mode
Thu Feb 29 20:21:31 2024 kern.info kernel: [ 5883.995103] br-lan: port 2(wlan0) entered disabled state
Thu Feb 29 20:21:31 2024 daemon.notice netifd: Network device 'wlan0' link is down
Thu Feb 29 20:21:31 2024 daemon.notice ttyd[11295]: rops_handle_POLLIN_netlink: DELADDR
Thu Feb 29 20:21:31 2024 daemon.info avahi-daemon[7896]: Interface wlan0.IPv6 no longer relevant for mDNS.
Thu Feb 29 20:21:31 2024 daemon.info avahi-daemon[7896]: Leaving mDNS multicast group on interface wlan0.IPv6 with address fe80::d61b:81ff:fe66:9521.
Thu Feb 29 20:21:31 2024 daemon.info avahi-daemon[7896]: Withdrawing address record for fe80::d61b:81ff:fe66:9521 on wlan0.
Thu Feb 29 20:21:31 2024 daemon.debug ModemManager[9243]: hotplug: event reported: action=remove, name=wlan0, subsystem=net
Thu Feb 29 20:21:31 2024 daemon.notice netifd: Wireless device 'radio0' is now down
Thu Feb 29 20:21:32 2024 user.notice wan_drop: current proto
Thu Feb 29 20:21:33 2024 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0) --> new PHY
Thu Feb 29 20:21:33 2024 daemon.debug ModemManager[9474]: hotplug: event reported: action=add, name=wlan0, subsystem=net
Thu Feb 29 20:21:33 2024 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Thu Feb 29 20:21:33 2024 daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->HT_SCAN
Thu Feb 29 20:21:33 2024 kern.info kernel: [ 5885.516874] br-lan: port 2(wlan0) entered blocking state
Thu Feb 29 20:21:33 2024 kern.info kernel: [ 5885.516876] br-lan: port 2(wlan0) entered disabled state
Thu Feb 29 20:21:33 2024 kern.info kernel: [ 5885.516928] device wlan0 entered promiscuous mode
Thu Feb 29 20:21:33 2024 kern.info kernel: [ 5885.950456] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Thu Feb 29 20:21:33 2024 kern.info kernel: [ 5885.950518] br-lan: port 2(wlan0) entered blocking state
Thu Feb 29 20:21:33 2024 kern.info kernel: [ 5885.950520] br-lan: port 2(wlan0) entered forwarding state
Thu Feb 29 20:21:33 2024 daemon.notice hostapd: wlan0: interface state HT_SCAN->ENABLED
Thu Feb 29 20:21:33 2024 daemon.notice hostapd: wlan0: AP-ENABLED
Thu Feb 29 20:21:34 2024 daemon.notice netifd: Wireless device 'radio0' is now up
Thu Feb 29 20:21:34 2024 daemon.notice netifd: Network device 'wlan0' link is up
Thu Feb 29 20:21:35 2024 daemon.info avahi-daemon[7896]: Joining mDNS multicast group on interface wlan0.IPv6 with address fe80::d61b:81ff:fe66:9521.
Thu Feb 29 20:21:35 2024 daemon.info avahi-daemon[7896]: New relevant interface wlan0.IPv6 for mDNS.
Thu Feb 29 20:21:35 2024 daemon.info avahi-daemon[7896]: Registering new address record for fe80::d61b:81ff:fe66:9521 on wlan0.*.
Thu Feb 29 20:21:36 2024 daemon.info [7997]: <info>  [base-manager] couldn't check support for device '/sys/devices/pci0000:00/0000:00:10.0': not supported by any plugin
Thu Feb 29 20:21:38 2024 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Thu Feb 29 20:21:38 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 2 names
Thu Feb 29 20:21:45 2024 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED 9a:06:47:be:8c:69 2
Thu Feb 29 20:21:45 2024 daemon.info hostapd: wlan0: STA 9a:06:47:be:8c:69 IEEE 802.11: associated (aid 1)
Thu Feb 29 20:21:45 2024 kern.info kernel: [ 5898.059179] rtw89_8852ae 0000:00:10.0: timed out to flush queues
Thu Feb 29 20:21:45 2024 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 9a:06:47:be:8c:69
Thu Feb 29 20:21:45 2024 daemon.info hostapd: wlan0: STA 9a:06:47:be:8c:69 RADIUS: starting accounting session 5001D8A5ADCF4D3F
Thu Feb 29 20:21:45 2024 daemon.info hostapd: wlan0: STA 9a:06:47:be:8c:69 WPA: pairwise key handshake completed (RSN)
Thu Feb 29 20:21:45 2024 daemon.notice hostapd: wlan0: EAPOL-4WAY-HS-COMPLETED 9a:06:47:be:8c:69
Thu Feb 29 20:22:03 2024 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 9a:06:47:be:8c:69
Thu Feb 29 20:22:04 2024 kern.info kernel: [ 5916.509053] rtw89_8852ae 0000:00:10.0: timed out to flush queues
Thu Feb 29 20:22:04 2024 kern.info kernel: [ 5916.799041] rtw89_8852ae 0000:00:10.0: timed out to flush queues
Thu Feb 29 20:22:13 2024 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED 9a:06:47:be:8c:69 2
Thu Feb 29 20:22:13 2024 daemon.info hostapd: wlan0: STA 9a:06:47:be:8c:69 IEEE 802.11: associated (aid 1)
Thu Feb 29 20:22:13 2024 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 9a:06:47:be:8c:69
Thu Feb 29 20:22:13 2024 daemon.info hostapd: wlan0: STA 9a:06:47:be:8c:69 RADIUS: starting accounting session 5D9FB1E6AA845066
Thu Feb 29 20:22:13 2024 daemon.info hostapd: wlan0: STA 9a:06:47:be:8c:69 WPA: pairwise key handshake completed (RSN)
Thu Feb 29 20:22:13 2024 daemon.notice hostapd: wlan0: EAPOL-4WAY-HS-COMPLETED 9a:06:47:be:8c:69
Thu Feb 29 20:22:31 2024 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 9a:06:47:be:8c:69
[... cut ...]
Thu Feb 29 20:22:55 2024 daemon.info hostapd: wlan0: STA 9a:06:47:be:8c:69 IEEE 802.11: associated (aid 1)
Thu Feb 29 20:22:55 2024 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 9a:06:47:be:8c:69
Thu Feb 29 20:22:55 2024 daemon.info hostapd: wlan0: STA 9a:06:47:be:8c:69 RADIUS: starting accounting session 763341981136591C
Thu Feb 29 20:22:55 2024 daemon.info hostapd: wlan0: STA 9a:06:47:be:8c:69 WPA: pairwise key handshake completed (RSN)
Thu Feb 29 20:22:55 2024 daemon.notice hostapd: wlan0: EAPOL-4WAY-HS-COMPLETED 9a:06:47:be:8c:69

jjm2473, 2024-02-29:
试试选择AC 80Mhz

jjm2473, 2024-02-29:
还有选择国家为CN

NokiPeng, 2024-02-29:
还是一样
[image]

Thu Feb 29 20:45:05 2024 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 9a:06:47:be:8c:69
Thu Feb 29 20:45:05 2024 daemon.warn odhcp6c[24579]: Server returned IA_PD status 'No Prefix Available (prefix not avaliable)'
Thu Feb 29 20:45:05 2024 kern.info kernel: [ 7298.136938] rtw89_8852ae 0000:00:10.0: timed out to flush queues
Thu Feb 29 20:45:06 2024 kern.info kernel: [ 7298.436923] rtw89_8852ae 0000:00:10.0: timed out to flush queues
Thu Feb 29 20:45:06 2024 user.notice firewall: Reloading firewall due to ifupdate of WAN6 (eth1)
Thu Feb 29 20:45:07 2024 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Thu Feb 29 20:45:07 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 2 names
Thu Feb 29 20:45:08 2024 daemon.warn odhcp6c[24579]: Server returned IA_PD status 'No Prefix Available (prefix not avaliable)'
Thu Feb 29 20:45:09 2024 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED 9a:06:47:be:8c:69 2
Thu Feb 29 20:45:09 2024 daemon.info hostapd: wlan0: STA 9a:06:47:be:8c:69 IEEE 802.11: associated (aid 1)
Thu Feb 29 20:45:09 2024 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 9a:06:47:be:8c:69
Thu Feb 29 20:45:09 2024 daemon.info hostapd: wlan0: STA 9a:06:47:be:8c:69 RADIUS: starting accounting session 834EFE58167C40DE
Thu Feb 29 20:45:09 2024 daemon.info hostapd: wlan0: STA 9a:06:47:be:8c:69 WPA: pairwise key handshake completed (RSN)
Thu Feb 29 20:45:09 2024 daemon.notice hostapd: wlan0: EAPOL-4WAY-HS-COMPLETED 9a:06:47:be:8c:69
Thu Feb 29 20:45:10 2024 daemon.warn odhcp6c[24579]: Server returned IA_PD status 'No Prefix Available (prefix not avaliable)'
Thu Feb 29 20:45:12 2024 daemon.warn odhcp6c[24579]: Server returned IA_PD status 'No Prefix Available (prefix not avaliable)'
Thu Feb 29 20:45:15 2024 daemon.warn odhcp6c[24579]: Server returned IA_PD status 'No Prefix Available (prefix not avaliable)'
Thu Feb 29 20:45:17 2024 daemon.warn odhcp6c[24579]: Server returned IA_PD status 'No Prefix Available (prefix not avaliable)'
Thu Feb 29 20:45:19 2024 daemon.warn odhcp6c[24579]: Server returned IA_PD status 'No Prefix Available (prefix not avaliable)'
Thu Feb 29 20:45:20 2024 daemon.warn odhcp6c[24579]: Server returned IA_PD status 'No Prefix Available (prefix not avaliable)'
Thu Feb 29 20:45:22 2024 daemon.warn odhcp6c[24579]: Server returned IA_PD status 'No Prefix Available (prefix not avaliable)'
Thu Feb 29 20:45:24 2024 daemon.warn odhcp6c[24579]: Server returned IA_PD status 'No Prefix Available (prefix not avaliable)'
Thu Feb 29 20:45:25 2024 daemon.warn odhcp6c[24579]: Server returned IA_PD status 'No Prefix Available (prefix not avaliable)'
Thu Feb 29 20:45:27 2024 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 9a:06:47:be:8c:69
Thu Feb 29 20:45:27 2024 kern.info kernel: [ 7320.166666] rtw89_8852ae 0000:00:10.0: timed out to flush queues
Thu Feb 29 20:45:27 2024 daemon.warn odhcp6c[24579]: Server returned IA_PD status 'No Prefix Available (prefix not avaliable)'
Thu Feb 29 20:45:28 2024 kern.info kernel: [ 7320.466661] rtw89_8852ae 0000:00:10.0: timed out to flush queues
Thu Feb 29 20:45:30 2024 daemon.warn odhcp6c[24579]: Server returned IA_PD status 'No Prefix Available (prefix not avaliable)'
Thu Feb 29 20:45:32 2024 daemon.warn odhcp6c[24579]: Server returned IA_PD status 'No Prefix Available (prefix not avaliable)'
Thu Feb 29 20:45:34 2024 daemon.warn odhcp6c[24579]: Server returned IA_PD status 'No Prefix Available (prefix not avaliable)'
Thu Feb 29 20:45:37 2024 daemon.warn odhcp6c[24579]: Server returned IA_PD status 'No Prefix Available (prefix not avaliable)'
Thu Feb 29 20:45:38 2024 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED 9a:06:47:be:8c:69 2
Thu Feb 29 20:45:38 2024 daemon.info hostapd: wlan0: STA 9a:06:47:be:8c:69 IEEE 802.11: associated (aid 1)
Thu Feb 29 20:45:38 2024 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 9a:06:47:be:8c:69
Thu Feb 29 20:45:38 2024 daemon.info hostapd: wlan0: STA 9a:06:47:be:8c:69 RADIUS: starting accounting session 56E5011FA9DFB4E7
Thu Feb 29 20:45:38 2024 daemon.info hostapd: wlan0: STA 9a:06:47:be:8c:69 WPA: pairwise key handshake completed (RSN)
Thu Feb 29 20:45:38 2024 daemon.notice hostapd: wlan0: EAPOL-4WAY-HS-COMPLETED 9a:06:47:be:8c:69
Thu Feb 29 20:45:40 2024 daemon.warn odhcp6c[24579]: Server returned IA_PD status 'No Prefix Available (prefix not avaliable)'
Thu Feb 29 20:45:42 2024 daemon.warn odhcp6c[24579]: Server returned IA_PD status 'No Prefix Available (prefix not avaliable)'
Thu Feb 29 20:45:44 2024 daemon.warn odhcp6c[24579]: Server returned IA_PD status 'No Prefix Available (prefix not avaliable)'
Thu Feb 29 20:45:46 2024 daemon.warn odhcp6c[24579]: Server returned IA_PD status 'No Prefix Available (prefix not avaliable)'
Thu Feb 29 20:45:49 2024 daemon.warn odhcp6c[24579]: Server returned IA_PD status 'No Prefix Available (prefix not avaliable)'
Thu Feb 29 20:45:50 2024 daemon.warn odhcp6c[24579]: Server returned IA_PD status 'No Prefix Available (prefix not avaliable)'
Thu Feb 29 20:45:53 2024 daemon.warn odhcp6c[24579]: Server returned IA_PD status 'No Prefix Available (prefix not avaliable)'
Thu Feb 29 20:45:56 2024 daemon.warn odhcp6c[24579]: Server returned IA_PD status 'No Prefix Available (prefix not avaliable)'
Thu Feb 29 20:45:56 2024 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 9a:06:47:be:8c:69
Thu Feb 29 20:45:57 2024 kern.info kernel: [ 7349.596323] rtw89_8852ae 0000:00:10.0: timed out to flush queues
[... cut ...]
Thu Feb 29 20:47:04 2024 daemon.warn odhcp6c[24579]: Server returned IA_PD status 'No Prefix Available (prefix not avaliable)'
Thu Feb 29 20:47:05 2024 daemon.warn odhcp6c[24579]: Server returned IA_PD status 'No Prefix Available (prefix not avaliable)'
Thu Feb 29 20:47:08 2024 daemon.warn odhcp6c[24579]: Server returned IA_PD status 'No Prefix Available (prefix not avaliable)'
Thu Feb 29 20:47:10 2024 daemon.warn odhcp6c[24579]: Server returned IA_PD status 'No Prefix Available (prefix not avaliable)'

jjm2473, 2024-02-29:
试试执行这个：
```shell
echo 'rtw89pci disable_aspm_l1=y disable_aspm_l1ss=y disable_clkreq=y' >>/etc/modules.d/rtw89-oot
reboot
```

NokiPeng, 2024-02-29:
添加了，还是一样
[image]

jjm2473, 2024-03-01:
那么再试试这个：
```shell
echo 'rtw89core disable_ps_mode=y' >>/etc/modules.d/rtw89-oot
reboot
```

根据官方的issue https://github.com/lwfinger/rtw89/issues/8

NokiPeng, 2024-03-01:
改完之后连无线网卡都驱动不了，连频道都少了
[image]
[image]

jjm2473, 2024-03-01:
这不是频道少了，而是因为国家码没配置成功，显示的是其他国家的频道列表。你试试AC模式情况下选择WAP2加密

NokiPeng, 2024-03-01:
原来是我重新刷个系统，未启用无线网络，所以无线网卡不工作，现在可以正常工作了，未掉线，主要就是两行代码

echo 'rtw89pci disable_aspm_l1=y disable_aspm_l1ss=y disable_clkreq=y' >>/etc/modules.d/rtw89-oot
echo 'rtw89core disable_ps_mode=y' >>/etc/modules.d/rtw89-oot

还有就是AP的设置
[image]

基本可以拉满宽带
[image]
[image]

jjm2473, 2024-03-01:
也就是说，你的机器加上 
```
echo 'rtw89pci disable_aspm_l1=y disable_aspm_l1ss=y disable_clkreq=y' >>/etc/modules.d/rtw89-oot
echo 'rtw89core disable_ps_mode=y' >>/etc/modules.d/rtw89-oot
```
就能正常使用，

那么试试不要第二行，也就是删掉 'rtw89core disable_ps_mode=y' ，以及试试第一行改成 'rtw89pci disable_clkreq=y'，看看哪种能正常工作，参数越少越好。

NokiPeng, 2024-03-01:
改好了，可以正常运行:beers:
注释了前两行
[image]

jjm2473, 2024-03-01:
OK，那之后看看其他人的反馈，是不是要把这个参数作为默认。
 
 
目前，为了避免以后固件升级导致 `/etc/modules.d/rtw89-oot` 参数被重置，执行这个命令：
```shell
echo 'options rtw89pci disable_clkreq=y' >> /etc/modules.conf
rm -f /overlay/upper/etc/modules.d/rtw89-oot
reboot
```

awenzh, 2024-03-01:
我也有8852be网卡，现在是只要修改了参数就能用吗？

jjm2473, 2024-03-01:
@awenzh 升级最新固件，可能不需要改参数都能用，那个参数应该只是为了讨好某些BIOS的

yuebai998, 2024-03-04:
8852be，我用的最新固件，开客户端模式，一直掉线重连。附日志：[    0.012602] ACPI: Reserving SSDT table memory at [mem 0x73128000-0x7312b3d2]
[    0.012602] ACPI: Reserving SSDT table memory at [mem 0x7311a000-0x73127486]
[    0.012603] ACPI: Reserving HPET table memory at [mem 0x731ad000-0x731ad037]
[    0.012604] ACPI: Reserving APIC table memory at [mem 0x73119000-0x731191db]
[    0.012605] ACPI: Reserving MCFG table memory at [mem 0x73118000-0x7311803b]
[    0.012605] ACPI: Reserving SSDT table memory at [mem 0x73116000-0x73117f19]
[    0.012606] ACPI: Reserving UEFI table memory at [mem 0x73201000-0x73201047]
[    0.012607] ACPI: Reserving NHLT table memory at [mem 0x73115000-0x7311502c]
[    0.012608] ACPI: Reserving LPIT table memory at [mem 0x73114000-0x731140cb]
[    0.012608] ACPI: Reserving SSDT table memory at [mem 0x73110000-0x73112a82]
[    0.012609] ACPI: Reserving SSDT table memory at [mem 0x7310d000-0x7310f356]
[    0.012610] ACPI: Reserving DBGP table memory at [mem 0x7310c000-0x7310c033]
[    0.012611] ACPI: Reserving DBG2 table memory at [mem 0x7310b000-0x7310b053]
[    0.012611] ACPI: Reserving SSDT table memory at [mem 0x73109000-0x7310a918]
[    0.012612] ACPI: Reserving DMAR table memory at [mem 0x73108000-0x73108087]
[    0.012613] ACPI: Reserving FPDT table memory at [mem 0x73107000-0x73107043]
[    0.012614] ACPI: Reserving SSDT table memory at [mem 0x73106000-0x73106739]
[    0.012614] ACPI: Reserving SSDT table memory at [mem 0x73102000-0x73105ae9]
[    0.012615] ACPI: Reserving SSDT table memory at [mem 0x730fe000-0x731019d9]
[    0.012616] ACPI: Reserving SSDT table memory at [mem 0x730fd000-0x730fd143]
[    0.012617] ACPI: Reserving PHAT table memory at [mem 0x730fc000-0x730fc5f0]
[    0.012617] ACPI: Reserving TPM2 table memory at [mem 0x730fb000-0x730fb04b]
[    0.012618] ACPI: Reserving WSMT table memory at [mem 0x73113000-0x73113027]
[    0.012632] ACPI: Local APIC address 0xfee00000
[    0.012643] Zone ranges:
[    0.012644]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
[    0.012645]   DMA32    [mem 0x0000000001000000-0x00000000ffffffff]
[    0.012646]   Normal   [mem 0x0000000100000000-0x000000027fbfffff]
[    0.012648] Movable zone start for each node
[    0.012648] Early memory node ranges
[    0.012649]   node   0: [mem 0x0000000000001000-0x000000000009dfff]
[    0.012650]   node   0: [mem 0x000000000009f000-0x000000000009ffff]
[    0.012651]   node   0: [mem 0x0000000000100000-0x000000004268efff]
[    0.012651]   node   0: [mem 0x0000000042690000-0x00000000430c9fff]
[    0.012652]   node   0: [mem 0x00000000430cb000-0x000000006ffccfff]
[    0.012653]   node   0: [mem 0x0000000073eff000-0x0000000073efffff]
[    0.012653]   node   0: [mem 0x0000000100000000-0x000000027fbfffff]
[    0.012655] Initmem setup node 0 [mem 0x0000000000001000-0x000000027fbfffff]
[    0.012656] On node 0 totalpages: 2030442
[    0.012657]   DMA zone: 64 pages used for memmap
[    0.012657]   DMA zone: 24 pages reserved
[    0.012658]   DMA zone: 3998 pages, LIFO batch:0
[    0.012658]   DMA32 zone: 7104 pages used for memmap
[    0.012659]   DMA32 zone: 454604 pages, LIFO batch:63
[    0.012659]   Normal zone: 24560 pages used for memmap
[    0.012660]   Normal zone: 1571840 pages, LIFO batch:63
[    0.012663] On node 0, zone DMA: 1 pages in unavailable ranges
[    0.012663] On node 0, zone DMA: 1 pages in unavailable ranges
[    0.012695] On node 0, zone DMA: 96 pages in unavailable ranges
[    0.014826] On node 0, zone DMA32: 1 pages in unavailable ranges
[    0.016198] On node 0, zone DMA32: 1 pages in unavailable ranges
[    0.016342] On node 0, zone DMA32: 16178 pages in unavailable ranges
[    0.027371] On node 0, zone Normal: 16640 pages in unavailable ranges
[    0.027383] On node 0, zone Normal: 1024 pages in unavailable ranges
[    0.027653] ACPI: PM-Timer IO Port: 0x1808
[    0.027655] ACPI: Local APIC address 0xfee00000
[    0.027660] ACPI: LAPIC_NMI (acpi_id[0x01] high edge lint[0x1])
[    0.027661] ACPI: LAPIC_NMI (acpi_id[0x02] high edge lint[0x1])
[    0.027662] ACPI: LAPIC_NMI (acpi_id[0x03] high edge lint[0x1])
[    0.027662] ACPI: LAPIC_NMI (acpi_id[0x04] high edge lint[0x1])
[    0.027663] ACPI: LAPIC_NMI (acpi_id[0x05] high edge lint[0x1])
[    0.027663] ACPI: LAPIC_NMI (acpi_id[0x06] high edge lint[0x1])
[    0.027664] ACPI: LAPIC_NMI (acpi_id[0x07] high edge lint[0x1])
[    0.027665] ACPI: LAPIC_NMI (acpi_id[0x08] high edge lint[0x1])
[    0.027665] ACPI: LAPIC_NMI (acpi_id[0x09] high edge lint[0x1])
[    0.027666] ACPI: LAPIC_NMI (acpi_id[0x0a] high edge lint[0x1])
[    0.027666] ACPI: LAPIC_NMI (acpi_id[0x0b] high edge lint[0x1])
[    0.027667] ACPI: LAPIC_NMI (acpi_id[0x0c] high edge lint[0x1])
[    0.027668] ACPI: LAPIC_NMI (acpi_id[0x0d] high edge lint[0x1])
[    0.027668] ACPI: LAPIC_NMI (acpi_id[0x0e] high edge lint[0x1])
[... 951 lines cut ...]
[  156.690510] device wlan0-1 entered promiscuous mode
[  156.690529] br-lan: port 2(wlan0-1) entered blocking state
[  156.690530] br-lan: port 2(wlan0-1) entered forwarding state
[  157.087161] br-lan: port 2(wlan0-1) entered disabled state
[  163.127833] device wlan0-1 left promiscuous mode
[  163.127926] br-lan: port 2(wlan0-1) entered disabled state
[  165.869217] wlan0: authenticate with 70:c6:dd:3b:2f:eb
[  166.055131] wlan0: send auth to 70:c6:dd:3b:2f:eb (try 1/3)
[  166.066550] wlan0: authenticated
[  166.067043] wlan0: associate with 70:c6:dd:3b:2f:eb (try 1/3)
[... cut ...]
[  176.337766] wlan0: deauthenticating from 70:c6:dd:3b:2f:eb by local choice (Reason: 3=DEAUTH_LEAVING)
[  176.958948] br-lan: port 2(wlan0-1) entered blocking state
[  176.958951] br-lan: port 2(wlan0-1) entered disabled state
[  176.958993] device wlan0-1 entered promiscuous mode
[  176.959010] br-lan: port 2(wlan0-1) entered blocking state
[  176.959011] br-lan: port 2(wlan0-1) entered forwarding state
[  177.356087] br-lan: port 2(wlan0-1) entered disabled state
[  183.457608] device wlan0-1 left promiscuous mode
[  183.457669] br-lan: port 2(wlan0-1) entered disabled state

jjm2473, 2024-03-04:
@yuebai998 看起来是hostapd挂了，应该不是驱动问题。试试修改加密方式为WPA2，修改管理帧有关的配置，没解决的话可能要等hostapd修复。

jjm2473, 2024-03-04:
@yuebai998 你同时启用了AP和客户端吗？

yuebai998, 2024-03-04:
是的，我第一次使用无线功能，不太熟悉。
我现在移除了AP功能，目前正常，短时间内没有断线重连情况出现了。

jjm2473, 2024-03-04:
@yuebai998 执行`iw list`看看，有些网卡不支持同时开启AP和客户端，不过也可能只是因为你的AP配置错误。

yuebai998, 2024-03-04:
root@iStoreOS:~# iw list
Wiphy phy0
        wiphy index: 0
        max # scan SSIDs: 8
        max scan IEs length: 394 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports AP-side u-APSD.
        Device supports T-DLS.
        Available Antennas: TX 0x3 RX 0x3
        Configured Antennas: TX 0x3 RX 0x3
        Supported interface modes:
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * P2P-client
                 * P2P-GO
        Band 1:
                Capabilities: 0x19ef
                        RX LDPC
                        HT20/HT40
                        SM Power Save disabled
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 7935 bytes
                        DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: No restriction (0x00)
                HT Max RX data rate: 300 Mbps
                HT TX/RX MCS rate indexes supported: 0-15, 32
                HE Iftypes: managed
                        HE MAC Capabilities (0x0801100a8060):
                                +HTC HE Supported
                                Trigger Frame MAC Padding Duration: 2
                                All Ack
                                BSR
                                Maximum A-MPDU Length Exponent: 2
                                OPS
                                A-MSDU in A-MPDU
                        HE PHY Capabilities: (0x02701e1f0dc0270e91bd00):
                                HE40/2.4GHz
                                Device Class: 1
                                LDPC Coding in Payload
                                HE SU PPDU with 1x HE-LTF and 0.8us GI
                                NDP with 4x HE-LTF and 3.2us GI
                                STBC Tx <= 80MHz
                                STBC Rx <= 80MHz
                                Doppler Tx
                                DCM Max Constellation: 3
                                DCM Max NSS Tx: 1
                                DCM Max Constellation Rx: 3
                                SU Beamformee
                                Beamformee STS <= 80Mhz: 3
                                Ng = 16 SU Feedback
                                Ng = 16 MU Feedback
                                Codebook Size SU Feedback
                                Codebook Size MU Feedback
                                Triggered SU Beamforming Feedback
                                Partial Bandwidth Extended Range
                                Power Boost Factor ar
                                HE SU PPDU & HE PPDU 4x HE-LTF 0.8us GI
                                Max NC: 1
                                HE ER SU PPDU 4x HE-LTF 0.8us GI
                                HE ER SU PPDU 1x HE-LTF 0.8us GI
[... 281 lines cut ...]
                        * 5885 MHz [177] (disabled)
        valid interface combinations:
                 * #{ managed } <= 1, #{ AP, P2P-client, P2P-GO } <= 1,
                   total <= 2, #channels <= 1
        HT Capability overrides:
                 * MCS: ff ff ff ff ff ff ff ff ff ff
                 * maximum A-MSDU length
                 * supported channel width
                 * short GI for 40 MHz
                 * max A-MPDU length exponent
                 * min MPDU start spacing
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Supported extended features:
                * [ RRM ]: RRM
                * [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
                * [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
                * [ CAN_REPLACE_PTK0 ]: can safely replace PTK 0 when rekeying
                * [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
                * [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
                * [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
root@iStoreOS:~# 
[image]

jjm2473, 2024-03-04:
@yuebai998 
```
        valid interface combinations:
                 * #{ managed } <= 1, #{ AP, P2P-client, P2P-GO } <= 1,
                   total <= 2, #channels <= 1
```

根据驱动的描述，是支持AP和客户端同时在线的，但是需要在同一个频道。一般要先作为客户端连接，再开AP，如果开AP失败，那就把无线的参数（频段，信道）改成跟客户端模式一致再试试。

yuebai998, 2024-03-04:
非常感谢您的帮助，这感染到了我。
我只是想用作客户端，却意外另外发现了一个问题，我很乐意为解决这个问题提供一些帮助。
我在确认客户端能正常使用的情况下，打开了AP开关，并配置参数与客户端一致，现在情况如图。reboot之后内核日志如下。
[image]

root@iStoreOS:~# dmesg
[    0.000000] Linux version 5.10.201 (build@buildserver) (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 2023070614) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP Thu Feb 29 10:13:27 2024
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=d55d2ed8-f283-61bd-8eb8-8f1bea502102 rootwait noinitrd
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009e000-0x000000000009efff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000009f000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000004268efff] usable
[    0.000000] BIOS-e820: [mem 0x000000004268f000-0x000000004268ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000042690000-0x00000000430c9fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000430ca000-0x00000000430cafff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000430cb000-0x000000006ffccfff] usable
[    0.000000] BIOS-e820: [mem 0x000000006ffcd000-0x00000000730ccfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000730cd000-0x00000000731aefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000731af000-0x000000007326efff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007326f000-0x0000000073e65fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000073e66000-0x0000000073efefff] type 20
[    0.000000] BIOS-e820: [mem 0x0000000073eff000-0x0000000073efffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000073f00000-0x0000000079ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007aa00000-0x000000007abfffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007b000000-0x00000000803fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000c0000000-0x00000000cfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed20000-0x00000000fed7ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000027fbfffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.80 by American Megatrends
[    0.000000] efi: ACPI=0x7324b000 ACPI 2.0=0x7324b014 TPMFinalLog=0x7321a000 SMBIOS=0x73bb0000 SMBIOS 3.0=0x73baf000 MEMATTR=0x6bcf0018 ESRT=0x6e365218 
[    0.000000] SMBIOS 3.6.0 present.
[    0.000000] DMI: TB WTR R1/WTR R1, BIOS HCN100-MI2 11/29/2023
[    0.000000] tsc: Detected 800.000 MHz processor
[    0.000000] tsc: Detected 806.400 MHz TSC
[    0.000215] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000217] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000227] last_pfn = 0x27fc00 max_arch_pfn = 0x400000000
[    0.000229] MTRR default type: write-back
[    0.000230] MTRR fixed ranges enabled:
[    0.000230]   00000-9FFFF write-back
[    0.000231]   A0000-BFFFF uncachable
[    0.000232]   C0000-FFFFF write-protect
[    0.000232] MTRR variable ranges enabled:
[    0.000233]   0 base 0080000000 mask 7F80000000 uncachable
[    0.000234]   1 base 007C000000 mask 7FFC000000 uncachable
[    0.000235]   2 base 2000000000 mask 6000000000 uncachable
[    0.000235]   3 base 1000000000 mask 7000000000 uncachable
[    0.000236]   4 base 0800000000 mask 7800000000 uncachable
[    0.000237]   5 base 0400000000 mask 7C00000000 uncachable
[    0.000237]   6 base 4000000000 mask 4000000000 uncachable
[    0.000238]   7 base 027FC00000 mask 7FFFC00000 uncachable
[    0.000238]   8 disabled
[    0.000239]   9 disabled
[    0.000335] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.000511] last_pfn = 0x73f00 max_arch_pfn = 0x400000000
[    0.011850] esrt: Reserving ESRT space from 0x000000006e365218 to 0x000000006e3652c8.
[    0.011855] e820: update [mem 0x6e365000-0x6e365fff] usable ==> reserved
[    0.011872] Using GB pages for direct mapping
[    0.011873] Incomplete global flushes, disabling PCID
[    0.012555] Secure boot disabled
[    0.012557] ACPI: Early table checksum verification disabled
[    0.012559] ACPI: RSDP 0x000000007324B014 000024 (v02 ALASKA)
[... 909 lines cut ...]
[   14.218048] auto rmmod i40e
[   14.258487] auto rmmod mpt3sas
[   14.261544] mpt3sas version 35.100.00.00 unloading
[   14.298038] auto rmmod ixgbe
[   15.052771] br-lan: port 2(wlan0-1) entered blocking state
[   15.052781] br-lan: port 2(wlan0-1) entered disabled state
[   15.052826] device wlan0-1 entered promiscuous mode
[   15.053378] device wlan0-1 left promiscuous mode
[   15.053397] br-lan: port 2(wlan0-1) entered disabled state
[   15.877557] igc 0000:02:00.0 eth1: NIC Link is Up 100 Mbps Full Duplex, Flow Control: RX/TX
[   15.877687] br-lan: port 1(eth1) entered blocking state
[   15.877692] br-lan: port 1(eth1) entered forwarding state
[   15.877869] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   19.947370] NFSD: Using /var/lib/nfs/v4recovery as the NFSv4 state recovery directory
[... cut ...]
[   23.561079] overlayfs: upperdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
[   23.561089] overlayfs: workdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
[   23.561094] overlayfs: lowerdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
[   23.561098] overlayfs: lowerdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
root@iStoreOS:~#

jjm2473, 2024-03-04:
看一下系统日志，hostapd是不是在执行DFS，也就是在扫描雷达频道，这时可能要等一分钟以后才会启动AP，选择149以上频道可以避免雷达扫描。

yuebai998, 2024-03-05:
关于您提到的一分钟的说法，我在36频道特意去观察了一下，仍旧无法工作。我也去试了下149及以上频道，可能是因为某些限制的缘故，在149频道上工作时频率到了6.5Ghz以上，实际上无法工作。我附近有设施，我不便调整国家码，遂作罢。
[image]

yuebai998, 2024-03-05:
这是还原设置到同时启用客户端和AP模式 信道36 ，开机两分钟左右的系统日志，不知道有没有能提供帮助的信息
root@iStoreOS:~# logread -l 300
Tue Mar  5 10:24:36 2024 daemon.debug ModemManager[8661]: hotplug: event reported: action=add, name=tty15, subsystem=tty
Tue Mar  5 10:24:36 2024 daemon.err ModemManager[8661]: hotplug: parent device sysfspath not found
Tue Mar  5 10:24:36 2024 daemon.debug ModemManager[8661]: hotplug: cached event found: action=add, name=tty43, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty43
Tue Mar  5 10:24:36 2024 daemon.debug ModemManager[8661]: hotplug: event reported: action=add, name=tty43, subsystem=tty
Tue Mar  5 10:24:36 2024 daemon.err ModemManager[8661]: hotplug: parent device sysfspath not found
Tue Mar  5 10:24:36 2024 daemon.debug ModemManager[8661]: hotplug: cached event found: action=add, name=ttyS3, subsystem=tty, sysfspath=/sys/devices/platform/serial8250/tty/ttyS3
Tue Mar  5 10:24:36 2024 daemon.debug ModemManager[8661]: hotplug: event reported: action=add, name=ttyS3, subsystem=tty
Tue Mar  5 10:24:36 2024 daemon.err ModemManager[8661]: hotplug: parent device sysfspath not found
Tue Mar  5 10:24:36 2024 daemon.debug ModemManager[8661]: hotplug: cached event found: action=add, name=tty33, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty33
Tue Mar  5 10:24:36 2024 daemon.debug ModemManager[8661]: hotplug: event reported: action=add, name=tty33, subsystem=tty
Tue Mar  5 10:24:36 2024 daemon.err ModemManager[8661]: hotplug: parent device sysfspath not found
Tue Mar  5 10:24:36 2024 daemon.debug ModemManager[8661]: hotplug: cached event found: action=add, name=tty61, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty61
Tue Mar  5 10:24:36 2024 daemon.debug ModemManager[8661]: hotplug: event reported: action=add, name=tty61, subsystem=tty
Tue Mar  5 10:24:36 2024 daemon.err ModemManager[8661]: hotplug: parent device sysfspath not found
Tue Mar  5 10:24:36 2024 daemon.debug ModemManager[8661]: hotplug: cached event found: action=add, name=tty23, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty23
Tue Mar  5 10:24:36 2024 daemon.debug ModemManager[8661]: hotplug: event reported: action=add, name=tty23, subsystem=tty
Tue Mar  5 10:24:36 2024 daemon.err ModemManager[8661]: hotplug: parent device sysfspath not found
Tue Mar  5 10:24:36 2024 daemon.debug ModemManager[8661]: hotplug: cached event found: action=add, name=ttyS10, subsystem=tty, sysfspath=/sys/devices/platform/serial8250/tty/ttyS10
Tue Mar  5 10:24:36 2024 daemon.debug ModemManager[8661]: hotplug: event reported: action=add, name=ttyS10, subsystem=tty
Tue Mar  5 10:24:36 2024 daemon.err ModemManager[8661]: hotplug: parent device sysfspath not found
Tue Mar  5 10:24:36 2024 daemon.debug ModemManager[8661]: hotplug: cached event found: action=add, name=tty7, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty7
Tue Mar  5 10:24:36 2024 daemon.debug ModemManager[8661]: hotplug: event reported: action=add, name=tty7, subsystem=tty
Tue Mar  5 10:24:36 2024 daemon.err ModemManager[8661]: hotplug: parent device sysfspath not found
Tue Mar  5 10:24:36 2024 daemon.debug ModemManager[8661]: hotplug: cached event found: action=add, name=tty51, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty51
Tue Mar  5 10:24:36 2024 daemon.debug ModemManager[8661]: hotplug: event reported: action=add, name=tty51, subsystem=tty
Tue Mar  5 10:24:36 2024 daemon.err ModemManager[8661]: hotplug: parent device sysfspath not found
Tue Mar  5 10:24:36 2024 daemon.debug ModemManager[8661]: hotplug: cached event found: action=add, name=tty13, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty13
Tue Mar  5 10:24:36 2024 daemon.debug ModemManager[8661]: hotplug: event reported: action=add, name=tty13, subsystem=tty
Tue Mar  5 10:24:36 2024 daemon.err ModemManager[8661]: hotplug: parent device sysfspath not found
Tue Mar  5 10:24:36 2024 daemon.debug ModemManager[8661]: hotplug: cached event found: action=add, name=tty41, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty41
Tue Mar  5 10:24:36 2024 daemon.debug ModemManager[8661]: hotplug: event reported: action=add, name=tty41, subsystem=tty
Tue Mar  5 10:24:36 2024 daemon.err ModemManager[8661]: hotplug: parent device sysfspath not found
Tue Mar  5 10:24:36 2024 daemon.debug ModemManager[8661]: hotplug: cached event found: action=add, name=ttyS1, subsystem=tty, sysfspath=/sys/devices/platform/serial8250/tty/ttyS1
Tue Mar  5 10:24:36 2024 daemon.debug ModemManager[8661]: hotplug: event reported: action=add, name=ttyS1, subsystem=tty
Tue Mar  5 10:24:36 2024 daemon.err ModemManager[8661]: hotplug: parent device sysfspath not found
Tue Mar  5 10:24:36 2024 daemon.debug ModemManager[8661]: hotplug: cached event found: action=add, name=tty31, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty31
Tue Mar  5 10:24:36 2024 daemon.debug ModemManager[8661]: hotplug: event reported: action=add, name=tty31, subsystem=tty
Tue Mar  5 10:24:36 2024 daemon.err ModemManager[8661]: hotplug: parent device sysfspath not found
Tue Mar  5 10:24:36 2024 daemon.debug ModemManager[8661]: hotplug: cached event found: action=add, name=tty21, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty21
Tue Mar  5 10:24:36 2024 daemon.debug ModemManager[8661]: hotplug: event reported: action=add, name=tty21, subsystem=tty
Tue Mar  5 10:24:36 2024 daemon.err ModemManager[8661]: hotplug: parent device sysfspath not found
[... cut ...]
Tue Mar  5 10:25:02 2024 daemon.err nmbd[12468]:   *****
Tue Mar  5 10:25:14 2024 authpriv.info dropbear[13802]: Child connection from 192.168.100.174:36053
Tue Mar  5 10:25:14 2024 authpriv.notice dropbear[13802]: Password auth succeeded for 'root' from 192.168.100.174:36053
Tue Mar  5 10:25:14 2024 authpriv.info dropbear[13812]: Child connection from 192.168.100.174:36054
Tue Mar  5 10:25:14 2024 authpriv.notice dropbear[13812]: Password auth succeeded for 'root' from 192.168.100.174:36054
root@iStoreOS:~# logread -l 300

jjm2473, 2024-03-05:
@yuebai998 日志完全没看到hostapd，可能被其他日志刷了，可以点一下wifi页面的重启按钮。

一般频道100以上就不用DFS了。

yuebai998, 2024-03-05:
好的 点击重启按钮后 日志如下
root@iStoreOS:~# logread
Tue Mar  5 10:24:36 2024 daemon.debug ModemManager[8661]: hotplug: cached event found: action=add, name=wlan0-1, subsystem=net, sysfspath=/sys/devices/pci0000:00/0000:00:1d.1/0000:03:00.0/net/wlan0-1
Tue Mar  5 10:24:36 2024 daemon.debug ModemManager[8661]: hotplug: event reported: action=add, name=wlan0-1, subsystem=net
Tue Mar  5 10:24:38 2024 daemon.info dnsmasq[1]: started, version 2.89 cachesize 1000
Tue Mar  5 10:24:38 2024 daemon.info dnsmasq[1]: DNS service limited to local subnets
Tue Mar  5 10:24:38 2024 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset no-nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Tue Mar  5 10:24:38 2024 daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Tue Mar  5 10:24:38 2024 daemon.info dnsmasq-dhcp[1]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Tue Mar  5 10:24:38 2024 daemon.info dnsmasq[1]: using only locally-known addresses for test
Tue Mar  5 10:24:38 2024 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Tue Mar  5 10:24:38 2024 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Tue Mar  5 10:24:38 2024 daemon.info dnsmasq[1]: using only locally-known addresses for local
Tue Mar  5 10:24:38 2024 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Tue Mar  5 10:24:38 2024 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Tue Mar  5 10:24:38 2024 daemon.info dnsmasq[1]: using only locally-known addresses for bss.pandora.xiaomi.com
Tue Mar  5 10:24:38 2024 daemon.info dnsmasq[1]: using only locally-known addresses for o2o.api.xiaomi.com
Tue Mar  5 10:24:38 2024 daemon.info dnsmasq[1]: using 13 more local addresses
Tue Mar  5 10:24:38 2024 daemon.warn dnsmasq[1]: no servers found in /tmp/resolv.conf.d/resolv.conf.auto, will retry
Tue Mar  5 10:24:38 2024 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Tue Mar  5 10:24:38 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 2 names
Tue Mar  5 10:24:38 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c.9503 - 0 names
Tue Mar  5 10:24:38 2024 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Tue Mar  5 10:24:38 2024 daemon.info [8690]: <info>  [base-manager] couldn't check support for device '/sys/devices/pci0000:00/0000:00:1d.1/0000:03:00.0': not supported by any plugin
Tue Mar  5 10:24:38 2024 daemon.err rmmod: module is not loaded
Tue Mar  5 10:24:38 2024 daemon.info smartd[12228]: smartd 7.2 2020-12-30 r5155 [x86_64-linux-5.10.201] (localbuild)
Tue Mar  5 10:24:38 2024 daemon.info smartd[12228]: Copyright (C) 2002-20, Bruce Allen, Christian Franke, www.smartmontools.org
Tue Mar  5 10:24:38 2024 daemon.info smartd[12228]: Opened configuration file /etc/smartd.conf
Tue Mar  5 10:24:38 2024 daemon.crit smartd[12228]: File /etc/smartd.conf line 3 (drive DEFAULT): Missing argument to -n Directive
Tue Mar  5 10:24:38 2024 daemon.crit smartd[12228]: Valid arguments to -n Directive are:
Tue Mar  5 10:24:38 2024 daemon.crit smartd[12228]: never[,N][,q], sleep[,N][,q], standby[,N][,q], idle[,N][,q]
Tue Mar  5 10:24:38 2024 daemon.crit smartd[12228]: Configuration file /etc/smartd.conf has fatal syntax errors.
Tue Mar  5 10:24:38 2024 daemon.info smartd[12228]: Monitoring 0 ATA/SATA, 0 SCSI/SAS and 0 NVMe devices
Tue Mar  5 10:24:38 2024 user.notice adb-enablemodem: INFO: waiting for device
Tue Mar  5 10:24:38 2024 daemon.notice procd: /etc/rc.d/S99dockerd: Warning: Section @zone[1] (wan) cannot resolve device of network 'wwan'
Tue Mar  5 10:24:38 2024 daemon.notice procd: /etc/rc.d/S99dockerd: Warning: Section @rule[11] (allow_wan_ssh) does not specify a protocol, assuming TCP+UDP
Tue Mar  5 10:24:38 2024 daemon.notice procd: /etc/rc.d/S99dockerd:  * Clearing IPv4 filter table
Tue Mar  5 10:24:38 2024 daemon.notice procd: /etc/rc.d/S99dockerd:  * Clearing IPv4 nat table
Tue Mar  5 10:24:39 2024 daemon.notice procd: /etc/rc.d/S99dockerd:  * Clearing IPv4 mangle table
Tue Mar  5 10:24:39 2024 daemon.notice procd: /etc/rc.d/S99dockerd:  * Clearing IPv4 raw table
Tue Mar  5 10:24:39 2024 daemon.notice procd: /etc/rc.d/S99dockerd:  * Populating IPv4 filter table
Tue Mar  5 10:24:39 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Rule 'Allow-DHCP-Renew'
Tue Mar  5 10:24:39 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Rule 'Allow-Ping'
Tue Mar  5 10:24:39 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Rule 'Allow-IGMP'
Tue Mar  5 10:24:39 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Rule 'Allow-IPSec-ESP'
Tue Mar  5 10:24:39 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Rule 'Allow-ISAKMP'
Tue Mar  5 10:24:39 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Rule 'linkease'
Tue Mar  5 10:24:39 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Rule 'allow_wan_ssh'
Tue Mar  5 10:24:39 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Forward 'lan' -> 'wan'
Tue Mar  5 10:24:39 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Forward 'docker' -> 'wan'
Tue Mar  5 10:24:39 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Forward 'docker' -> 'lan'
Tue Mar  5 10:24:39 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Forward 'lan' -> 'docker'
Tue Mar  5 10:24:39 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'lan'
[... cut ...]
Tue Mar  5 13:01:18 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 2 names
Tue Mar  5 13:01:18 2024 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 2 names
Tue Mar  5 13:01:18 2024 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Tue Mar  5 13:01:19 2024 daemon.info [8690]: <info>  [base-manager] couldn't check support for device '/sys/devices/pci0000:00/0000:00:1d.1/0000:03:00.0': not supported by any plugin
Tue Mar  5 13:01:46 2024 user.info : luci: accepted login on / for root from 192.168.100.174

yuebai998, 2024-03-05:
另外 在地区设置为中国的情况下，可选的频道如下
[image]
我这边不方便调整国家代码，频道这块故没法继续测试了。

yuebai998, 2024-03-05:
我去了解了一下DFS，然后单独开AP模式 149频道，能发射信号，能接入手机等客户端。然后我添加了客户端模式，配置都写149，但是现在客户端工作在36频道，工作正常，AP端不工作。详情附图。
[image]
大佬辛苦了。

yuebai998, 2024-03-05:
[image]

yuebai998, 2024-03-05:
[image]

yuebai998, 2024-03-05:
附该情况下系统日志
root@iStoreOS:~# logread -l 300
Tue Mar  5 13:49:32 2024 kern.info kernel: [   18.487238] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd:  * Clearing IPv6 filter table
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd:  * Clearing IPv6 nat table
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd:  * Clearing IPv6 mangle table
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd:  * Populating IPv6 filter table
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Rule 'Allow-DHCPv6'
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Rule 'Allow-MLD'
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Rule 'Allow-ICMPv6-Input'
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Rule 'Allow-ICMPv6-Forward'
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Rule 'Allow-IPSec-ESP'
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Rule 'Allow-ISAKMP'
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Rule 'linkease'
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Forward 'lan' -> 'wan'
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Forward 'docker' -> 'wan'
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Forward 'docker' -> 'lan'
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Forward 'lan' -> 'docker'
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'lan'
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'wan'
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'docker'
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd:  * Populating IPv6 nat table
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd: Warning: fw3_ipt_rule_append(): Can't find target 'prerouting_lan_rule'
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd: Warning: fw3_ipt_rule_append(): Can't find target 'postrouting_lan_rule'
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd: Warning: fw3_ipt_rule_append(): Can't find target 'prerouting_wan_rule'
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd: Warning: fw3_ipt_rule_append(): Can't find target 'postrouting_wan_rule'
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd: Warning: fw3_ipt_rule_append(): Can't find target 'prerouting_docker_rule'
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd: Warning: fw3_ipt_rule_append(): Can't find target 'postrouting_docker_rule'
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd: Warning: fw3_ipt_rule_append(): Can't find target 'prerouting_rule'
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd: Warning: fw3_ipt_rule_append(): Can't find target 'postrouting_rule'
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'lan'
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'wan'
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'docker'
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd:  * Populating IPv6 mangle table
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'lan'
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'wan'
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'docker'
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd:  * Set tcp_ecn to off
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd:  * Set tcp_syncookies to on
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd:  * Set tcp_window_scaling to on
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd:  * Running script '/usr/share/miniupnpd/firewall.include'
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd:  * Running script '/var/etc/shadowsocksr.include'
Tue Mar  5 13:49:32 2024 daemon.notice procd: /etc/rc.d/S99dockerd:    ! Skipping due to path error: No such file or directory
Tue Mar  5 13:49:32 2024 daemon.info [8683]: <info>  [base-manager] couldn't check support for device '/sys/devices/pci0000:00/0000:00:1c.0/0000:01:00.0': not supported by any plugin
Tue Mar  5 13:49:32 2024 daemon.info [8683]: <info>  [base-manager] couldn't check support for device '/sys/devices/pci0000:00/0000:00:1d.0/0000:02:00.0': not supported by any plugin
Tue Mar  5 13:49:32 2024 daemon.info [8683]: <info>  [base-manager] couldn't check support for device '/sys/devices/pci0000:00/0000:00:1d.1/0000:03:00.0': not supported by any plugin
Tue Mar  5 13:49:33 2024 daemon.err nmbd[12585]: [2024/03/05 13:49:33.024974,  0] ../../source3/nmbd/nmbd.c:901(main)
Tue Mar  5 13:49:33 2024 daemon.err nmbd[12585]:   nmbd version 4.18.8 started.
Tue Mar  5 13:49:33 2024 daemon.err nmbd[12585]:   Copyright Andrew Tridgell and the Samba Team 1992-2023
Tue Mar  5 13:49:33 2024 daemon.err smbd[12583]: [2024/03/05 13:49:33.147815,  0] ../../source3/smbd/server.c:1746(main)
Tue Mar  5 13:49:33 2024 daemon.err smbd[12583]:   smbd version 4.18.8 started.
[... cut ...]
Tue Mar  5 13:52:04 2024 daemon.notice hostapd: handle_probe_req: send failed
Tue Mar  5 13:52:04 2024 daemon.notice hostapd: handle_probe_req: send failed
Tue Mar  5 13:52:06 2024 daemon.err hostapd: Wrong coupling between HT and VHT/HE channel setting
Tue Mar  5 13:52:08 2024 daemon.notice hostapd: handle_probe_req: send failed
Tue Mar  5 13:52:08 2024 daemon.notice hostapd: handle_probe_req: send failed
Tue Mar  5 13:52:12 2024 daemon.err hostapd: Wrong coupling between HT and VHT/HE channel setting
root@iStoreOS:~#

jjm2473, 2024-03-05:
看起来AP模式必须要在100频道以上，100以下DFS都会失败（可能驱动或硬件有bug）。

但是 @yuebai998 你的客户端模式又要用36频道，网卡不支持同时两个频道，所以你这肯定没法同时开AP和客户端了，除非你把客户端连的那个AP也改成100以上频道。

yuebai998, 2024-03-05:
嗯嗯 明白了，我去看了下恩山的帖子，有人对DFS做了测试。
88522BE折腾告一段落，我的MT7922到货了。嘿嘿
[image]
附链接https://www.right.com.cn/forum/thread-8307497-1-1.html
最后，感谢jjm2473大佬，大佬高产！

yuebai998, 2024-03-05:
[image]
还是8852BE，新的问题出现了，不知您是否感兴趣。在我重启、wan口接入互联网之后。有两个radio出现了，一个5G一个像2.4G！

jjm2473, 2024-03-05:
@yuebai998 可能你换了个模块？其中一个应该是无效的，只是之前的模块留下的配置，点一下扫描就知道了。

yuebai998, 2024-03-05:
emmm，我还没换模块呢，我就加了个m2，然后接了个网线到wan口。

yuebai998, 2024-03-05:
[log.txt](https://github.com/istoreos/istoreos/files/14492616/log.txt)
还是文本文件形式优雅点？ 开机的内核日志，点击新设备扫描及重启的系统日志。

yuebai998, 2024-03-05:
我在diskman里没找到我的m2固态硬盘，所以我多出来的网卡是硬盘？哈哈哈哈哈哈哈哈

jjm2473, 2024-03-08:
@yuebai998 出现两个radio的话，看看 `/etc/config/wireless` 的内容，可能设备路径变了或者mac地址变了，导致系统不认为两个模块是同一个

closed 2024-04-08 by jjm2473