istoreos/istoreos #1375: 2.5G usb网卡掉线问题

xmyaoshu, 2024-03-01:
x86架构
固件版本iStoreOS 22.03.6 2024022310
内核版本5.10.2012.5g
网卡
2.5G usb网卡 8156b在软件包中安装了8152，使用过程经常掉线。使用便携工具修复系统以后可以连接不一会儿又掉线。
不停的跳出以下错误。跪求大神教教，感激不尽。
Fri Mar  1 09:52:55 2024 kern.warn kernel: [  130.360507] r8152 3-2.2:1.0 eth1: failed tx_urb -22
Fri Mar  1 09:52:55 2024 kern.warn kernel: [  130.540658] xhci_hcd 0000:00:10.0: WARN waiting for error on ep to be cleared
Fri Mar  1 09:52:55 2024 kern.warn kernel: [  130.540677] r8152 3-2.2:1.0 eth1: failed tx_urb -22
Fri Mar  1 09:52:55 2024 kern.warn kernel: [  130.755472] xhci_hcd 0000:00:10.0: WARN waiting for error on ep to be cleared
Fri Mar  1 09:52:55 2024 kern.warn kernel: [  130.755490] r8152 3-2.2:1.0 eth1: failed tx_urb -22
Fri Mar  1 09:52:55 2024 kern.warn kernel: [  130.758310] xhci_hcd 0000:00:10.0: WARN waiting for error on ep to be cleared
Fri Mar  1 09:52:55 2024 kern.warn kernel: [  130.758326] r8152 3-2.2:1.0 eth1: failed tx_urb -22
Fri Mar  1 09:52:55 2024 kern.warn kernel: [  130.761852] xhci_hcd 0000:00:10.0: WARN waiting for error on ep to be cleared
Fri Mar  1 09:52:55 2024 kern.warn kernel: [  130.761870] r8152 3-2.2:1.0 eth1: failed tx_urb -22
Fri Mar  1 09:52:56 2024 kern.warn kernel: [  131.254556] xhci_hcd 0000:00:10.0: WARN waiting for error on ep to be cleared
Fri Mar  1 09:52:56 2024 kern.warn kernel: [  131.254574] r8152 3-2.2:1.0 eth1: failed tx_urb -22
Fri Mar  1 09:52:56 2024 kern.warn kernel: [  131.261892] xhci_hcd 0000:00:10.0: WARN waiting for error on ep to be cleared
Fri Mar  1 09:52:56 2024 kern.warn kernel: [  131.261910] r8152 3-2.2:1.0 eth1: failed tx_urb -22
Fri Mar  1 09:52:56 2024 kern.warn kernel: [  131.267074] xhci_hcd 0000:00:10.0: WARN waiting for error on ep to be cleared
Fri Mar  1 09:52:56 2024 kern.warn kernel: [  131.267092] r8152 3-2.2:1.0 eth1: failed tx_urb -22
Fri Mar  1 09:52:56 2024 kern.warn kernel: [  131.319920] xhci_hcd 0000:00:10.0: WARN waiting for error on ep to be cleared
Fri Mar  1 09:52:56 2024 kern.warn kernel: [  131.319939] r8152 3-2.2:1.0 eth1: failed tx_urb -22
Fri Mar  1 09:52:56 2024 kern.warn kernel: [  131.761618] xhci_hcd 0000:00:10.0: WARN waiting for error on ep to be cleared
Fri Mar  1 09:52:56 2024 kern.warn kernel: [  131.761635] r8152 3-2.2:1.0 eth1: failed tx_urb -22
Fri Mar  1 09:52:56 2024 kern.warn kernel: [  131.764640] xhci_hcd 0000:00:10.0: WARN waiting for error on ep to be cleared
Fri Mar  1 09:52:56 2024 kern.warn kernel: [  131.764655] r8152 3-2.2:1.0 eth1: failed tx_urb -22
Fri Mar  1 09:52:56 2024 kern.warn kernel: [  131.770498] xhci_hcd 0000:00:10.0: WARN waiting for error on ep to be cleared
Fri Mar  1 09:52:56 2024 kern.warn kernel: [  131.770514] r8152 3-2.2:1.0 eth1: failed tx_urb -22
Fri Mar  1 09:52:56 2024 kern.warn kernel: [  131.929867] xhci_hcd 0000:00:10.0: WARN waiting for error on ep to be cleared
Fri Mar  1 09:52:56 2024 kern.warn kernel: [  131.929888] r8152 3-2.2:1.0 eth1: failed tx_urb -22
Fri Mar  1 09:52:57 2024 kern.warn kernel: [  132.264988] xhci_hcd 0000:00:10.0: WARN waiting for error on ep to be cleared
Fri Mar  1 09:52:57 2024 kern.warn kernel: [  132.265006] r8152 3-2.2:1.0 eth1: failed tx_urb -22
Fri Mar  1 09:52:57 2024 kern.warn kernel: [  132.267857] xhci_hcd 0000:00:10.0: WARN waiting for error on ep to be cleared
Fri Mar  1 09:52:57 2024 kern.warn kernel: [  132.267872] r8152 3-2.2:1.0 eth1: failed tx_urb -22
Fri Mar  1 09:52:57 2024 kern.warn kernel: [  132.272563] xhci_hcd 0000:00:10.0: WARN waiting for error on ep to be cleared
Fri Mar  1 09:52:57 2024 kern.warn kernel: [  132.272582] r8152 3-2.2:1.0 eth1: failed tx_urb -22
Fri Mar  1 09:52:57 2024 kern.warn kernel: [  132.317188] xhci_hcd 0000:00:10.0: WARN waiting for error on ep to be cleared
Fri Mar  1 09:52:57 2024 kern.warn kernel: [  132.317206] r8152 3-2.2:1.0 eth1: failed tx_urb -22
Fri Mar  1 09:52:57 2024 kern.warn kernel: [  132.767929] xhci_hcd 0000:00:10.0: WARN waiting for error on ep to be cleared
Fri Mar  1 09:52:57 2024 kern.warn kernel: [  132.767944] r8152 3-2.2:1.0 eth1: failed tx_urb -22
Fri Mar  1 09:52:57 2024 kern.warn kernel: [  132.771183] xhci_hcd 0000:00:10.0: WARN waiting for error on ep to be cleared
Fri Mar  1 09:52:57 2024 kern.warn kernel: [  132.771196] r8152 3-2.2:1.0 eth1: failed tx_urb -22
Fri Mar  1 09:52:57 2024 kern.warn kernel: [  132.778297] xhci_hcd 0000:00:10.0: WARN waiting for error on ep to be cleared
Fri Mar  1 09:52:57 2024 kern.warn kernel: [  132.778312] r8152 3-2.2:1.0 eth1: failed tx_urb -22
Fri Mar  1 09:52:58 2024 kern.warn kernel: [  133.271638] xhci_hcd 0000:00:10.0: WARN waiting for error on ep to be cleared
Fri Mar  1 09:52:58 2024 kern.warn kernel: [  133.271653] r8152 3-2.2:1.0 eth1: failed tx_urb -22
Fri Mar  1 09:52:58 2024 kern.warn kernel: [  133.274609] xhci_hcd 0000:00:10.0: WARN waiting for error on ep to be cleared
Fri Mar  1 09:52:58 2024 kern.warn kernel: [  133.274623] r8152 3-2.2:1.0 eth1: failed tx_urb -22
Fri Mar  1 09:52:58 2024 kern.warn kernel: [  133.282255] xhci_hcd 0000:00:10.0: WARN waiting for error on ep to be cleared
Fri Mar  1 09:52:58 2024 kern.warn kernel: [  133.282265] r8152 3-2.2:1.0 eth1: failed tx_urb -22
Fri Mar  1 09:52:58 2024 kern.warn kernel: [  133.642817] xhci_hcd 0000:00:10.0: WARN waiting for error on ep to be cleared
Fri Mar  1 09:52:58 2024 kern.warn kernel: [  133.642835] r8152 3-2.2:1.0 eth1: failed tx_urb -22
Fri Mar  1 09:52:58 2024 kern.warn kernel: [  133.776218] xhci_hcd 0000:00:10.0: WARN waiting for error on ep to be cleared
Fri Mar  1 09:52:58 2024 kern.warn kernel: [  133.776237] r8152 3-2.2:1.0 eth1: failed tx_urb -22
Fri Mar  1 09:52:58 2024 kern.warn kernel: [  133.781329] xhci_hcd 0000:00:10.0: WARN waiting for error on ep to be cleared
Fri Mar  1 09:52:58 2024 kern.warn kernel: [  133.781340] r8152 3-2.2:1.0 eth1: failed tx_urb -22
Fri Mar  1 09:52:58 2024 kern.warn kernel: [  133.786494] xhci_hcd 0000:00:10.0: WARN waiting for error on ep to be cleared
Fri Mar  1 09:52:58 2024 kern.warn kernel: [  133.786512] r8152 3-2.2:1.0 eth1: failed tx_urb -22
Fri Mar  1 09:52:59 2024 kern.warn kernel: [  134.279257] xhci_hcd 0000:00:10.0: WARN waiting for error on ep to be cleared
Fri Mar  1 09:52:59 2024 kern.warn kernel: [  134.279272] r8152 3-2.2:1.0 eth1: failed tx_urb -22
Fri Mar  1 09:52:59 2024 kern.warn kernel: [  134.284634] xhci_hcd 0000:00:10.0: WARN waiting for error on ep to be cleared
Fri Mar  1 09:52:59 2024 kern.warn kernel: [  134.284686] r8152 3-2.2:1.0 eth1: failed tx_urb -22
Fri Mar  1 09:52:59 2024 kern.warn kernel: [  134.286837] xhci_hcd 0000:00:10.0: WARN waiting for error on ep to be cleared
Fri Mar  1 09:52:59 2024 kern.warn kernel: [  134.286855] r8152 3-2.2:1.0 eth1: failed tx_urb -22
Fri Mar  1 09:52:59 2024 kern.warn kernel: [  134.308582] xhci_hcd 0000:00:10.0: WARN waiting for error on ep to be cleared

jjm2473, 2024-03-01:
不要安装驱动，r8152本身就集成了，可能就是因为你安装的驱动才出问题

xmyaoshu, 2024-03-01:
> 不用安装驱动，r8152本身就集成了，可能就是因为你安装的驱动才出问题

就是在软件包里搜索5152安装的，如果不安装内网传输速度只有四百兆，也会掉网的情况。还有在杂项设置里开启8125和关闭，都试了都会出现掉网的情况。不知道啥情况，另一个千兆usb网卡就正常。经常就是显示已经连接，然后实际掉线，在电脑端断开重新启动就无法连接。
[image]

jjm2473, 2024-03-01:
反正不要安装软件包里的r8152驱动，还有r8125跟usb网卡无关，是给pcie网卡用的。固件集成的r8152是厂商驱动，一般都比内核的好，你从软件包里面安装的就是内核的。

xmyaoshu, 2024-03-01:
> 一般来说不要安装负载里的r8152驱动，还有r8125跟usb中断无关，是给pcie中断用的。固定集成的r8152是厂商驱动，一般都比内核的好，你从负载里面安装的就是内核的。

谢谢，已经移除了，就是内网跑不满上传跑400下载速度有个1.5g。

jjm2473, 2024-03-01:
不一定是软路由驱动有问题啊，可能对端设备有问题，或者usb接口问题，以及网线。x86 跑 RTL8156B 一般没啥问题，正常2.35G左右。

xmyaoshu, 2024-03-01:
> 不一定是软路由驱动有问题啊，可能对端设备有问题，或者usb接口问题，以及网线。x86跑RTL8156B一般没啥问题，正常2.35G左右。

当我安装那个驱动的时候是可以接近跑满的，移除后就不行，我尝试着更新我电脑端的驱动，还是一样，我明天其他电脑试试。

closed 2024-04-08 by jjm2473

chazikai24, 2025-01-29:
> 不一定是软路由驱动有问题啊，可能对端设备有问题，或者usb接口问题，以及网线。x86 跑 RTL8156B 一般没啥问题，正常2.35G左右。

22.03.7 2024122712版本也发现这样，上传测不满。准确说，应该是发往istoreos的流量，只有两三百兆。之前一直没留意，没发现这个问题。
如果启用杂项设置里的8125驱动，就没问题。
另外测试24.10RC4测试版，上传与下载都没问题。
我的是PCIE 8125B网卡。