istoreos/istoreos #1379: DHCPv6 NAT 方式，ipv6内网无法访问外网，防火墙缺少规则

138161, 2024-03-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：DHCPv6 NAT 方式，ipv6内网无法访问外网，防火墙缺少规则

### (2) 路由器型号和固件版本

A：

### (3) 详细日志和/或截图

A：防火墙需要添加以下内容
ip6tables -t nat -A POSTROUTING -o eth0 -j MASQUERADE

希望在以后的版本中解决

jjm2473, 2024-03-04:
ipv6 nat不是要自己配置吗，有谁用ipv6还nat

closed 2024-03-04 by jjm2473

jjm2473, 2024-03-04:
防火墙是支持ipv6 nat的，但是太少人用，所以界面都没有。用命令配置参考  https://openwrt.org/docs/guide-user/network/ipv6/ipv6.nat6

138161, 2024-03-18:
谢谢谢谢，wan和lan全中继情况下，win10电脑不定时会连不上ipv6，有时一天，有时几天，需要我在win10里定时ping -6 www.baidu.com或者wan口地址，被搞得烦死，所以就NAT了，稳定省心，反正istore下就几台电脑，没有杂七杂八的设备，其实只用ipv4就行了

jjm2473, 2024-03-18:
mac也有类似的问题，从抓包来看，有时候mac不会回应RS，导致路由器不知道ipv6地址对应的设备。目前还不知道是mac的bug还是openwrt的bug