istoreos/istoreos #1384: 首页的“已连接设备”数量为0

shixin-679, 2024-03-03:
设备及系统信息：
    设备型号：FriendlyElec NanoPi R5S
    固件版本：iStoreOS 22.03.6 2024030112
    内核版本：5.10.201
    已启动：0天0小时32分28秒
    系统时间：2024-03-04 02:25:38

问题描述：
    在首页的右上角“已连接设备”这个视图中，设备的数量一直为0；

    这个视图需要的“设备路由”和“带宽监控”都开起来了；

    这个是配置问题还是BUG呀？

[image]

[image]

jjm2473, 2024-03-04:
这个只统计lan口侧的arp数量，可能你的lan口没设备

closed 2024-03-04 by jjm2473

shixin-679, 2024-03-04:
> 这个只统计lan口侧的arp数量，可能你的lan口没有设备

有呀，eth1和eth2都由设备，eth1直连的电脑，eth2连的无线路由器  eth1和eth2是一个网段

jjm2473, 2024-03-05:
状态-路由表，看看arp表有没有lan的

shixin-679, 2024-03-05:
> 状态-路由表，看看arp表有没有lan的

[image]
有的

tonghm0808, 2024-07-06:
> 设备及系统信息： 设备型号：FriendlyElec NanoPi R5S 固件版本：iStoreOS 22.03.6 2024030112 内核版本：5.10.201 已启动：0天0小时32分28秒 系统时间：2024-03-04 02:25:38
>
> 问题描述： 在首页的右上角“已连接设备”这个视图中，设备的数量一直为0；
>
> ```
> 这个视图需要的“设备路由”和“带宽监控”都开起来了； [...]

应该是接口lan的名字 不对  这个首页的监控只监控接口名称为lan的接口下面的设备    自己创建的接口LAN  LAN1之类的名字都不行  简直无语

guanzhilian, 2024-11-09:
> 应该是接口lan的名字 不对 这个首页的监控只监控接口名称为lan的接口下面的设备 自己创建的接口LAN LAN1之类的名字都不行 简直无语

我也遇到同样的问题，创建lan1口后， 已连接设备就无法统计Lan1口设备。 lan1的设备占了整个网络主机数量的70%。看不到很郁闷。