istoreos/istoreos #1389: iStore拨号，隔段时间自动重拨

shixin-679, 2024-03-04:
设备信息：
    设备型号：FriendlyElec NanoPi R5S
    固件版本：iStoreOS 22.03.6 2024030112
    内核版本：5.10.201

宽带信息：
    成都电信100M

问题描述：
    在iStore拨号的情况下，会出现经常性的重拨，有时候重拨很快，几秒恢复联网，有时会出现重拨完全失败，根本播不上号；每次只要重拨号后，公网IP地址都会发生变动，在玩PUBG时就会出现“网络延迟检测”；

    我看了下，在重拨时，MAC地址没有自动发生变化，都是重拨前的MAC地址；

    我也问了运营商，装维师傅说可能是光猫盲插功能开启后，桥接的情况下会出现这种问题，关闭盲插，我的光猫LAN口就只有1口和4口，但是在盲插功能关闭后，还是会出现重拨。

    运营商二级反馈说没有针对桥接拨号进行任何限制，所以搞不懂了。

=================================================

    上周五出现了重拨一直失败的情况，就是一直播不了号，我也在WAN口抓了PPPOED协议包，下面是截图，是有iStore发出的PADI包，没有收到运营商的PADO包，日志显示也是“Timeout waiting for PADO packets”，符合现象。
[image]
[image]
[image]

=================================================

    3月4日晚上20:10:24又出现了重拨的情况，查了很多文档，有的说是MAC地址的问题，我修改了WAN口MAC后，也会出现隔段时间重拨，有时1天3-5次，有时2天共计几次，无规律地发生，以下是3月4日晚上20:10:24故障时的日志信息。

    WAN口配置截图：
[image]
[image]
[image]

    已安装的插件截图：
[image]

   最新的系统日志：
Mon Mar  4 11:45:29 2024 user.notice root: This is a Test Log!
Mon Mar  4 11:56:34 2024 authpriv.info dropbear[27154]: Exit (root) from <192.168.0.2:3637>: Error reading: Connection reset by peer
Mon Mar  4 12:07:01 2024 user.info : luci: accepted login on / for root from 240e:399:b7f1:9c20:e4f7:72e0:5687:c429
Mon Mar  4 12:08:47 2024 daemon.info netdata[24683]: Initializing file /var/cache/netdata/ipv4.fragsout/main.db.
Mon Mar  4 12:08:47 2024 daemon.info netdata[24683]: Initializing file /var/cache/netdata/ipv4.fragsout/FragOKs.db.
Mon Mar  4 12:08:47 2024 daemon.info netdata[24683]: Initializing file /var/cache/netdata/ipv4.fragsout/FragFails.db.
Mon Mar  4 12:08:47 2024 daemon.info netdata[24683]: Initializing file /var/cache/netdata/ipv4.fragsout/FragCreates.db.
Mon Mar  4 12:49:16 2024 daemon.err nlbwmon[15830]: Netlink receive failure: Out of memory
Mon Mar  4 12:49:16 2024 daemon.err nlbwmon[15830]: Unable to dump conntrack: No buffer space available
Mon Mar  4 13:36:51 2024 daemon.err nlbwmon[15830]: Netlink receive failure: Object busy
Mon Mar  4 13:36:51 2024 daemon.err nlbwmon[15830]: Unable to dump conntrack: I/O error
Mon Mar  4 13:51:03 2024 user.info : luci: accepted login on /admin/quickstart for root from 240e:399:b7f1:9c20:e4f7:72e0:5687:c429
Mon Mar  4 14:07:55 2024 daemon.err nlbwmon[15830]: Netlink receive failure: Object busy
Mon Mar  4 14:07:55 2024 daemon.err nlbwmon[15830]: Unable to dump conntrack: I/O error
Mon Mar  4 14:43:29 2024 daemon.err nlbwmon[15830]: Netlink receive failure: Out of memory
Mon Mar  4 14:43:29 2024 daemon.err nlbwmon[15830]: Unable to dump conntrack: No buffer space available
Mon Mar  4 17:21:47 2024 daemon.err nlbwmon[15830]: Netlink receive failure: Out of memory
Mon Mar  4 17:21:47 2024 daemon.err nlbwmon[15830]: Unable to dump conntrack: No buffer space available
Mon Mar  4 18:19:40 2024 user.info : luci: accepted login on / for root from 240e:399:b7f1:9c20:e4f7:72e0:5687:c429
Mon Mar  4 19:53:03 2024 daemon.err nlbwmon[15830]: Netlink receive failure: Out of memory
Mon Mar  4 19:53:03 2024 daemon.err nlbwmon[15830]: Unable to dump conntrack: No buffer space available
Mon Mar  4 20:10:24 2024 kern.info kernel: [65831.163212] rk_gmac-dwmac fe2a0000.ethernet eth0: Link is Down
Mon Mar  4 20:10:24 2024 daemon.notice netifd: Network device 'eth0' link is down
Mon Mar  4 20:10:24 2024 daemon.notice netifd: Interface 'wan' has link connectivity loss
Mon Mar  4 20:10:24 2024 daemon.notice netifd: Interface 'wan6' has link connectivity loss
Mon Mar  4 20:10:24 2024 daemon.info pppd[14617]: Terminating on signal 15
Mon Mar  4 20:10:24 2024 daemon.info pppd[14617]: Connect time 1096.2 minutes.
Mon Mar  4 20:10:24 2024 daemon.info pppd[14617]: Sent 2723159205 bytes, received 3194932688 bytes.
Mon Mar  4 20:10:24 2024 daemon.notice ttyd[14309]: rops_handle_POLLIN_netlink: DELADDR
Mon Mar  4 20:10:24 2024 daemon.info avahi-daemon[9569]: Withdrawing address record for 100.70.246.115 on pppoe-wan.
Mon Mar  4 20:10:24 2024 daemon.info avahi-daemon[9569]: Leaving mDNS multicast group on interface pppoe-wan.IPv4 with address 100.70.246.115.
Mon Mar  4 20:10:24 2024 daemon.info avahi-daemon[9569]: Interface pppoe-wan.IPv4 no longer relevant for mDNS.
Mon Mar  4 20:10:24 2024 daemon.notice ttyd[14309]: rops_handle_POLLIN_netlink: DELADDR
[... 135 lines cut ...]
Tue Mar  5 00:37:45 2024 kern.info kernel: [81872.978439] r8125: eth1: link down
Tue Mar  5 00:37:45 2024 daemon.notice netifd: Network device 'eth1' link is down
Tue Mar  5 00:37:45 2024 kern.info kernel: [81873.002076] br-lan: port 1(eth1) entered disabled state
Tue Mar  5 00:37:47 2024 kern.info kernel: [81875.099853] r8125: eth1: link up
Tue Mar  5 00:37:47 2024 kern.info kernel: [81875.100144] br-lan: port 1(eth1) entered blocking state
Tue Mar  5 00:37:47 2024 kern.info kernel: [81875.100201] br-lan: port 1(eth1) entered forwarding state
Tue Mar  5 00:37:47 2024 daemon.notice netifd: Network device 'eth1' link is up
Tue Mar  5 00:37:51 2024 kern.info kernel: [81879.505162] r8125: eth1: link down
Tue Mar  5 00:37:51 2024 kern.info kernel: [81879.528115] br-lan: port 1(eth1) entered disabled state
Tue Mar  5 00:37:51 2024 daemon.notice netifd: Network device 'eth1' link is down
Tue Mar  5 00:37:55 2024 daemon.notice netifd: Network device 'eth1' link is up
Tue Mar  5 00:37:55 2024 kern.info kernel: [81882.918918] r8125: eth1: link up
Tue Mar  5 00:37:55 2024 kern.info kernel: [81882.919381] br-lan: port 1(eth1) entered blocking state
Tue Mar  5 00:37:55 2024 kern.info kernel: [81882.919410] br-lan: port 1(eth1) entered forwarding state
Tue Mar  5 00:37:59 2024 authpriv.info dropbear[29778]: Exit (root) from <192.168.0.2:10609>: Error reading: Connection reset by peer
Tue Mar  5 01:10:20 2024 user.info : luci: accepted login on / for root from 240e:399:b7f1:bc00:a444:3a39:7792:cee3
Tue Mar  5 01:55:19 2024 daemon.err AdGuardHome[8829]: 2024/03/04 17:55:19.138161 [info] saving filter 1 contents to: /tmp/adguardhome/data/filters/1.txt
Tue Mar  5 01:55:19 2024 daemon.err AdGuardHome[8829]: 2024/03/04 17:55:19.138667 [info] updated filter 1: 1297334 bytes, 62938 rules
Tue Mar  5 01:55:20 2024 daemon.err AdGuardHome[8829]: 2024/03/04 17:55:20.739036 [info] Updated filter #1.  Rules: 62715 -> 62938
Tue Mar  5 02:13:12 2024 user.info : luci: accepted login on /admin/status/logs for root from 240e:399:b7f1:bc00:a444:3a39:7792:cee3
Tue Mar  5 02:28:34 2024 user.info : luci: accepted login on / for root from 192.168.0.2

=================================================

    请大佬帮忙看看是怎么回事，运营商那边也说是没有特殊限制，实在找不到原因了，以前我提过一个Issue，在那之后我重装了iStore，问题还是没解决，依然会出现重拨的情况。

shixin-679, 2024-03-04:
Issue中字符限制了，这是最新的内核日志：
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[    0.000000] Linux version 5.10.201 (build@buildserver) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 2023071019) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP Thu Feb 29 10:13:27 2024
[    0.000000] Machine model: FriendlyElec NanoPi R5S
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000fe660000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] OF: fdt: Reserved memory: failed to reserve memory for node 'drm-logo@00000000': base 0x0000000000000000, size 0 MiB
[    0.000000] OF: fdt: Reserved memory: failed to reserve memory for node 'drm-cubic-lut@00000000': base 0x0000000000000000, size 0 MiB
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000010000000, size 8 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x00000000efffffff]
[    0.000000]   node   0: [mem 0x00000001f0000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x00000001ffffffff]
[    0.000000] On node 0 totalpages: 1048064
[    0.000000]   DMA zone: 16376 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 982528 pages, LIFO batch:63
[    0.000000]   Normal zone: 16384 pages used for memmap
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 22 pages/cpu s49448 r8192 d32472 u90112
[    0.000000] pcpu-alloc: s49448 r8192 d32472 u90112 alloc=22*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1015304
[    0.000000] Kernel command line: earlycon=uart8250,mmio32,0xfe660000 console=ttyFIQ0 root=PARTUUID=5452574f-02 rw rootwait
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x00000000ec000000-0x00000000f0000000] (64MB)
[    0.000000] Memory: 4017068K/4192256K available (11902K kernel code, 1758K rwdata, 3676K rodata, 1984K init, 809K bss, 166996K reserved, 8192K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=4.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 30 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 320 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] GICv3: Distributor has no Range Selector support
[    0.000000] GICv3: 16 PPIs implemented
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000fd460000
[    0.000000] ITS [mem 0xfd440000-0xfd45ffff]
[    0.000000] ITS@0x00000000fd440000: allocated 8192 Devices @1810000 (indirect, esz 8, psz 64K, shr 0)
[    0.000000] ITS@0x00000000fd440000: allocated 32768 Interrupt Collections @1820000 (flat, esz 2, psz 64K, shr 0)
[    0.000000] ITS: using cache flushing for cmd queue
[    0.000000] GICv3: using LPI property table @0x0000000001830000
[    0.000000] GIC: using cache flushing for LPI property table
[    0.000000] GICv3: CPU0: using allocated LPI pending table @0x0000000001840000
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000006] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.001857] Console: colour dummy device 80x25
[    0.002335] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=80000)
[    0.003316] pid_max: default: 32768 minimum: 301
[    0.003909] Mount-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.004629] Mountpoint-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[... 1129 lines cut ...]
[30756.036963] br-lan: port 1(eth1) entered blocking state
[30756.036981] br-lan: port 1(eth1) entered forwarding state
[30760.335565] r8125: eth1: link down
[30760.361570] br-lan: port 1(eth1) entered disabled state
[30763.882774] r8125: eth1: link up
[30763.882990] br-lan: port 1(eth1) entered blocking state
[30763.883025] br-lan: port 1(eth1) entered forwarding state
[... cut ...]
[81872.978439] r8125: eth1: link down
[81873.002076] br-lan: port 1(eth1) entered disabled state
[81875.099853] r8125: eth1: link up
[81875.100144] br-lan: port 1(eth1) entered blocking state
[81875.100201] br-lan: port 1(eth1) entered forwarding state
[81879.505162] r8125: eth1: link down
[81879.528115] br-lan: port 1(eth1) entered disabled state
[81882.918918] r8125: eth1: link up
[81882.919381] br-lan: port 1(eth1) entered blocking state
[81882.919410] br-lan: port 1(eth1) entered forwarding state

shixin-679, 2024-03-04:
有个现象：如果重拨一直失败的话，重启光猫也解决不了，但是我重新插拔网线，拨号就成功了，我也换过光猫接口和成品网线，重拨现象也一直存在；还有就是有时候重拨失败，WAN口显示的是“已断开”或者是“1000MB的速率”，但是拨号成功后，因为我是100MB的宽带，WAN口速率就显示为100MB了。

jjm2473, 2024-03-08:
你的eth0是wan口吗？怎么只有百兆

jjm2473, 2024-03-08:
WAN口速率就显示为100MB肯定有问题，速率是链路协商出来的，跟运营商带宽无关，只跟硬件有关。换个网线试试。

shixin-679, 2024-03-08:
> 你的eth0是wan口吗？怎么只有百兆

ETH0是1G的，光猫也是1G的，但是我的宽带是100M的

shixin-679, 2024-03-08:
> WAN口速率就显示为100MB肯定有问题，速率是链路协商出来的，跟运营商带宽无关，只跟硬件有关。换个网线试试。

换过网线 网络正常时  协商出来都是100M

jjm2473, 2024-03-08:
反正协商速度跟带宽无关。还有问题找运营商，让他们带网线来测试。

azhe, 2024-03-11:
rpi4有同样的问题，尝试过无数版本。唯一没有问题的，是2023年2月的版本。

hy817120, 2024-03-18:
是要隔段几天要重新拨号一次, 不过容易拨号失败 不拨号
以前我也遇到过, 后来这样设置,就没出现过不拨号的情况
你可以试试
[image]
[image]

jjm2473, 2024-04-07:
跟IPv6应该没啥关系，LCP检查可能有用

loofeal, 2024-06-10:
问题解决了吗？我最近也是 这样的PPPOE 也掉线重播次数不正常。感觉比以前频繁了

babdoy, 2024-06-24:
> [shixin-679](/shixin-679)

有解决吗？我也遇到了，只要PPPOE调查就会疯狂重新拨号，造成运营商拨号次数上限，无法联网。

lvjinjie, 2025-11-05:
兄弟我也碰到了类似的问题，T_T   我换了路由器了，吃鸡最后一打一，一开枪瞬间就掉线，一晚上掉线5次以上