istoreos/istoreos #1392: 鸭子的ht2有适配计划嘛？

dalaoshi777, 2024-03-05:
鸭子的ht2有适配计划嘛？

leux0, 2024-03-06:
给你个参考，抛砖引玉：https://leux.cc/doc/iStoreOS%E9%80%82%E9%85%8DHT2.html

[image]

dalaoshi777, 2024-03-06:
> 给你个参考，抛砖引玉：https://leux.cc/doc/iStoreOS%E9%80%82%E9%85%8DHT2.html
> 
> ![2024-03-06_115318](https://private-user-images.githubusercontent.com/19219159/310355199-b850513f-65c9-480a-9e32-9dea2e03b933.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDk3MTE5MzUsIm5iZiI6MTcwOTcxMTYzNSwicGF0aCI6Ii8xOTIxOTE1OS8zMTAzNTUxOTktYjg1MDUxM2YtNjVjOS00ODBhLTllMzItOWRlYTJlMDNiOTMzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzA2VDA3NTM1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWEyMTU4OTFiNjk0ZDUzMWFiZDc4YTA2ZDZlMjQzZGViZGMyOTc1MjA0NTNlZDA5ZjFkNGJlMWU5MmFkMGM1YzQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.RnL2Ca-hTZ4Krum1gyFoFxomd_xhPZuEaiEo9pnoAJc)

可以研究下

jjm2473, 2024-03-07:
ht2不是单网口吗，做电视盒才最合适。

dalaoshi777, 2024-03-07:
> ht2不是单网口吗，做电视盒才最合适。

感觉挂个ssd当存储服务器也挺棒的😁

closed 2024-03-07 by dalaoshi777