istoreos/istoreos #1393: 占美N3150机器1000兆跑不满，RTL8111与intel 210混合网卡

PlutoNameless, 2024-03-05:
Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：占美N3150的双机器，一个网口是RTL8111，一个网口是Intel 210，公网speedtest和内网iperf3测试，内网速度只有700Mbps不到，外网速度300Mbps出头；
尝试过交换LAN和WAN网口，测试下来速度不变；
尝试过修改turboacc和防火墙-常规设置-软件流量分载，问题没有得到解决。

### (2) 路由器型号和固件版本

A：占美N3150，istoreos x86_64_efi istoreos-22.03.6-2024012613

### (3) 详细日志和/或截图

A：
[image]
[image]
[image]
[image]
[image]

PlutoNameless, 2024-03-06:
修正一些该问题的错误描述：
该机器为双8111E，而并非标题写的210与8111混合；
更多的信息在另外一个issues下贴出
https://github.com/istoreos/istoreos/issues/904?notification_referrer_id=NT_kwDOAFdeh7I3MTYxNzYzNjg1OjU3MjU4MzE#issuecomment-1980418503

jjm2473, 2024-03-07:
你还是测试下openwrt官方版本能跑多少，毕竟内网才700M也太慢了

jjm2473, 2024-03-07:
还有，iperf3测速的时候，客户端试试`-R`参数，看看服务端发送的速率以及丢包，或者你换个linux的客户端设备测速，这样才能看到双向的丢包率。

PlutoNameless, 2024-03-07:
> 还有，iperf3测速的时候，客户端试试`-R`参数，看看服务端发送的速率以及丢包，或者你换个linux的客户端设备测速，这样才能看到双向的丢包率。

turboacc是我称呼失误，本意是开启了「软件流量分载」，测试结果是开关「软件流量分载」并在每次测试前重启不会影响测试速度；

在网络-接口-全局网络设置里面把“数据包引导”勾选上，重启后，速度会有50Mbps的下降；

切换8168和8169驱动，速度上没有带来变化；

官方openwrt的版本我没有测试过，不过使用的第三方包当中，仅有Homelede 5.4.138内核下的版本测试速度是正常的，Homelde其它内核版本速度也异常；

我手上没有Linux设备，我用Mac上的iperf3输出的内容和Windows的字段是一样的；

以下截图是我在Windows下的测试；

-c 192.168.100.1 -V -u -b 1000M -P 10
[image]

-c 192.168.100.1 -V -u -b 1000M -P 10 -R
[image]

-c 192.168.100.1 -V -P 10
[image]

-c 192.168.100.1 -V -P 10 -R
[image]

这是我当前系统的内核日志
`[    0.000000] Linux version 5.10.201 (build@buildserver) (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 2023070614) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP Thu Feb 29 10:13:27 2024
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=0e191dfe-1c24-1a32-0441-44afa227da02 rootwait noinitrd
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000003efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000003f000-0x000000000003ffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000000040000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000003fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000040000000-0x00000000401fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000040200000-0x000000007ac44fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ac45000-0x000000007acbdfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007acbe000-0x000000007af22fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007af23000-0x000000007b877fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007b878000-0x000000007bb3bfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007bb3c000-0x000000007bb8efff] type 20
[    0.000000] BIOS-e820: [mem 0x000000007bb8f000-0x000000007bd56fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007bd57000-0x000000007bfc6fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007bfc7000-0x000000007bffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fea00000-0x00000000feafffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed03000-0x00000000fed03fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed06000-0x00000000fed06fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed08000-0x00000000fed09fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1cfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed80000-0x00000000fedbffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffb00000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.40 by American Megatrends
[    0.000000] efi: ESRT=0x7bb38218 ACPI=0x7b7f4000 ACPI 2.0=0x7b7f4000 SMBIOS=0xf05b0 
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: To be filled by O.E.M. To be filled by O.E.M./Aptio CRB, BIOS 5.11 08/22/2015
[    0.000000] tsc: Detected 1600.000 MHz processor
[    0.003205] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.003214] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.003238] last_pfn = 0x7c000 max_arch_pfn = 0x400000000
[    0.003245] MTRR default type: uncachable
[    0.003247] MTRR fixed ranges enabled:
[    0.003250]   00000-9FFFF write-back
[    0.003253]   A0000-BFFFF uncachable
[    0.003255]   C0000-FFFFF write-protect
[    0.003258] MTRR variable ranges enabled:
[... 823 lines cut ...]
[   44.150032] overlayfs: lowerdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
[   44.150067] overlayfs: lowerdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
[   63.974598] pppoe-wan: renamed from ppp0
[  160.654311] pppoe-wan: renamed from ppp0
[  191.164519] pppoe-wan: renamed from ppp0
[  308.354403] pppoe-wan: renamed from ppp0
[  334.852403] r8168: eth0: link down
[  345.256563] r8168: eth0: link up
[  352.522855] r8168: eth0: link down
[  410.776483] r8168: eth0: link up
[  421.094435] pppoe-wan: renamed from ppp0
[ 9917.730905] br-lan: port 1(eth1) entered disabled state
[ 9917.732536] device eth1 left promiscuous mode
[ 9917.732594] br-lan: port 1(eth1) entered disabled state
[ 9918.055693] eth1: 0xffffc9000007d000, 00:e0:4c:68:2a:f0, IRQ 120
[ 9918.120123] br-lan: port 1(eth1) entered blocking state
[ 9918.120131] br-lan: port 1(eth1) entered disabled state
[ 9918.120357] device eth1 entered promiscuous mode
[ 9921.244649] r8168: eth1: link up
[ 9921.245144] br-lan: port 1(eth1) entered blocking state
[ 9921.245159] br-lan: port 1(eth1) entered listening state
[ 9929.710253] br-lan: port 1(eth1) entered learning state
[ 9938.029713] br-lan: port 1(eth1) entered forwarding state
[ 9938.029739] br-lan: topology change detected, propagating
[ 9938.030188] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready`

jjm2473, 2024-03-08:
我用h28k测试了下，没有速度问题，稳定900多Mbps（未开启流量分载）。这是pci信息，pcie网卡也是RTL8111： 
[image]

理论上h28k的cpu性能应该比x86要低的（Arm64 四核 2GHz）。

jjm2473, 2024-03-08:
你试试不同版本的固件，20240301升级了r8168驱动。

PlutoNameless, 2024-03-08:
> 你试试不同版本的固件，20240301升级了r8168驱动。

此前我在使用istoreos-22.03.6-2024012613也存在该问题；并且速度上并没有发生区别；
这是我的网络拓扑结构
[image]

今天我还额外发现一个问题，WindowsPC下测速速度不正常，但是在MBP上速度就快很多；

MBP无线 -c 192.168.100.1 -V
[image]

MBP无线 -c 192.168.100.1 -V -R
[image]

MBP有线 -c 192.168.100.1 -V
[image]

MBP有线 -c 192.168.100.1 -V -R
[image]

PlutoNameless, 2024-03-08:
> 我用h28k测试了下，没有速度问题，稳定900多Mbps（未开启流量分载）。这是pci信息，pcie网卡也是RTL8111： <img alt="image" width="1103" src="https://private-user-images.githubusercontent.com/3896703/311138069-21936184-78b8-4feb-bb54-c4a6c0d2b0f4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDk4ODY5MTIsIm5iZiI6MTcwOTg4NjYxMiwicGF0aCI6Ii8zODk2NzAzLzMxMTEzODA2OS0yMTkzNjE4NC03OGI4LTRmZWItYmI1NC1jNGE2YzBkMmIwZjQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDMwOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAzMDhUMDgzMDEyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NjhjNzcwZjhmNDBhMzc3ZDE2YjlmMzFmYzA4NGVkMjAxNWE3MGUxMGQwMTQwZTgxMGVlZjE4MjdkOGI2MzAzMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.RfzDXQB-rmMJ_-80lonsgFKwCOHxNOlSDTRAfL-KKrg">
> 
> 理论上h28k的cpu性能应该比x86要低的（Arm64 四核 2GHz）。

我的Windows电脑主板的网卡是 RTL8111H，当前驱动版本是1168.15.717.2023；

PlutoNameless, 2024-03-08:
> 我的Windows电脑主板的网卡是 RTL8111H，当前驱动版本是1168.15.717.2023；

当我把驱动升级到1168.16.1123.2023之后，WIndows电脑可以跑到910Mbps了，不过-R模式下还是只有730Mbps

jjm2473, 2024-03-08:
那就在-R方式下加上-t0测试，然后用htop监控CPU使用率

jjm2473, 2024-03-08:
要么恢复出厂设置，不要安装插件测试，可以用ventoy任意创建多个干净的系统。

PlutoNameless, 2024-03-13:
> 那就在-R方式下加上-t0测试，然后用htop监控CPU使用率

可能是cpu性能不够问题？在跑iperf3的时候，会有两个核心占用率在85%以上。