istoreos/istoreos #1396: 因为固件升级到最近，导致相关下载软件（aria2，qb，transmission）下载失败，错误内容基本一至

Hwatu, 2024-03-07:
## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
因为固件升级到最新，导致相关下载软件（aria2，qb，transmission）下载失败，错误内容基本一至。
磁盘没有问题，可手动创建删除文件，只是下载软件无法开辟空间，日志如下：
A：

### (2) 路由器型号和固件版本
型号： Intel(R) Celeron(R) J4125 CPU @ 2.00GHz
固件： iStoreOS 22.03.6 2024030112 / LuCI istoreos-22.03 branch git-24.057.30647-6659faa
A：

### (3) 详细日志和/或截图
(W) 2024-03-06T03:07:09 - 文件错误警报。Torrent："x.wmv". 文件： "/mnt/usb2_2-1/download/x.wmv.!qB". 原因："x.wmv(W) 2024-03-06T03:07:09 - 文件错误警报。Torrent："x.wmv". 文件： "/mnt/usb2_2-1/download/x.wmv.!qB". 原因："x.wmv file_fallocate (/mnt/usb2_2-1/download/x.wmv.!qB) error: Not supported"
(W) 2024-03-06T03:08:12 - UPnP/NAT-PMP 端口映射失败。消息： "could not map port using UPnP[183.129.23.229]: no router found"
(W) 2024-03-06T03:08:12 - UPnP/NAT-PMP 端口映射失败。消息： "could not map port using UPnP[183.129.23.229]: no router found"
(N) 2024-03-07T01:07:13 - 发起了 qBittorrent 终止操作 file_fallocate (/mnt/usb2_2-1/download/x.wmv.!qB) error: Not supported"
(W) 2024-03-06T03:08:12 - UPnP/NAT-PMP 端口映射失败。消息： "could not map port using UPnP[183.129.23.229]: no router found"
(W) 2024-03-06T03:08:12 - UPnP/NAT-PMP 端口映射失败。消息： "could not map port using UPnP[183.129.23.229]: no router found"
(N) 2024-03-07T01:07:13 - 发起了 qBittorrent 终止操作
A：

jjm2473, 2024-03-07:
磁盘没挂载吧或者空间不够了。

closed 2024-03-07 by jjm2473

Hwatu, 2024-03-08:
> 磁盘没挂载吧或者空间不够了。

磁盘没有问题，可手动创建删除文件，只是下载软件无法开辟空间

jjm2473, 2024-03-08:
执行`lsblk`看看

jjm2473, 2024-03-08:
要么你就修改下文件预分配的方式，不要预分配

JohnRayGZ, 2025-09-16:
UPnP/NAT-PMP 端口映射失败。消息： "could not map port using UPnP
这不是很明显吗，高版本miniupnpd不接受端口转发了

jjm2473, 2025-09-17:
端口转发失败又不会导致文件下载失败。前面的文件分配错误才是真正的原因。