istoreos/istoreos #1404: MT7916(AW7916-NPD) 开除36以外的信道，下载速度有问题

Siriling, 2024-03-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：MT7916(AW7916-NPD) 开除36以外的信道，内网测速或者下载文件会断流，速度不过10mbps，速度有问题。
复现条件：
1. 7916开AX模式，信道44-48（只测试了这几个），通道宽度160
2. 手机需要支持160，并且连接速率有2400
3. 连接WiFi进行测速或者下载，出现断流，速度直降10mbps以下

### (2) 路由器型号和固件版本

A：x86，istoreos-22.03.6-2024030112-x86-64-squashfs-combined-efi.img

### (3) 详细日志和/或截图

A：
WiFi设置图
[image]

手机连接图
[image]

测速图
[image]

[image]

jjm2473, 2024-03-15:
使用157信道试试

closed 2024-04-07 by jjm2473

HC153Yuanyuan, 2024-11-24:
请问找到解决方法了吗？我的只要设为160M带宽吞吐就掉到10Mbps以内，换成80M就能跑到1000Mbps。 做AP和做STA现象一致。

HC153Yuanyuan, 2024-11-24:
> 使用157信道试试

157信道带宽最大只有80M

Siriling, 2024-11-24:
> 请问找到解决方法了吗？我的只要设为160M带宽吞吐就掉到10Mbps以内，换成80M就能跑到1000Mbps。 做AP和做STA现象一致。

没解决，is的这个驱动一直存在这个问题

jjm2473, 2024-12-28:
如果是驱动问题，那可能要等24版才会更新驱动了