istoreos/istoreos #1409: 我在istoreos的docker里启动immich容器时报错localtime exists

DcMmShan, 2024-03-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述
我用终端命令安装了docker compose
DOCKER_CONFIG=${DOCKER_CONFIG:-$HOME/.docker} 
mkdir -p $DOCKER_CONFIG/cli-plugins 
curl -SL https://github.com/docker/compose/releases/download/v2.24.7/docker-compose-linux-x86_64 -o $DOCKER_CONFIG/cli-plugins/docker-compose 
chmod +x $DOCKER_CONFIG/cli-plugins/docker-compose

再用命令安装immich，拉取镜像没问题
开始启动 docker compose up -d
启动容器时报错，共需要启动五个容器
 ✔ Container immich_postgres          Running                                                        0.0s
 ✔ Container immich_machine_learning  Running                                                        0.0s
 ✔ Container immich_redis             Running                                                        0.0s
 ⠹ Container immich_microservices     Starting                                                       3.2s
 ⠹ Container immich_server            Starting                                                       3.2s
最后两个报错，报错信息为：
Error response from daemon: error while creating mount source path '/etc/localtime': mkdir /etc/localtime: file exists

现在附上immich的.env和docker-compose.yml

.env：
   # You can find documentation for all the supported env variables at https://immich.app/docs/install/environment-variables
   # The location where your uploaded files are stored
   UPLOAD_LOCATION=/3/immich/library
   # The Immich version to use. You can pin this to a specific version like "v1.71.0"
   IMMICH_VERSION=release
   # Connection secret for postgres. You should change it to a random password
   DB_PASSWORD=postgres
   # The values below this line do not need to be changed
   ###################################################################################
   DB_HOSTNAME=immich_postgres
   DB_USERNAME=postgres
   DB_DATABASE_NAME=immich
   REDIS_HOSTNAME=immich_redis

docker-compose.yml：
      version: '3.8'
   #
   # WARNING: Make sure to use the docker-compose.yml of the current release:
   #
   # https://github.com/immich-app/immich/releases/latest/download/docker-compose.yml
   #
   # The compose file on main may not be compatible with the latest release.
   #
   name: immich
   services:
     immich-server:
       container_name: immich_server
       image: ghcr.io/immich-app/immich-server:${IMMICH_VERSION:-release}
    command: ['start.sh', 'immich']
    volumes:
      - ${UPLOAD_LOCATION}:/usr/src/app/upload
      - /etc/localtime:/etc/localtime:ro
     env_file:
      - .env
     ports:
      - 2283:3001
     depends_on:
      - redis
      - database
    restart: always
[... 100 lines cut ...]
Tue Mar 12 09:05:23 2024 daemon.err dockerd[12983]: time="2024-03-12T01:05:23.731302460Z" level=info msg="loading plugin \"io.containerd.internal.v1.shutdown\"..." runtime=io.containerd.runc.v2 type=io.containerd.internal.v1
Tue Mar 12 09:05:23 2024 daemon.err dockerd[12983]: time="2024-03-12T01:05:23.731397860Z" level=info msg="loading plugin \"io.containerd.ttrpc.v1.task\"..." runtime=io.containerd.runc.v2 type=io.containerd.ttrpc.v1
Tue Mar 12 09:05:23 2024 daemon.err dockerd[12983]: time="2024-03-12T01:05:23.731714561Z" level=info msg="starting signal loop" namespace=moby path=/tmp/run/docker/containerd/daemon/io.containerd.runtime.v2.task/moby/060cc1b7c82e2eff13752dccc72eca2f96a434a4370e46172c59c0be4790da6d pid=21080 runtime=io.containerd.runc.v2
Tue Mar 12 09:09:11 2024 kern.info kernel: [  406.582075] br-5ab05ebce491: port 4(vethe86e165) entered blocking state
Tue Mar 12 09:09:11 2024 kern.info kernel: [  406.582076] br-5ab05ebce491: port 4(vethe86e165) entered disabled state
Tue Mar 12 09:09:11 2024 kern.info kernel: [  406.582125] device vethe86e165 entered promiscuous mode
Tue Mar 12 09:09:11 2024 daemon.debug ModemManager[24372]: hotplug: event reported: action=add, name=vethfd154ab, subsystem=net
Tue Mar 12 09:09:11 2024 daemon.err ModemManager[24372]: hotplug: parent device sysfspath not found
Tue Mar 12 09:09:11 2024 daemon.debug ModemManager[24428]: hotplug: event reported: action=add, name=vethe86e165, subsystem=net
Tue Mar 12 09:09:11 2024 daemon.err ModemManager[24428]: hotplug: parent device sysfspath not found
Tue Mar 12 09:09:12 2024 daemon.err dockerd[12983]: time="2024-03-12T01:09:12.169311505Z" level=info msg="loading plugin \"io.containerd.event.v1.publisher\"..." runtime=io.containerd.runc.v2 type=io.containerd.event.v1
Tue Mar 12 09:09:12 2024 daemon.err dockerd[12983]: time="2024-03-12T01:09:12.169459007Z" level=info msg="loading plugin \"io.containerd.internal.v1.shutdown\"..." runtime=io.containerd.runc.v2 type=io.containerd.internal.v1
Tue Mar 12 09:09:12 2024 daemon.err dockerd[12983]: time="2024-03-12T01:09:12.169531807Z" level=info msg="loading plugin \"io.containerd.ttrpc.v1.task\"..." runtime=io.containerd.runc.v2 type=io.containerd.ttrpc.v1
Tue Mar 12 09:09:12 2024 daemon.err dockerd[12983]: time="2024-03-12T01:09:12.169892910Z" level=info msg="starting signal loop" namespace=moby path=/tmp/run/docker/containerd/daemon/io.containerd.runtime.v2.task/moby/a807ea1295eae4819999c8bbe69a228cd7f7fc7ff6352acae12308b9c9ff9fbf pid=24487 runtime=io.containerd.runc.v2
Tue Mar 12 09:09:12 2024 daemon.err dockerd[12983]: time="2024-03-12T01:09:12.186118242Z" level=warning msg="cleaning up after shim disconnected" id=a807ea1295eae4819999c8bbe69a228cd7f7fc7ff6352acae12308b9c9ff9fbf namespace=moby
Tue Mar 12 09:09:12 2024 daemon.err dockerd[12983]: time="2024-03-12T01:09:12.193283200Z" level=warning msg="cleanup warnings time=\"2024-03-12T01:09:12Z\" level=info msg=\"starting signal loop\" namespace=moby pid=24532 runtime=io.containerd.runc.v2\ntime=\"2024-03-12T01:09:12Z\" level=warning msg=\"failed to read init pid file\" error=\"open /tmp/run/docker/containerd/daemon/io.containerd.runtime.v2.task/moby/a807ea1295eae4819999c8bbe69a228cd7f7fc7ff6352acae12308b9c9ff9fbf/init.pid: no such file or directory\" runtime=io.containerd.runc.v2\n"
Tue Mar 12 09:09:12 2024 daemon.err dockerd[12983]: time="2024-03-12T01:09:12.193621303Z" level=error msg="copy shim log" error="read /proc/self/fd/41: file already closed"
Tue Mar 12 09:09:12 2024 daemon.err dockerd[12983]: time="2024-03-12T01:09:12.196035122Z" level=error msg="stream copy error: reading from a closed fifo"
Tue Mar 12 09:09:12 2024 daemon.err dockerd[12983]: time="2024-03-12T01:09:12.196135823Z" level=error msg="stream copy error: reading from a closed fifo"
Tue Mar 12 09:09:12 2024 daemon.debug ModemManager[24546]: hotplug: event reported: action=remove, name=vethfd154ab, subsystem=net
Tue Mar 12 09:09:12 2024 kern.info kernel: [  407.569708] br-5ab05ebce491: port 4(vethe86e165) entered disabled state
Tue Mar 12 09:09:12 2024 kern.info kernel: [  407.570061] device vethe86e165 left promiscuous mode
Tue Mar 12 09:09:12 2024 kern.info kernel: [  407.570062] br-5ab05ebce491: port 4(vethe86e165) entered disabled state
Tue Mar 12 09:09:12 2024 daemon.debug ModemManager[24566]: hotplug: event reported: action=remove, name=vethe86e165, subsystem=net
Tue Mar 12 09:09:12 2024 daemon.err dockerd[12983]: time="2024-03-12T01:09:12.866816868Z" level=error msg="a807ea1295eae4819999c8bbe69a228cd7f7fc7ff6352acae12308b9c9ff9fbf cleanup: failed to delete container from containerd: no such container"

jjm2473, 2024-03-15:
一般openwrt没有/etc/localtime文件的，所以不要-v挂载就行了

DcMmShan, 2024-03-15:
> 一般openwrt没有/etc/localtime文件的，所以不要-v挂载就行了

用的是docker compose
该应用的docker-compose.yml文件如下
version: '3.8'

#
# WARNING: Make sure to use the docker-compose.yml of the current release:
#
# https://github.com/immich-app/immich/releases/latest/download/docker-compose.yml
#
# The compose file on main may not be compatible with the latest release.
#

name: immich

services:
  immich-server:
    container_name: immich_server
    image: ghcr.io/immich-app/immich-server:${IMMICH_VERSION:-release}
    command: ['start.sh', 'immich']
    volumes:
      - ${UPLOAD_LOCATION}:/usr/src/app/upload
      - /etc/localtime:/etc/localtime:ro
      - "F:/backup/msphoto:/mnt/media/msphoto:ro" # import path in Windows system.
    env_file:
      - .env
    ports:
      - 2283:3001
    depends_on:
      - redis
      - database
    restart: always

  immich-microservices:
    container_name: immich_microservices
    image: ghcr.io/immich-app/immich-server:${IMMICH_VERSION:-release}
    # extends: # uncomment this section for hardware acceleration - see https://immich.app/docs/features/hardware-transcoding
    #   file: hwaccel.transcoding.yml
    #   service: cpu # set to one of [nvenc, quicksync, rkmpp, vaapi, vaapi-wsl] for accelerated transcoding
    command: ['start.sh', 'microservices']
    volumes:
      - ${UPLOAD_LOCATION}:/usr/src/app/upload
      - /etc/localtime:/etc/localtime:ro
      - "F:/backup/msphoto:/mnt/media/msphoto:ro" # import path in Windows system.
    env_file:
      - .env
    depends_on:
      - redis
      - database
    restart: always

  immich-machine-learning:
    container_name: immich_machine_learning
    # For hardware acceleration, add one of -[armnn, cuda, openvino] to the image tag.
    # Example tag: ${IMMICH_VERSION:-release}-cuda
    image: ghcr.io/immich-app/immich-machine-learning:${IMMICH_VERSION:-release}
    #extends: # uncomment this section for hardware acceleration - see https://immich.app/docs/features/ml-hardware-acceleration
       #file: hwaccel.ml.yml
       #service: cpu # set to one of [armnn, cuda, openvino, openvino-wsl] for accelerated inference - use the `-wsl` version for WSL2 where applicable
    volumes:
      - model-cache:/cache
    env_file:
      - .env
    restart: always

  redis:
    container_name: immich_redis
    image: registry.hub.docker.com/library/redis:6.2-alpine@sha256:51d6c56749a4243096327e3fb964a48ed92254357108449cb6e23999c37773c5
    restart: always

  database:
    container_name: immich_postgres
    image: registry.hub.docker.com/tensorchord/pgvecto-rs:pg14-v0.2.0@sha256:90724186f0a3517cf6914295b5ab410db9ce23190a2d9d0b9dd6463e3fa298f0
    environment:
      POSTGRES_PASSWORD: ${DB_PASSWORD}
      POSTGRES_USER: ${DB_USERNAME}
      POSTGRES_DB: ${DB_DATABASE_NAME}
    volumes:
      - pgdata:/var/lib/postgresql/data
    restart: always

volumes:
  pgdata:
  model-cache:

没有找到-v这个参数，用易有云文件管理器打开查看/etc有localtime这个文件，但无法下载无法编辑，ll指令显示/etc/localtime软连接到/tmp/localtime。
[image]

请问是还需要修改什么参数吗，还是OpenWrt不支持docker compose，亦或istoreOS不支持immich

jjm2473, 2024-03-16:
靠，你到底会不会用docker compose？
你自己看的教程不是有用docker部署的吗，里面不是有-v参数？
-v是docker命令的，在docker compose的yml文件里面就是volumes。

既然你都知道用docker命令做对比了，却不知道docker compose的yml对应的位置，搞笑吗？

openwrt的zoneinfo也是可以安装的，但是分成了多个不同的包名，在软件包里面搜索就知道了。

DcMmShan, 2024-03-16:
> 靠，你到底会不会用docker compose？
> 你自己看的教程不是有用docker部署的吗，里面不是有-v参数？
> -v是docker命令的，在docker compose的yml文件里面就是volumes。
> 
> 既然你都知道用docker命令做对比了，却不知道docker compose的yml对应的位置，搞笑吗？
> 
> openwrt的zoneinfo也是可以安装的，但是分成了多个不同的包名，在软件包里面搜索就知道了。

好的感谢感谢，小白有些还是不太会😂

DcMmShan, 2024-03-16:
# 解决了nice感谢大佬指导
[image]
# 解决步骤如下图所示：
[image]

jjm2473, 2024-03-16:
zoneinfo不是必须的，别用-v挂载就行了

closed 2024-06-27 by jjm2473

bh-lay, 2024-10-11:
The same error occurs when launching with the docker compose file using Docker Desktop on a Mac mini.
[image]

jjm2473, 2024-10-24:
Don't use `-v '*:/etc/localtime'`, or deleate these lines from docker-compose.yml