istoreos/istoreos #1410: 【功能请求】请求提供IStoreOS的Docker X86镜像

tan9710630, 2024-03-12:
我已经确认 ：
* [x] 没有类似的issue

## 需求描述

我运行IStoreOS在虚拟机内部，试试AIO模式进行部署，内网服务器除开IStoreOS之外存在多台Linux虚拟机组成K8S集群，所有服务都是在集群之中运行，然后使用ingress暴露地址；外部访问则使用IStoreOS做端口映射，这里存在两个问题：

1. K8S本质上多个节点均衡部署所有的服务，所以做端口映射的时候内部IP不能是固定IP，而应该使用服务所在节点IP，为此我得想法是将IStoreOS加入到集群，然后用CoreDNS分配流量，但是IStoreOS不支持部署REK2，所以我便想反过来将IStoreOS放在容器之中运行；
2. 因为路由直接拨号，所以外部访问内部服务的入口都是IStoreOS，所以如果可以将IStoreOS放在容器的话，则可以实现Ingress直接转发，借由集群提供taerfik ingress无论是负责均衡、证书、NAT穿透、防火墙都可以一步到位，最小化完成该架构的部署；

所以基于上述考虑，我希望官方能够明白我这样建议的目的，对于这个特性的需求应该也不仅仅是我一家之言，如果IStoreOS能够提供云原生的运行模式，无论是主路由还是旁路由都会出现大量的玩法，结合容器镜像私服甚至可以实现完整的版本控制，极大方便了用户管理路由。希望官方能够考虑的我提议。

jjm2473, 2024-03-15:
docker容器有几个问题：
1. 内核功能没法控制，不知道哪些内核模块可用，例如防火墙相关的模块
2. 升级功能没法使用，就算通过升级镜像的方式升级固件版本，但是安装的软件都会丢失，配置文件也不一定能保留

用虚拟机方式更合适

jjm2473, 2024-03-15:
这些问题不解决的话，就没啥实用价值

closed 2024-03-15 by jjm2473