istoreos/istoreos #1411: 关于MKT-7603U的支持

Sliccic, 2024-03-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：手头有一个闲置的360随身wifi 3，插在电脑上看可以当无线网卡，芯片是MKT-7603U，现在直接插在设备上不会出现无线，重启也不会
### (2) 路由器型号和固件版本

A：设备是r5c，固件版本是iStoreOs 22.03.6 2024030112

### (3) 详细日志和/或截图

A：root@iStoreOS:~# lsusb -tv
/:  Bus 08.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
    ID 1d6b:0002  
/:  Bus 07.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
    ID 1d6b:0002  
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    ID 1d6b:0003  
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
    ID 1d6b:0002  
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    ID 1d6b:0003  
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
    ID 1d6b:0002  
    |__ Port 1: Dev 2, If 0, Class=, Driver=, 480M
        ID 148f:760b  
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
    ID 1d6b:0001  
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
    ID 1d6b:0001

jjm2473, 2024-03-15:
你的lsusb都没任何设备

jjm2473, 2024-03-15:
哦，是`148f:760b`

Sliccic, 2024-03-16:
是很鸡肋的东西，上古产物了，本着不浪费的精神，我该如何让我的设备支持他呢？网上看了一圈都没有很好的方法

jjm2473, 2024-03-16:
自己编译固件，把驱动打上去

Sliccic, 2024-03-16:
好吧，我研究研究，目前只查到用它当网卡连别的无线网的方案，我也不晓得跟把他当热点是不是一回事

jjm2473, 2024-03-16:
这种弱鸡wifi做AP就更弱鸡了，就算能开AP估计信号也一般。真要做AP，用MT7915/6，MT7921/2等比较合适。

Sliccic, 2024-03-16:
好的感谢，那我就放弃了

jjm2473, 2024-10-24:
@IIIef 如果你的设备是x86或者树莓派，或者友善的r2s，r4s，直接在软件包安装`kmod-mt7601u`就行了