istoreos/istoreos #1419: docker 出现问题如何解决。。。甚至导致开不了机

pili-book, 2024-03-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
今天上午忽然不能访问onenva，回家发现内网也无法登录路由地址，所以重启，但是重启好多次才勉强点亮，进入路由页面发现，docker容器全部停摆了，不知道如何产生这样的问题，很多天了一直很稳定忽然搞个这样的事情。。。。而且这几天也没有动过docker，应该说没动过路由设置，使用docke的容易本身不应该有这样的问题吧。。想不明白，不知道是不是bug了

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
[image]
[image]

### (2) 路由器型号和固件版本

A：

FriendlyElec NanoPi R6S

iStoreOS 22.03.6 2024012613
### (3) 详细日志和/或截图

A：

jjm2473, 2024-03-15:
你咋这么多莫名其妙的问题。
可能中毒了，可能文件损坏了，可能硬盘坏了。

closed 2024-03-15 by jjm2473

pili-book, 2024-03-15:
我也纳闷啊。。。。。啥也没折腾。。。好好的自己就这样了，搞不清楚这是arm机器就是不稳定还是怎么滴

pili-book, 2024-03-15:
请问下你在这行这么久，你说arm
的软路由和86的软路由稳定性哪个更好了

 
Closed #1419 as completed.

jjm2473, 2024-03-15:
建议别用USB移动硬盘

pili-book, 2024-03-15:
没有啊，就是本身的emmc

jjm2473, 2024-03-16:
那可能中毒了吧

pili-book, 2024-03-16:
最近没动过。。。如何中毒呢。。。。，大哥你别开玩笑