istoreos/istoreos #1422: 是否支持ax101

benzo970210, 2024-03-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志和/或截图

A：

jjm2473, 2024-03-15:
不知道，应该不支持，至少没看到哪个驱动或fw写了ax101

hawkli, 2024-06-04:
截止今天istoreOS的LINUX内核版本是5.10，而AX101的LINUX驱动据说要6.2以上才支持，所以看来要么官方更新核心版本，要么就是驱动出来，不然openwrt没可能支持ax101。
残念！

jjm2473, 2024-06-04:
不一定要升级内核的，现在openwrt的wifi框架都是从高版本内核backport的，而不是直接使用当前内核的wifi驱动，也许之后openwrt更新wifi驱动就能支持了

NatsuZoraKanata, 2024-07-23:
> 不知道，应该不支持，至少没看到哪个驱动或fw写了ax101

我搜索到了iwlwifi-firmware-ax101
https://pkgs.staging.openwrt.org/package/master/base/mips_24kc/iwlwifi-firmware-ax101
请问有没有可能增加到咱istore里面？

jjm2473, 2024-07-23:
@NatsuZoraKanata firmware是通用的，你可以直接下载安装。但是驱动是否支持又是另一回事了。

NatsuZoraKanata, 2024-07-26:
> @NatsuZoraKanata firmware是通用的，你可以直接下载安装。但是驱动是否支持又是另一回事了，还要自己测试下。

@jjm2473 尝试下载下来试了试 https://downloads.openwrt.org/snapshots/packages/x86_64/base/iwlwifi-firmware-ax101_20240513-r1_x86_64.ipk
用opkg install，重启后无法驱动，内核日志如下：
[  350.068629] iwlwifi: No config found for PCI dev 54f0/0244, rev=0x370, rfid=0x10c000
[  350.069258] iwlwifi: probe of 0000:00:14.3 failed with error -22
对openwrt这方面不太懂，请问需要进一步配置吗？是否是因为这个驱动对硬件ID没有识别？

jjm2473, 2024-07-29:
那可能驱动还没支持吧，等下个版本固件更新以后，看看开机日志有没有变化

wkABXY, 2024-08-03:
不支持呢，研究了好久 浪费我好多时间