istoreos/istoreos #1426: 迅雷下载导致 NTFS 文件系统损坏

lizhi10, 2024-03-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：0204030112 版本，使用迅雷下载下载至 NTFS 硬盘的文件夹以后发现下载文件夹在 iStoreOS  的文件管理器里找不到，在 Windows 挂载 SMB 共享里也看不到，把硬盘接在 Windows 中，无法访问下载文件夹，显示“文件或目录损坏且无法读取”， 使用 chkdsk 命令以后显示文件系统错误，修复以后正常。
所以 NTFS 文件系统支持还是不太完美。

### (2) 路由器型号和固件版本

A：畅网 N5105

### (3) 详细日志和/或截图

A：

jjm2473, 2024-03-18:
可能只是因为磁盘掉线或者不正常断电，要么供电不足。

jjm2473, 2024-03-18:
跟NTFS没太大关系

lizhi10, 2024-03-20:
> 跟NTFS没太大关系

这种情况发生过两次，一次是带有独立供电的硬盘柜，铁威马的，供电应该还是合格的，一次是内置硬盘，用 chkdsk 命令检查以后也没有发生丢文件，就是以前不可见的文件变得可见了，文件也没有损坏。如果是不正常断电的话，应该会有文件损坏的。

jjm2473, 2024-04-03:
如果你怀疑ntfs，那就试试格式化成ext4再测试