istoreos/istoreos #1427: istoreos主机重启后，kvm虚拟机不能自动启动

lxhfems, 2024-03-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：istoreos主机重启后，kvm里面的虚拟机不能自动启动，要手动点开机。 尝试过下面截图中的  自动启动功能，不管禁用还是启用都无效。

### (2) 路由器型号和固件版本

A：x86 Generic   /
iStoreOS 22.03.6 2024031514

### (3) 详细日志和/或截图

A：
[image]

y648, 2024-03-18:
同样的问题.研究不懂.

MaximilianXuan, 2024-03-24:
我也是遇到这个问题

yanjinsongyyq, 2024-04-02:
如何解决呢？

soulripper13, 2024-06-21:
https://github.com/istoreos/istoreos/issues/1663

journeyaway, 2025-01-05:
在XML里加自启动也是不行，重启后自动运行的命令就自动删除了

journeyaway, 2025-01-05:
看了kvm的说明：注意： 开机自启动的操作，仅限于被注册的虚拟机，临时虚拟机没有该功能，估计在istore里的虚拟机都是临时的