istoreos/istoreos #1429: Proxmox直通三个网卡到iStoreOS内，ETH3不是WAN口

CWWK-TechnicalSupport, 2024-03-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：Proxmox直通三个网卡到iStoreOS内，ETH3不是WAN口

### (2) 路由器型号和固件版本

A：畅网工控机，istoreos-22.03.6-2024031514

### (3) 详细日志和/或截图

A：无

jjm2473, 2024-03-18:
istoreos不会知道哪个是wan口，只会把系统识别到的第一个网卡作为wan口。

closed 2024-03-18 by jjm2473

CWWK-TechnicalSupport, 2024-03-19:
> istoreos不会知道哪个是wan口，只会把系统识别到的第一个网卡作为wan口。

第一个口不一定是ETH0，爱快就没问题，只会把第一个做为LAN

jjm2473, 2024-03-19:
那你试试把第一个口贴上wan标志，看看爱快能不能识别到，真是搞笑。

istoreos就是把第一个口认为wan口，跟你的机器怎么标的无关。同样，爱快只是把最后一个口作为wan口，也不会管你的机器怎么标记的。

jjm2473, 2024-03-19:
凭什么认为爱快的策略就是正确的？我要是生产一个机器，第一个网口就是wan口，爱快还能识别？

CWWK-TechnicalSupport, 2024-03-20:
> 凭什么认为爱快的策略就是正确的？我要是生产一个机器，第一个网口就是wan口，爱快还能识别？

爱快是把第一个口当LAN，剩下的都没加入LAN或WAN

jjm2473, 2024-03-20:
所以呢，爱快不是一样要进去配置？

CWWK-TechnicalSupport, 2024-03-21:
> 所以呢，爱快不是一样要进去配置？

现在问题是直通以后，ETH0应该是管理口被当成了WAN口
[image]

jjm2473, 2024-04-03:
x86没有网口顺序的说法，看的是pcie的地址，以及驱动加载的顺序。还有，istoreos没有什么管理口，就是lan口管理的，如果你不知道管理口干嘛的，那就别给虚拟机。

jjm2473, 2024-04-03:
总之，istoreos会把第一个识别到到网卡，也就是eth0作为wan口，其他都作为lan口，你只要按照这个逻辑给虚拟机分配网卡就行了。虚拟机内部不可能知道你的物理网卡的顺序，以及逻辑上的作用，连物理机都不知道（只能知道pcie的地址）。