istoreos/istoreos #1432: r6s acme差件

pili-book, 2024-03-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
；有没有r6s适用的acme 插件，istore商店里没这个插件，要是能集成到自带的商店就更好了，网上看到有但是下载下来不能用。。。。小白，搞不定，只能求助官方增加acme 证书申请插件了

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：nanopi r6s

### (3) 详细日志和/或截图

A：

jjm2473, 2024-03-18:
用docker部署不就没问题了

pili-book, 2024-03-18:
大哥，我一个小白，有插件不是更加简单，目前我是用docker实现了，但是过程相对比较插件肯定更加复杂，还得代码配合还得那个去一个网站申请弄一个部署码要是插件能行，我觉得肯定会简单点吧，至少还不用靠那个网站

pili-book, 2024-03-18:
> 用docker部署不就没问题了

我从网上搜到的,我感觉比docker简单多了
[image]

jjm2473, 2024-03-19:
那就直接用acme插件啊，软件包里面不就可以安装？
运营商限制，大部分人都没法开放80端口，只能使用DNS方式申请，但是不同的DNS服务商提供的API和参数都不一样，就算用插件还是要查文档。

pili-book, 2024-03-19:
软件包？在哪里，怎么安装，详细说说，小白真找不见

pili-book, 2024-03-19:
网上找的下了一个，就是不能用

jjm2473, 2024-03-19:
下个屁啊，acme是openwrt官方支持的插件。

pili-book, 2024-03-19:
> 下个屁啊，acme是openwrt官方支持的插件。

[image]
这个吗？装哪个了

jjm2473, 2024-03-19:
软件包里先点击更新列表，然后搜索acme。

官方也有教程 https://openwrt.org/docs/guide-user/services/tls/acmesh

jjm2473, 2024-03-19:
@pili-book luci表示的就是界面啊，安装以后其他依赖一般都会自动安装。但是看起来dnsapi也要安装，因为要用dns申请。

closed 2024-03-19 by jjm2473

pili-book, 2024-03-19:
> @pili-book luci表示的就是界面啊，安装以后其他依赖一般都会自动安装。但是看起来dnsapi也要安装，因为要用dn，
好的谢谢，我看看研究下

pili-book, 2024-03-19:
> 软件包里先点击更新列表，然后搜索acme。
> 
> 官方也有教程 https://openwrt.org/docs/guide-user/services/tls/acmesh

好的谢谢我研究下