istoreos/istoreos #1433: 插件，上网时间控制timecontrol功能玄学了？

abullstar, 2024-03-18:
## 2. 详细叙述

### (1) 具体问题

A：上网时间控制功能，比如设定对电视设定，插件功能不规律的紊乱，也不是完全失效，有的时候有效，有的时候无效，即使关闭运行，再次启用，在设定的时间段依旧无效。另：同插件访问限制，黑名单，即开就有效。

### (2) 路由器型号和固件版本

A：x86软路由年后更新固件后如此，前两天更新了固件，问题还是没有解决。

jjm2473, 2024-04-08:
出问题的时候，把防火墙规则导出来看看。执行`iptables-save > /tmp/iptables.rules`，然后把`/tmp/iptables.rules`发出来看看。以及把你的timecontrol的配置发出来看看

abullstar, 2024-05-16:
# Generated by iptables-save v1.8.7 on Thu May 16 10:07:32 2024
*raw
:PREROUTING ACCEPT [11013269:8271601246]
:OUTPUT ACCEPT [143411:14380983]
:zone_lan_helper - [0:0]
-A PREROUTING -i br-LAN -m comment --comment "!fw3: lan CT helper assignment" -j zone_lan_helper
-A zone_lan_helper -p udp -m comment --comment "!fw3: Amanda backup and archiving proto" -m udp --dport 10080 -j CT --helper amanda
-A zone_lan_helper -p tcp -m comment --comment "!fw3: FTP passive connection tracking" -m tcp --dport 21 -j CT --helper ftp
-A zone_lan_helper -p udp -m comment --comment "!fw3: RAS proto tracking" -m udp --dport 1719 -j CT --helper RAS
-A zone_lan_helper -p tcp -m comment --comment "!fw3: Q.931 proto tracking" -m tcp --dport 1720 -j CT --helper Q.931
-A zone_lan_helper -p tcp -m comment --comment "!fw3: IRC DCC connection tracking" -m tcp --dport 6667 -j CT --helper irc
-A zone_lan_helper -p tcp -m comment --comment "!fw3: PPTP VPN connection tracking" -m tcp --dport 1723 -j CT --helper pptp
-A zone_lan_helper -p tcp -m comment --comment "!fw3: SIP VoIP connection tracking" -m tcp --dport 5060 -j CT --helper sip
-A zone_lan_helper -p udp -m comment --comment "!fw3: SIP VoIP connection tracking" -m udp --dport 5060 -j CT --helper sip
-A zone_lan_helper -p udp -m comment --comment "!fw3: SNMP monitoring connection tracking" -m udp --dport 161 -j CT --helper snmp
-A zone_lan_helper -p udp -m comment --comment "!fw3: TFTP connection tracking" -m udp --dport 69 -j CT --helper tftp
COMMIT
# Completed on Thu May 16 10:07:32 2024
# Generated by iptables-save v1.8.7 on Thu May 16 10:07:32 2024
*nat
:PREROUTING ACCEPT [5488:535332]
:INPUT ACCEPT [2085:150623]
:OUTPUT ACCEPT [3977:284853]
:POSTROUTING ACCEPT [289:21051]
:MINIUPNPD - [0:0]
:MINIUPNPD-POSTROUTING - [0:0]
:TIMECONTROL - [0:0]
:TIMECONTROL_REJECT - [0:0]
:TIMECONTROL_RULES - [0:0]
:postrouting_docker_rule - [0:0]
:postrouting_itv_rule - [0:0]
:postrouting_lan_rule - [0:0]
:postrouting_ont_rule - [0:0]
:postrouting_rule - [0:0]
:postrouting_wan_rule - [0:0]
:prerouting_docker_rule - [0:0]
:prerouting_itv_rule - [0:0]
:prerouting_lan_rule - [0:0]
:prerouting_ont_rule - [0:0]
:prerouting_rule - [0:0]
:prerouting_wan_rule - [0:0]
:zone_docker_postrouting - [0:0]
:zone_docker_prerouting - [0:0]
:zone_itv_postrouting - [0:0]
:zone_itv_prerouting - [0:0]
:zone_lan_postrouting - [0:0]
:zone_lan_prerouting - [0:0]
:zone_ont_postrouting - [0:0]
:zone_ont_prerouting - [0:0]
:zone_wan_postrouting - [0:0]
:zone_wan_prerouting - [0:0]
-A PREROUTING -j TIMECONTROL
-A PREROUTING -m comment --comment "!fw3: Custom prerouting rule chain" -j prerouting_rule
-A PREROUTING -i br-LAN -m comment --comment "!fw3" -j zone_lan_prerouting
-A PREROUTING -i pppoe-WAN -m comment --comment "!fw3" -j zone_wan_prerouting
-A PREROUTING -i eth1 -m comment --comment "!fw3" -j zone_itv_prerouting
-A PREROUTING -i eth0 -m comment --comment "!fw3" -j zone_ont_prerouting
-A PREROUTING -i docker0 -m comment --comment "!fw3" -j zone_docker_prerouting
-A POSTROUTING -m comment --comment "!fw3: Custom postrouting rule chain" -j postrouting_rule
-A POSTROUTING -o br-LAN -m comment --comment "!fw3" -j zone_lan_postrouting
-A POSTROUTING -o pppoe-WAN -m comment --comment "!fw3" -j zone_wan_postrouting
-A POSTROUTING -o eth1 -m comment --comment "!fw3" -j zone_itv_postrouting
-A POSTROUTING -o eth0 -m comment --comment "!fw3" -j zone_ont_postrouting
-A POSTROUTING -o docker0 -m comment --comment "!fw3" -j zone_docker_postrouting
-A MINIUPNPD -p tcp -m tcp --dport 10238 -j DNAT --to-destination 10.0.0.169:8897
-A MINIUPNPD -p udp -m udp --dport 26844 -j DNAT --to-destination 10.0.0.178:26844
-A MINIUPNPD -p tcp -m tcp --dport 44125 -j DNAT --to-destination 10.0.0.169:8897
-A MINIUPNPD-POSTROUTING -s 10.0.0.169/32 -p tcp -m tcp --sport 8897 -j MASQUERADE --to-ports 10238
-A MINIUPNPD-POSTROUTING -s 10.0.0.178/32 -p udp -m udp --sport 26844 -j MASQUERADE --to-ports 26844
-A MINIUPNPD-POSTROUTING -s 10.0.0.169/32 -p tcp -m tcp --sport 8897 -j MASQUERADE --to-ports 44125
[... 377 lines cut ...]
-A zone_wan_input -p udp -m udp --dport 2345 -m comment --comment "!fw3: OPEN-xunlei@2345" -j ACCEPT
-A zone_wan_input -p tcp -m tcp --dport 16601 -m comment --comment "!fw3: OPEN-Lucky666@16601" -j ACCEPT
-A zone_wan_input -p udp -m udp --dport 16601 -m comment --comment "!fw3: OPEN-Lucky666@16601" -j ACCEPT
-A zone_wan_input -p tcp -m tcp --dport 3002 -m comment --comment "!fw3: OPEN-SunPanel@3002" -j ACCEPT
-A zone_wan_input -p udp -m udp --dport 3002 -m comment --comment "!fw3: OPEN-SunPanel@3002" -j ACCEPT
-A zone_wan_input -p tcp -m tcp --dport 3000 -m comment --comment "!fw3: OPEN-AdGuardHome@3000" -j ACCEPT
-A zone_wan_input -p udp -m udp --dport 3000 -m comment --comment "!fw3: OPEN-AdGuardHome@3000" -j ACCEPT
-A zone_wan_input -p tcp -m tcp --dport 8090 -m comment --comment "!fw3: qBittorrent@8090" -j ACCEPT
-A zone_wan_input -p udp -m udp --dport 8090 -m comment --comment "!fw3: qBittorrent@8090" -j ACCEPT
-A zone_wan_input -s 10.0.0.169/32 -p tcp -m tcp --sport 5678 --dport 5678 -m comment --comment "!fw3: xiaoya@5678" -j ACCEPT
-A zone_wan_input -s 10.0.0.169/32 -p udp -m udp --sport 5678 --dport 5678 -m comment --comment "!fw3: xiaoya@5678" -j ACCEPT
-A zone_wan_input -s 10.0.0.169/32 -p tcp -m tcp --sport 6789 --dport 6789 -m comment --comment "!fw3: xiaoya@6789" -j ACCEPT
-A zone_wan_input -s 10.0.0.169/32 -p udp -m udp --sport 6789 --dport 6789 -m comment --comment "!fw3: xiaoya@6789" -j ACCEPT
[... cut ...]
-A zone_wan_input -m conntrack --ctstate DNAT -m comment --comment "!fw3: Accept port redirections" -j ACCEPT
-A zone_wan_input -j MINIUPNPD
-A zone_wan_input -j MINIUPNPD
-A zone_wan_input -m comment --comment "!fw3" -j zone_wan_src_DROP
-A zone_wan_output -m comment --comment "!fw3: Custom wan output rule chain" -j output_wan_rule
-A zone_wan_output -m comment --comment "!fw3" -j zone_wan_dest_ACCEPT
-A zone_wan_src_DROP -i pppoe-WAN -m comment --comment "!fw3" -j DROP
COMMIT
# Completed on Thu May 16 10:07:32 2024

abullstar, 2024-05-16:
[image]

jjm2473, 2024-05-16:
可能内核的时区有问题，不过好像也没什么方式能检查内核的时区。你试试把时间都减8小时看看

jjm2473, 2024-05-16:
还有个方式，就是执行下`hwclock --systz`，然后再重新应用下规则。

jjm2473, 2024-05-16:
执行下`date`命令看看时间和时区对不对

abullstar, 2024-05-16:
root@N100:~# hwclock --systz
root@N100:~# date
Thu May 16 12:28:02 CST 2024

abullstar, 2024-06-13:
大佬，这个上网时间控制，能否修正一下。