istoreos/istoreos #1436: R5C固件的RTC实时时钟似乎不工作

bai1828, 2024-03-19:
固件版本: iStoreOS 22.03.6 2024030112
RTC电池装上，在WAN口断网时，断电一段时间后在启动时间会回到2017年
尝试刷了FriendlyWRT同样的方法测试，没有这个现象。

jjm2473, 2024-03-19:
一般都是从网络同步时钟的

jjm2473, 2024-04-07:
我确认了RTC是存在的，而且有两个（rk808和hym8563），驱动至少几个月前就加上了，可能你不会用罢了。要用`hwclock`命令来修改硬件时间，具体自己查询这个命令的用法。

[image]
[image]
[image]

closed 2024-04-07 by jjm2473

bai1828, 2024-04-07:
谢谢指导

bai1828, 2024-04-07:
那有没有一种可能 在网络同步时间后，时间并没有被写入到RTC里呢

bai1828, 2024-04-07:
直观点的现象就是 断电重启后 docker容器的重启状态会编程好几年前启动的
[image]

bai1828, 2024-04-07:
> 我确认了RTC是存在的，而且有两个（rk808和hym8563），驱动至少几个月前就加上了，可能你不会用罢了。要用`hwclock`命令来修改硬件时间，具体自己查询这个命令的用法。
> 
> <img alt="image" width="309" src="https://private-user-images.githubusercontent.com/3896703/320253828-79e182a5-0965-4af6-9f24-994be549c6f1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI0Njg3OTgsIm5iZiI6MTcxMjQ2ODQ5OCwicGF0aCI6Ii8zODk2NzAzLzMyMDI1MzgyOC03OWUxODJhNS0wOTY1LTRhZjYtOWYyNC05OTRiZTU0OWM2ZjEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQwNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MDdUMDU0MTM4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MThmNWU2OTIxZTQxZDEyYmMwYmExZWFmMGE4ZjhmODMzNGE1MzkxNGU2YzNjZmZlNTcxOTJjOTlmYTA3OTRiZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.5QZRaHk0Bg23VSobmOjiRPFSkMAc7v4Ld6XO3Rq0SEU"> <img alt="image" width="327" src="https://private-user-images.githubusercontent.com/3896703/320253847-66fdf994-bdc7-4053-a7a4-461f6a3b46f0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI0Njg3OTgsIm5iZiI6MTcxMjQ2ODQ5OCwicGF0aCI6Ii8zODk2NzAzLzMyMDI1Mzg0Ny02NmZkZjk5NC1iZGM3LTQwNTMtYTdhNC00NjFmNmEzYjQ2ZjAucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQwNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MDdUMDU0MTM4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Njk1ZjI0Njc2YWZlZTI3MDkzYzRjMzQyYzgyYzAxOGZiMWMyM2I5Mjk1NTBhM2ZjMTI2OTgwMDZmNDdlMjY2YiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.Plyj7U0D9psqfZ7xK-31YDU1vLvDVmqZD8dUW_SQbXw"> <img alt="image" width="363" src="https://private-user-images.githubusercontent.com/3896703/320253857-4c3603e0-fa98-4c04-8c20-1c9a94b24990.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI0Njg3OTgsIm5iZiI6MTcxMjQ2ODQ5OCwicGF0aCI6Ii8zODk2NzAzLzMyMDI1Mzg1Ny00YzM2MDNlMC1mYTk4LTRjMDQtOGMyMC0xYzlhOTRiMjQ5OTAucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQwNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MDdUMDU0MTM4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MzJjZTQ3NmNmNDU4YWMyNzUyMGM4YjNiZDEwZjkyMzRmMzIxMjFjYmQxYWIyMWFmZjZjNGUxZTY4Y2E3OGEyNSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.1lqJc8Gb9Fpp1SVpxm_8YqeRt8uNo4-yMFJ-0Us7K_A">

> 我确认了RTC是存在的，而且有两个（rk808和hym8563），驱动至少几个月前就加上了，可能你不会用罢了。要用`hwclock`命令来修改硬件时间，具体自己查询这个命令的用法。
> 
> <img alt="image" width="309" src="https://private-user-images.githubusercontent.com/3896703/320253828-79e182a5-0965-4af6-9f24-994be549c6f1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI0Njg3OTgsIm5iZiI6MTcxMjQ2ODQ5OCwicGF0aCI6Ii8zODk2NzAzLzMyMDI1MzgyOC03OWUxODJhNS0wOTY1LTRhZjYtOWYyNC05OTRiZTU0OWM2ZjEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQwNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MDdUMDU0MTM4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MThmNWU2OTIxZTQxZDEyYmMwYmExZWFmMGE4ZjhmODMzNGE1MzkxNGU2YzNjZmZlNTcxOTJjOTlmYTA3OTRiZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.5QZRaHk0Bg23VSobmOjiRPFSkMAc7v4Ld6XO3Rq0SEU"> <img alt="image" width="327" src="https://private-user-images.githubusercontent.com/3896703/320253847-66fdf994-bdc7-4053-a7a4-461f6a3b46f0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI0Njg3OTgsIm5iZiI6MTcxMjQ2ODQ5OCwicGF0aCI6Ii8zODk2NzAzLzMyMDI1Mzg0Ny02NmZkZjk5NC1iZGM3LTQwNTMtYTdhNC00NjFmNmEzYjQ2ZjAucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQwNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MDdUMDU0MTM4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Njk1ZjI0Njc2YWZlZTI3MDkzYzRjMzQyYzgyYzAxOGZiMWMyM2I5Mjk1NTBhM2ZjMTI2OTgwMDZmNDdlMjY2YiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.Plyj7U0D9psqfZ7xK-31YDU1vLvDVmqZD8dUW_SQbXw"> <img alt="image" width="363" src="https://private-user-images.githubusercontent.com/3896703/320253857-4c3603e0-fa98-4c04-8c20-1c9a94b24990.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI0Njg3OTgsIm5iZiI6MTcxMjQ2ODQ5OCwicGF0aCI6Ii8zODk2NzAzLzMyMDI1Mzg1Ny00YzM2MDNlMC1mYTk4LTRjMDQtOGMyMC0xYzlhOTRiMjQ5OTAucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQwNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MDdUMDU0MTM4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MzJjZTQ3NmNmNDU4YWMyNzUyMGM4YjNiZDEwZjkyMzRmMzIxMjFjYmQxYWIyMWFmZjZjNGUxZTY4Y2E3OGEyNSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.1lqJc8Gb9Fpp1SVpxm_8YqeRt8uNo4-yMFJ-0Us7K_A">

买了个R4S 也是加上RTC电池并刷了IStoreOS  经测试后不会出现这个问题，R5C会，目前ARM就只有这两个设备

bai1828, 2024-04-07:
> 我确认了RTC是存在的，而且有两个（rk808和hym8563），驱动至少几个月前就加上了，可能你不会用罢了。要用`hwclock`命令来修改硬件时间，具体自己查询这个命令的用法。
>
> <img alt="image" width="309" [...]

实测R4S不会出现这个问题，R5C会

jjm2473, 2024-04-07:
> 网络同步时间后，时间并没有被写入到RTC里呢

本来就没做这个功能，毕竟没几个设备有RTC，你自己用hwclock设置下就行了，反正也不是经常要写入的。如果你想做的更完美，可以在`/etc/hotplug.d/ntp`里面添加个脚本写入。

我这里的r5c没有电池，所以没法测试到底rtc0和rtc1哪个才是真正有用的/有连接电池（或者两个都有用），现在两个都驱动了。

jjm2473, 2024-04-07:
istoreos本来就是为了给网络设备用的，不管做软路由还是NAS都要联网，时间同步有NTP就行了，RTC不是强需求。

bai1828, 2024-04-07:
> istoreos本来就是为了给网络设备用的，不管做软路由还是NAS都要联网，时间同步有NTP就行了，RTC不是强需求。

是的，这个确实是个少数场景，因为之前开发的Docker在启动时有部分功能涉及到时间验证，会报错，目前是通过修改代码跑起来了。

bai1828, 2024-04-07:
> > 网络同步时间后，时间并没有被写入到RTC里呢
> 
> 本来就没做这个功能，毕竟没几个设备有RTC，你自己用hwclock设置下就行了，反正也不是经常要写入的。如果你想做的更完美，可以在`/etc/hotplug.d/ntp`里面添加个脚本写入。
> 
> 我这里的r5c没有电池，所以没法测试到底rtc0和rtc1哪个才是真正有用的/有连接电池（或者两个都有用），现在两个都驱动了。

这是个好办法，另外 如果有需要的话我可以配合测试r5c的电池连接方式，告诉我如何检测即可

jjm2473, 2024-04-07:
就是用hwclock设置RTC时钟啊，然后再按你之前的测试方法看还会不会出现时间被重置的情况。RTC时钟有两个，hwclock可以分别测试，也可以都设置。

bai1828, 2024-04-07:
测出来了 是rtc1
[image]

bai1828, 2024-04-07:
> 就是用hwclock设置RTC时钟啊，然后再按你之前的测试方法看还会不会出现时间被重置的情况。RTC时钟有两个，hwclock可以分别测试，也可以都设置。

使用hwclock --systohc同步一次时间，然后断开WAN和电源，等待一段时间后在次上电得到的这个结果

bai1828, 2024-04-07:
此时系统主页的系统时间还是2017-08-04，我猜测应该是读取了RTC0的时间，而实际R5C应该读取的是RTC1

jjm2473, 2024-04-07:
所以，rtc1是有电池的是吧，那我之后把rtc0删掉

bai1828, 2024-04-07:
[image]

用这种方式同步两个rtc后断电，反复测了几次，rtc0的时间都会丢失，而rtc1不会。结论和上一张截图一样。

jjm2473, 2024-04-07:
OK

bai1828, 2024-04-07:
> 所以，rtc1是有电池的是吧，那我之后把rtc0删掉

R5C和R5S不知道会不会有区别，这是要注意的一个点

jjm2473, 2024-04-07:
一样的，厂家也嫌麻烦，不会搞太多差异

bai1828, 2024-04-07:
期待下一个版本更新。

jjm2473, 2024-04-07:
靠，我发现rk808驱动默认不能通过dtb禁用rtc。

你还是先修改 `/etc/init.d/sysfixtime`，把rtc0改成rtc1吧，我看这个脚本已经可以满足你的需求的，会自动读取/保存时间，只是默认操作rtc0。

bai1828, 2024-04-07:
> 修改 `/etc/init.d/sysfixtime`

测试了一下，可行的，断电后重启，时间能正常了

jjm2473, 2024-04-12:
最新固件已经不需要任何配置了，rk35xx机型都会检查是否有rk808以外的RTC，有的话就会使用。