istoreos/istoreos #1441: 关于网络不定时重连Network device eth0 link is down

Somnr123, 2024-03-21:
升级新版后，出现十多小时不定时重连网络，日志如下，求解决办法：Thu Mar 21 08:34:49 2024 daemon.notice netifd: Network device 'eth0' link is down
Thu Mar 21 08:34:49 2024 daemon.notice netifd: Interface 'wan' has link connectivity loss
Thu Mar 21 08:34:49 2024 daemon.notice netifd: Interface 'wan6' has link connectivity loss
Thu Mar 21 08:34:49 2024 daemon.notice netifd: Interface 'vap_wan' has link connectivity loss
Thu Mar 21 08:34:49 2024 kern.info kernel: [308234.024476] igc 0000:01:00.0 eth0: NIC Link is Down
Thu Mar 21 08:34:49 2024 daemon.notice netifd: Interface 'wan6' is now down
Thu Mar 21 08:34:49 2024 daemon.info pppd[2891]: Terminating on signal 15
Thu Mar 21 08:34:49 2024 daemon.info pppd[2891]: Connect time 1129.1 minutes.
Thu Mar 21 08:34:49 2024 daemon.info pppd[2891]: Sent 1011241346 bytes, received 3588711405 bytes.
Thu Mar 21 08:34:49 2024 daemon.notice ttyd[26998]: rops_handle_POLLIN_netlink: DELADDR
Thu Mar 21 08:34:49 2024 daemon.notice ttyd[26998]: rops_handle_POLLIN_netlink: DELADDR
Thu Mar 21 08:34:49 2024 daemon.err odhcp6c[3041]: Failed to send RS (Network unreachable)
Thu Mar 21 08:34:49 2024 daemon.err odhcp6c[3041]: Failed to send RELEASE message to ff02::1:2 (Network unreachable)
Thu Mar 21 08:34:49 2024 daemon.notice netifd: Network device 'pppoe-wan' link is down
Thu Mar 21 08:34:49 2024 daemon.notice netifd: Network alias 'pppoe-wan' link is down
Thu Mar 21 08:34:49 2024 daemon.notice netifd: Interface 'wan_6' has link connectivity loss
Thu Mar 21 08:34:49 2024 daemon.notice ttyd[26998]: rops_handle_POLLIN_netlink: DELADDR
Thu Mar 21 08:34:49 2024 daemon.info avahi-daemon[9282]: Withdrawing address record for 100.75.174.255 on pppoe-wan.
Thu Mar 21 08:34:49 2024 daemon.notice netifd: wan_6 (3041): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "wan_6" } (Permission denied)
Thu Mar 21 08:34:49 2024 daemon.info avahi-daemon[9282]: Leaving mDNS multicast group on interface pppoe-wan.IPv4 with address 100.75.174.255.
Thu Mar 21 08:34:49 2024 daemon.info avahi-daemon[9282]: In

Somnr123, 2024-03-21:
[image]
[image]

Somnr123, 2024-03-21:
系统版本[image]

leftpoint, 2024-03-22:
我也出现这样的问题，但是我的是lan口
`[ 1107.645804] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[ 1661.935301] rk_gmac-dwmac ffbe0000.ethernet eth0: Link is Down
[ 1664.976971] rk_gmac-dwmac ffbe0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[54862.615446] rk_gmac-dwmac ffbe0000.ethernet eth0: Link is Down
[54865.657133] rk_gmac-dwmac ffbe0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[68751.277725] rk_gmac-dwmac ffbe0000.ethernet eth0: Link is Down
[68754.319402] rk_gmac-dwmac ffbe0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[68933.676690] rk_gmac-dwmac ffbe0000.ethernet eth0: Link is Down
[68936.718264] rk_gmac-dwmac ffbe0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[72258.403013] rk_gmac-dwmac ffbe0000.ethernet eth0: Link is Down
[72261.444686] rk_gmac-dwmac ffbe0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[78023.222213] rk_gmac-dwmac ffbe0000.ethernet eth0: Link is Down
[78026.263888] rk_gmac-dwmac ffbe0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[80422.784486] rk_gmac-dwmac ffbe0000.ethernet eth0: Link is Down
[80425.822968] rk_gmac-dwmac ffbe0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[84011.984967] rk_gmac-dwmac ffbe0000.ethernet eth0: Link is Down
[84015.026642] rk_gmac-dwmac ffbe0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[84299.769124] rk_gmac-dwmac ffbe0000.ethernet eth0: Link is Down
[84302.810850] rk_gmac-dwmac ffbe0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[84922.963939] rk_gmac-dwmac ffbe0000.ethernet eth0: Link is Down
[84926.005509] rk_gmac-dwmac ffbe0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[94447.243749] rk_gmac-dwmac ffbe0000.ethernet eth0: Link is Down
[94450.288455] rk_gmac-dwmac ffbe0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[147360.136200] rk_gmac-dwmac ffbe0000.ethernet eth0: Link is Down
[147363.177880] rk_gmac-dwmac ffbe0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[147750.267163] rk_gmac-dwmac ffbe0000.ethernet eth0: Link is Down
[147753.308843] rk_gmac-dwmac ffbe0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[161629.807198] rk_gmac-dwmac ffbe0000.ethernet eth0: Link is Down
[161632.848614] rk_gmac-dwmac ffbe0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[165392.293642] rk_gmac-dwmac ffbe0000.ethernet eth0: Link is Down
[165395.335037] rk_gmac-dwmac ffbe0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[166166.475101] rk_gmac-dwmac ffbe0000.ethernet eth0: Link is Down
[166169.516808] rk_gmac-dwmac ffbe0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[166212.074794] rk_gmac-dwmac ffbe0000.ethernet eth0: Link is Down`

pulller, 2024-04-10:
同样是5.10.201，问题同样，基本上几十分钟会断线一次。
Wed Apr 10 20:27:04 2024 daemon.info avahi-daemon[10331]: Interface pppoe-wan.IPv4 no longer relevant for mDNS.
Wed Apr 10 20:27:04 2024 daemon.info avahi-daemon[10331]: Joining mDNS multicast group on interface pppoe-wan.IPv4 with address 219.140.149.212.
Wed Apr 10 20:27:04 2024 daemon.info avahi-daemon[10331]: New relevant interface pppoe-wan.IPv4 for mDNS.
Wed Apr 10 20:27:04 2024 daemon.info avahi-daemon[10331]: Registering new address record for 219.140.149.212 on pppoe-wan.IPv4.
Wed Apr 10 20:27:04 2024 user.notice ddns-scripts[2484]: namecheap: PID '2484' started at 2024-04-10 20:27
Wed Apr 10 20:27:05 2024 daemon.err nmbd[1730]: [2024/04/10 20:27:05.566097,  0] ../../source3/nmbd/nmbd.c:59(terminate)
Wed Apr 10 20:27:05 2024 daemon.err nmbd[1730]:   Got SIGTERM: going down...
Wed Apr 10 20:27:05 2024 daemon.err nmbd[3196]: [2024/04/10 20:27:05.608929,  0] ../../source3/nmbd/nmbd.c:901(main)
Wed Apr 10 20:27:05 2024 daemon.err nmbd[3196]:   nmbd version 4.18.8 started.
Wed Apr 10 20:27:05 2024 daemon.err nmbd[3196]:   Copyright Andrew Tridgell and the Samba Team 1992-2023
Wed Apr 10 20:27:05 2024 daemon.err nmbd[3196]: [2024/04/10 20:27:05.612999,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Wed Apr 10 20:27:05 2024 daemon.err nmbd[3196]:   query_name_response: Multiple (2) responses received for a query on subnet 192.168.2.1 for name WORKGROUP<1d>.
Wed Apr 10 20:27:05 2024 daemon.err nmbd[3196]:   This response was from IP 192.168.2.2, reporting an IP address of 192.168.2.2.
Wed Apr 10 20:27:05 2024 daemon.err smbd[3198]: [2024/04/10 20:27:05.617587,  0] ../../source3/smbd/server.c:1746(main)
Wed Apr 10 20:27:05 2024 daemon.err smbd[3198]:   smbd version 4.18.8 started.
Wed Apr 10 20:27:05 2024 daemon.err smbd[3198]:   Copyright Andrew Tridgell and the Samba Team 1992-2023
Wed Apr 10 20:27:05 2024 cron.err crond[3450]: crond (busybox 1.35.0) started, log level 5
Wed Apr 10 20:27:05 2024 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Wed Apr 10 20:27:05 2024 daemon.info avahi-daemon[10331]: Joining mDNS multicast group on interface eth0.IPv6 with address fe80::62be:b4ff:fe14:ccbc.
Wed Apr 10 20:27:05 2024 daemon.info avahi-daemon[10331]: New relevant interface eth0.IPv6 for mDNS.
Wed Apr 10 20:27:05 2024 daemon.info avahi-daemon[10331]: Registering new address record for fe80::62be:b4ff:fe14:ccbc on eth0.*.
Wed Apr 10 20:27:08 2024 daemon.info dnsmasq[1]: started, version 2.89 cachesize 1000
Wed Apr 10 20:27:08 2024 daemon.info dnsmasq[1]: DNS service limited to local subnets
Wed Apr 10 20:27:08 2024 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset no-nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Wed Apr 10 20:27:08 2024 daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Wed Apr 10 20:27:08 2024 daemon.info dnsmasq-dhcp[1]: DHCP, IP range 192.168.2.100 -- 192.168.2.248, lease time 12h
Wed Apr 10 20:27:08 2024 daemon.info dnsmasq[1]: using only locally-known addresses for test
Wed Apr 10 20:27:08 2024 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Wed Apr 10 20:27:08 2024 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Wed Apr 10 20:27:08 2024 daemon.info dnsmasq[1]: using only locally-known addresses for local
Wed Apr 10 20:27:08 2024 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Wed Apr 10 20:27:08 2024 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Wed Apr 10 20:27:08 2024 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Wed Apr 10 20:27:08 2024 daemon.info dnsmasq[1]: reading /tmp/resolv.conf.d/resolv.conf.auto
Wed Apr 10 20:27:08 2024 daemon.info dnsmasq[1]: using nameserver 114.114.114.114#53
Wed Apr 10 20:27:08 2024 daemon.info dnsmasq[1]: using nameserver 8.8.8.8#53
Wed Apr 10 20:27:08 2024 daemon.info dnsmasq[1]: using only locally-known addresses for test
Wed Apr 10 20:27:08 2024 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Wed Apr 10 20:27:08 2024 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Wed Apr 10 20:27:08 2024 daemon.info dnsmasq[1]: using only locally-known addresses for local
Wed Apr 10 20:27:08 2024 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Wed Apr 10 20:27:08 2024 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Wed Apr 10 20:27:08 2024 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Wed Apr 10 20:27:08 2024 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Wed Apr 10 20:27:08 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 6 names
Wed Apr 10 20:27:08 2024 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 4 names
Wed Apr 10 20:27:08 2024 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Wed Apr 10 20:27:08 2024 daemon.info avahi-daemon[10331]: Leaving mDNS multicast group on interface pppoe-wan.IPv6 with address fe80::ad4b:dda2:16b:f62f.
Wed Apr 10 20:27:08 2024 daemon.info avahi-daemon[10331]: Joining mDNS multicast group on interface pppoe-wan.IPv6 with address 240e:368:1404:2792:ad4b:dda2:16b:f62f.
Wed Apr 10 20:27:08 2024 daemon.info avahi-daemon[10331]: Registering new address record for 240e:368:1404:2792:ad4b:dda2:16b:f62f on pppoe-wan.*.
[... cut ...]
Wed Apr 10 21:12:55 2024 daemon.err nmbd[14142]:   This response was from IP 192.168.2.2, reporting an IP address of 192.168.2.2.
Wed Apr 10 21:18:04 2024 daemon.err nmbd[14142]: [2024/04/10 21:18:04.265076,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Wed Apr 10 21:18:04 2024 daemon.err nmbd[14142]:   query_name_response: Multiple (2) responses received for a query on subnet 192.168.2.1 for name WORKGROUP<1d>.
Wed Apr 10 21:18:04 2024 daemon.err nmbd[14142]:   This response was from IP 192.168.2.2, reporting an IP address of 192.168.2.2.

luckyBricks, 2024-04-10:
Same issue here:

> Wed Apr 10 22:44:21 2024 daemon.notice netifd: Network device 'eth4' link is up
> Wed Apr 10 22:44:21 2024 kern.info kernel: [ 2588.029824] igc baaf:00:00.0 eth4: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX/TX
> Wed Apr 10 22:44:21 2024 kern.info kernel: [ 2588.029977] br-lan: port 4(eth4) entered blocking state
> Wed Apr 10 22:44:21 2024 kern.info kernel: [ 2588.029981] br-lan: port 4(eth4) entered forwarding state
> Wed Apr 10 22:44:22 2024 daemon.notice netifd: Network device 'eth3' link is up
> Wed Apr 10 22:44:22 2024 kern.info kernel: [ 2588.459809] igc c0a6:00:00.0 eth3: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX/TX
> Wed Apr 10 22:44:22 2024 kern.info kernel: [ 2588.459972] br-lan: port 3(eth3) entered blocking state
> Wed Apr 10 22:44:22 2024 kern.info kernel: [ 2588.459977] br-lan: port 3(eth3) entered forwarding state
> Wed Apr 10 22:44:23 2024 daemon.notice netifd: Network device 'eth3' link is down
> Wed Apr 10 22:44:23 2024 kern.info kernel: [ 2589.179687] igc c0a6:00:00.0 eth3: NIC Link is Down
> Wed Apr 10 22:44:23 2024 kern.info kernel: [ 2589.179848] br-lan: port 3(eth3) entered disabled state
> Wed Apr 10 22:44:26 2024 daemon.notice netifd: Network device 'eth4' link is down
> Wed Apr 10 22:44:26 2024 kern.info kernel: [ 2592.969299] igc baaf:00:00.0 eth4: NIC Link is Down
> Wed Apr 10 22:44:26 2024 kern.info kernel: [ 2592.969422] br-lan: port 4(eth4) entered disabled state
> Wed Apr 10 22:44:27 2024 daemon.notice netifd: Network device 'eth3' link is up
> Wed Apr 10 22:44:27 2024 kern.info kernel: [ 2593.099596] igc c0a6:00:00.0 eth3: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX/TX
> Wed Apr 10 22:44:27 2024 kern.info kernel: [ 2593.099763] br-lan: port 3(eth3) entered blocking state
> Wed Apr 10 22:44:27 2024 kern.info kernel: [ 2593.099768] br-lan: port 3(eth3) entered forwarding state
> Wed Apr 10 22:44:28 2024 daemon.notice netifd: Network device 'eth3' link is down
> Wed Apr 10 22:44:28 2024 kern.info kernel: [ 2594.669253] igc c0a6:00:00.0 eth3: NIC Link is Down
> Wed Apr 10 22:44:28 2024 kern.info kernel: [ 2594.669513] br-lan: port 3(eth3) entered disabled state
> Wed Apr 10 22:44:30 2024 daemon.notice netifd: Network device 'eth4' link is up
> Wed Apr 10 22:44:30 2024 kern.info kernel: [ 2596.099531] igc baaf:00:00.0 eth4: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX/TX
> Wed Apr 10 22:44:30 2024 kern.info kernel: [ 2596.099681] br-lan: port 4(eth4) entered blocking state
> Wed Apr 10 22:44:30 2024 kern.info kernel: [ 2596.099685] br-lan: port 4(eth4) entered forwarding state
> Wed Apr 10 22:44:31 2024 daemon.notice netifd: Network device 'eth4' link is down
> Wed Apr 10 22:44:31 2024 kern.info kernel: [ 2597.369232] igc baaf:00:00.0 eth4: NIC Link is Down
> Wed Apr 10 22:44:31 2024 kern.info kernel: [ 2597.369412] br-lan: port 4(eth4) entered disabled state
> Wed Apr 10 22:44:34 2024 daemon.notice netifd: Network device 'eth4' link is up
> Wed Apr 10 22:44:34 2024 kern.info kernel: [ 2600.519327] igc baaf:00:00.0 eth4: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX/TX
> Wed Apr 10 22:44:34 2024 kern.info kernel: [ 2600.519476] br-lan: port 4(eth4) entered blocking state
> Wed Apr 10 22:44:34 2024 kern.info kernel: [ 2600.519481] br-lan: port 4(eth4) entered forwarding state
> Wed Apr 10 22:44:44 2024 daemon.notice netifd: Network device 'eth3' link is up
> Wed Apr 10 22:44:44 2024 kern.info kernel: [ 2610.658977] igc c0a6:00:00.0 eth3: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX/TX
> Wed Apr 10 22:44:44 2024 kern.info kernel: [ 2610.659144] br-lan: port 3(eth3) entered blocking state
> Wed Apr 10 22:44:44 2024 kern.info kernel: [ 2610.659150] br-lan: port 3(eth3) entered forwarding state
> Wed Apr 10 22:44:44 2024 kern.info kernel: [ 2610.738683] igc c0a6:00:00.0 eth3: NIC Link is Down
> Wed Apr 10 22:44:45 2024 daemon.notice netifd: Network device 'eth3' link is down
> Wed Apr 10 22:44:45 2024 kern.info kernel: [ 2611.688785] br-lan: port 3(eth3) entered disabled state
> Wed Apr 10 22:44:48 2024 daemon.notice netifd: Network device 'eth3' link is up
> Wed Apr 10 22:44:48 2024 kern.info kernel: [ 2614.928941] igc c0a6:00:00.0 eth3: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX/TX
> Wed Apr 10 22:44:48 2024 kern.info kernel: [ 2614.929103] br-lan: port 3(eth3) entered blocking state
> Wed Apr 10 22:44:48 2024 kern.info kernel: [ 2614.929107] br-lan: port 3(eth3) entered forwarding state
> 

NIC continuously restarting and no connections could be established, no choice besides rolling back.

Somnr123, 2024-04-25:
光猫的问题，找运营商换了光猫，就没事了

closed 2024-04-25 by Somnr123