istoreos/istoreos #1446: 通过“易有云文件管理”上传文件出错

desion628, 2024-03-24:
## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 故障描述：上传文件到/mnt/disk4/media/movie（/mnt/disk4/为某个分区的挂载目录），提示“Failed because: Error: tus: failed to upload chunk at offset 1077415936, caused by [object ProgressEvent], originated from request (method: PATCH, url: /cgi-bin/luci/linkease/api/file/transmit/5396c016cc9a850cdec391d86ce16b09, response code: n/a, response text: n/a, request id: n/a)”

A：

### (2) 路由器型号：VMware, Inc. VMware Virtual Platform，固件版本：iStoreOS 22.03.6 2024031514 / LuCI istoreos-22.03 branch git-24.057.30647-6659faa

A：

### (3) 截图：
[image]

A：

jjm2473, 2024-06-27:
related https://github.com/istoreos/istoreos/issues/1654

liuwleo, 2024-06-27:
终端结果如下：

[image]

谢谢

seacatwu, 2024-07-15:
我用lucky反代会出现这个问题，但是本地局域网没这个问题。目前还不知道为什么

my1639963, 2024-07-25:
@jannson

my1639963, 2024-07-25:
<html>
<body>

架构 | Intel(R) Celeron(R) CPU 3855U @ 1.60GHz
-- | --
目标平台 | x86/64
固件版本 | iStoreOS 22.03.6 2024071911 / LuCI openwrt-22.03 branch git-24.067.02332-4c1ddfb
内核版本 | 5.10.201

</body>
</html>

文件上传时候强制使用http 协议，路由器uhttpd 配置https访问后无法正常上传文件

liuwleo, 2024-07-26:
> 收到，这个看起来是 BUG。我记得之前修复了。但不知道为何冒出来了。后续我来修复。

谢谢！

hapi01, 2024-10-07:
> @jannson
> 
> 架构	Intel(R) Celeron(R) CPU 3855U @ 1.60GHz
> 目标平台	x86/64
> 固件版本	iStoreOS 22.03.6 [2024071911](tel:2024071911) / LuCI openwrt-22.03 branch git-24.067.02332-4c1ddfb
> 内核版本	5.10.201
> 文件上传时强制使用http协议，路由器uhttpd配置https访问后无法正常上传文件
> 
> ![image](https://private-user-images.githubusercontent.com/16329362/351925808-5fcec385-4f6d-411e-b404-b581d804db3a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjgyODkwNTAsIm5iZiI6MTcyODI4ODc1MCwicGF0aCI6Ii8xNjMyOTM2Mi8zNTE5MjU4MDgtNWZjZWMzODUtNGY2ZC00MTFlLWI0MDQtYjU4MWQ4MDRkYjNhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDA3VDA4MTIzMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI2M2FjODkwMTQwNDNiMGYzZjQ2MzU3YmNiYmRjZTAzNTlkM2FhM2ZkOTJlOWMxYWE4OTYzY2MwNGY2ZjJmNjcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.8hKcoZBb2fKYA_70az_8nZpn12ldRnDncJb9okujmHE)

我没配置过https，重新安装一遍istore还是同样的报错，上传不了一点

liuwleo, 2024-11-23:
大佬，这个问题还修么😂