istoreos/istoreos #1447: 请问docker出问题如何修复？

heku66, 2024-03-24:
你好，感谢作者提供这么好的工具。
在使用docker时，任意一个操作【修改网段或修改daemon.json文件】错误会导致docker无法启动，不知道如何修复？谢谢。
Sun Mar 24 22:25:59 2024 daemon.err dockerd[3773]: time="2024-03-24T14:25:59Z" level=warning msg="containerd config version `1` has been deprecated and will be removed in containerd v2.0, please switch to version `2`, see https://github.com/containerd/containerd/blob/main/docs/PLUGINS.md#version-header"
Sun Mar 24 22:25:59 2024 daemon.err modprobe: failed to find a module named aufs
Sun Mar 24 22:25:59 2024 daemon.err dockerd[3773]: time="2024-03-24T14:25:59.753541745Z" level=warning msg="failed to load plugin io.containerd.snapshotter.v1.devmapper" error="devmapper not configured"
Sun Mar 24 22:25:59 2024 daemon.err dockerd[3773]: time="2024-03-24T14:25:59.755848508Z" level=warning msg="could not use snapshotter devmapper in metadata plugin" error="devmapper not configured"
Sun Mar 24 22:25:59 2024 daemon.err dockerd[3773]: time="2024-03-24T14:25:59.763227781Z" level=error msg="failed to initialize a tracing processor \"otlp\"" error="no OpenTelemetry endpoint: skip plugin"
Sun Mar 24 22:26:00 2024 daemon.info modprobe: nf_conntrack is already loaded
Sun Mar 24 22:26:00 2024 daemon.info modprobe: nf_conntrack_netlink is already loaded
Sun Mar 24 22:26:01 2024 daemon.err dockerd[3773]: failed to start daemon: Error initializing network controller: Error creating default "bridge" network: failed to allocate gateway (172.17.0.0): Address already in use

jjm2473, 2024-04-07:
'failed to allocate gateway (172.17.0.0): Address already in use'，你的系统里有其他网络接口是这个网段的吗？

heku66, 2024-04-07:
> > failed to allocate gateway (172.17.0.0): Address already in use
> 
> 你的系统里有其他网络接口是这个网段的吗？

不会不会，这个是新设置的网段，这平台里面的docker很容易出问题。

jjm2473, 2024-04-08:
执行`ip addr`把结果发出来，以及`daemon.json`的内容

jjm2473, 2024-04-08:
为啥你要自己修改daemon.json？后台不能配置？