istoreos/istoreos #1450: 关于istoreos作为旁路由端口转发问题

jasonnmy, 2024-03-25:
Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：在旁路由开clash情况下，无法通过国外ip访问内网服务器，连接超时，同时Clash中没有任何日志记录。猜测是外网进来的连接没有通过openWRT的PREROUTING链，而是通过主路由（10.0.0.1）的转发直接到了内网服务器（10.0.0.11）上。
当内网服务器接收到连接命令之后，再通过旁路由openWRT发出响应，这时连接进入到openWRT的PREROUTING链，最终进入到了Clash中，但是这个TCP连接只有响应，Clash应该是丢弃了（瞎猜）。然后超时了
如果我把主路由转发到旁路由，再通过旁路由转发到内网服务器，这时外网访问没问题了，但是内网通过域名访问就不行了（有些配置文件写的是域名地址，所以内网有些需要通过域名访问，而不是用内网ip直接访问）。但如果打开防火墙区域的ip动态伪装，这时候都能访问了，但是内网服务器收到的ip就永远是旁路由ip了，现在有办法能让他显示源ip吗？

### (2) 路由器型号和固件版本

A：iStoreOS 22.03.6 2024031514

### (3) 详细日志和/或截图

A：

jjm2473, 2024-04-07:
为什么有人会用旁路由做端口转发？主路由不能转发给目标设备？

jjm2473, 2024-04-07:
如果你要精细控制NAT，那就不要勾选ip动态伪装，而是自己配置NAT规则，你这种应该配置DNAT就行了。

jjm2473, 2024-04-07:
但是不NAT的话，可以会导致连接建立出问题。

jjm2473, 2024-04-07:
还有，为什么不在主路由做端口转发？

closed 2024-04-07 by jjm2473