istoreos/istoreos #1452: 最近几个版本Docker容器状态里的 CPU、内存使用率没有数值了

byteinfo, 2024-03-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：最近几个版本Docker容器状态里的 CPU、内存使用率没有数值了

### (2) 路由器型号和固件版本

A：R4S

### (3) 详细日志和/或截图

A：
[image]

jjm2473, 2024-04-07:
可能只是因为你这个容器里面没有相应的命令，或者命令不兼容，跟固件或docker无关。

closed 2024-04-07 by jjm2473

byteinfo, 2024-04-11:
> 可能只是因为你这个容器里面没有相应的命令，或者命令不兼容，跟固件或docker无关。

在20231027版本更新之前，状态里一直是空白，20231027开始显示正常，后面没注意哪个版本开始cpu，内存信息不显示了，容器还是之前的容器

jjm2473, 2024-04-12:
你自己升级过luci-app-dockerman软件包吗？

byteinfo, 2024-04-12:
> 你自己升级过luci-app-dockerman软件包吗？

应该是没有 v0.5.13-20231026