istoreos/istoreos #1454: R6s重启后错误的判断，看不懂那是什么错误

pili-book, 2024-03-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
Tue Mar 26 18:45:52 2024 daemon.debug ModemManager[7411]: hotplug: cached event found: action=add, name=tty56, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty56
Tue Mar 26 18:45:52 2024 daemon.debug ModemManager[7411]: hotplug: event reported: action=add, name=tty56, subsystem=tty
Tue Mar 26 18:45:52 2024 daemon.err ModemManager[7411]: hotplug: parent device sysfspath not found
Tue Mar 26 18:45:52 2024 daemon.debug ModemManager[7411]: hotplug: cached event found: action=add, name=tty18, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty18
Tue Mar 26 18:45:52 2024 daemon.debug ModemManager[7411]: hotplug: event reported: action=add, name=tty18, subsystem=tty
Tue Mar 26 18:45:52 2024 daemon.err ModemManager[7411]: hotplug: parent device sysfspath not found
Tue Mar 26 18:45:52 2024 daemon.debug ModemManager[7411]: hotplug: cached event found: action=add, name=tty1, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty1
Tue Mar 26 18:45:52 2024 daemon.debug ModemManager[7411]: hotplug: event reported: action=add, name=tty1, subsystem=tty
Tue Mar 26 18:45:52 2024 daemon.err ModemManager[7411]: hotplug: parent device sysfspath not found
Tue Mar 26 18:45:52 2024 daemon.debug ModemManager[7411]: hotplug: cached event found: action=add, name=tty46, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty46
Tue Mar 26 18:45:52 2024 daemon.debug ModemManager[7411]: hotplug: event reported: action=add, name=tty46, subsystem=tty
Tue Mar 26 18:45:52 2024 daemon.err ModemManager[7411]: hotplug: parent device sysfspath not found
Tue Mar 26 18:45:52 2024 daemon.debug ModemManager[7411]: hotplug: cached event found: action=add, name=tty36, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty36
Tue Mar 26 18:45:52 2024 daemon.debug ModemManager[7411]: hotplug: event reported: action=add, name=tty36, subsystem=tty
Tue Mar 26 18:45:52 2024 daemon.err ModemManager[7411]: hotplug: parent device sysfspath not found
Tue Mar 26 18:45:52 2024 daemon.debug ModemManager[7411]: hotplug: cached event found: action=add, name=tty26, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty26
Tue Mar 26 18:45:52 2024 daemon.debug ModemManager[7411]: hotplug: event reported: action=add, name=tty26, subsystem=tty
Tue Mar 26 18:45:52 2024 daemon.err ModemManager[7411]: hotplug: parent device sysfspath not found
Tue Mar 26 18:45:52 2024 daemon.debug ModemManager[7411]: hotplug: cached event found: action=add, name=tty54, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty54
Tue Mar 26 18:45:52 2024 daemon.debug ModemManager[7411]: hotplug: event reported: action=add, name=tty54, subsystem=tty
Tue Mar 26 18:45:52 2024 daemon.err ModemManager[7411]: hotplug: parent device sysfspath not found
Tue Mar 26 18:45:52 2024 daemon.debug ModemManager[7411]: hotplug: cached event found: action=add, name=tty16, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty16
Tue Mar 26 18:45:52 2024 daemon.debug ModemManager[7411]: hotplug: event reported: action=add, name=tty16, subsystem=tty
Tue Mar 26 18:45:52 2024 user.notice firewall: Reloading firewall due to ifup of vap_lan (vap-lan-peer)
Tue Mar 26 18:45:52 2024 daemon.err ModemManager[7411]: hotplug: parent device sysfspath not found
Tue Mar 26 18:45:52 2024 daemon.debug ModemManager[7411]: hotplug: cached event found: action=add, name=tty44, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty44
Tue Mar 26 18:45:52 2024 daemon.debug ModemManager[7411]: hotplug: event reported: action=add, name=tty44, subsystem=tty
Tue Mar 26 18:45:52 2024 daemon.err ModemManager[7411]: hotplug: parent device sysfspath not found
Tue Mar 26 18:45:52 2024 daemon.debug ModemManager[7411]: hotplug: cached event found: action=add, name=console, subsystem=tty, sysfspath=/sys/devices/virtual/tty/console
Tue Mar 26 18:45:52 2024 daemon.debug ModemManager[7411]: hotplug: event reported: action=add, name=console, subsystem=tty
Tue Mar 26 18:45:52 2024 daemon.err ModemManager[7411]: hotplug: parent device sysfspath not found
Tue Mar 26 18:45:52 2024 daemon.debug ModemManager[7411]: hotplug: cached event found: action=add, name=tty34, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty34
Tue Mar 26 18:45:52 2024 daemon.debug ModemManager[7411]: hotplug: event reported: action=add, name=tty34, subsystem=tty
Tue Mar 26 18:45:52 2024 daemon.err ModemManager[7411]: hotplug: parent device sysfspath not found
Tue Mar 26 18:45:52 2024 daemon.debug ModemManager[7411]: hotplug: cached event found: action=add, name=tty62, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty62
Tue Mar 26 18:45:52 2024 daemon.debug ModemManager[7411]: hotplug: event reported: action=add, name=tty62, subsystem=tty
Tue Mar 26 18:45:52 2024 user.notice adb-enablemodem: INFO: waiting for device
Tue Mar 26 18:45:52 2024 daemon.err ModemManager[7411]: hotplug: parent device sysfspath not found
Tue Mar 26 18:45:53 2024 daemon.debug ModemManager[7411]: hotplug: cached event found: action=add, name=tty24, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty24
Tue Mar 26 18:45:53 2024 daemon.debug ModemManager[7411]: hotplug: event reported: action=add, name=tty24, subsystem=tty
Tue Mar 26 18:45:53 2024 daemon.err ModemManager[7411]: hotplug: parent device sysfspath not found
Tue Mar 26 18:45:53 2024 daemon.debug ModemManager[7411]: hotplug: cached event found: action=add, name=tty8, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty8
Tue Mar 26 18:45:53 2024 daemon.debug ModemManager[7411]: hotplug: event reported: action=add, name=tty8, subsystem=tty
Tue Mar 26 18:45:53 2024 daemon.err ModemManager[7411]: hotplug: parent device sysfspath not found
Tue Mar 26 18:45:53 2024 daemon.debug ModemManager[7411]: hotplug: cached event found: action=add, name=tty52, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty52
Tue Mar 26 18:45:53 2024 daemon.debug ModemManager[7411]: hotplug: event reported: action=add, name=tty52, subsystem=tty
Tue Mar 26 18:45:53 2024 daemon.err ModemManager[7411]: hotplug: parent device sysfspath not found
Tue Mar 26 18:45:53 2024 daemon.debug ModemManager[7411]: hotplug: cached event found: action=add, name=tty14, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty14
Tue Mar 26 18:45:53 2024 daemon.debug ModemManager[7411]: hotplug: event reported: action=add, name=tty14, subsystem=tty
Tue Mar 26 18:45:53 2024 daemon.err ModemManager[7411]: hotplug: parent device sysfspath not found
Tue Mar 26 18:45:53 2024 daemon.debug ModemManager[7411]: hotplug: cached event found: action=add, name=tty42, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty42
Tue Mar 26 18:45:53 2024 daemon.debug ModemManager[7411]: hotplug: event reported: action=add, name=tty42, subsystem=tty
Tue Mar 26 18:45:53 2024 daemon.err ModemManager[7411]: hotplug: parent device sysfspath not found
Tue Mar 26 18:45:53 2024 daemon.debug ModemManager[7411]: hotplug: cached event found: action=add, name=tty32, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty32
Tue Mar 26 18:45:53 2024 daemon.debug ModemManager[7411]: hotplug: event reported: action=add, name=tty32, subsystem=tty
Tue Mar 26 18:45:53 2024 daemon.err ModemManager[7411]: hotplug: parent device sysfspath not found
Tue Mar 26 18:45:53 2024 daemon.debug ModemManager[7411]: hotplug: cached event found: action=add, name=tty60, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty60
Tue Mar 26 18:45:53 2024 daemon.debug ModemManager[7411]: hotplug: event reported: action=add, name=tty60, subsystem=tty
Tue Mar 26 18:45:53 2024 daemon.err ModemManager[7411]: hotplug: parent device sysfspath not found
Tue Mar 26 18:45:53 2024 daemon.debug ModemManager[7411]: hotplug: cached event found: action=add, name=ttyFIQ0, subsystem=tty, sysfspath=/sys/devices/platform/fiq_debugger.0/tty/ttyFIQ0
Tue Mar 26 18:45:53 2024 daemon.debug ModemManager[7411]: hotplug: event reported: action=add, name=ttyFIQ0, subsystem=tty
Tue Mar 26 18:45:53 2024 daemon.err ModemManager[7411]: hotplug: parent device sysfspath not found
Tue Mar 26 18:45:53 2024 daemon.debug ModemManager[7411]: hotplug: cached event found: action=add, name=tty22, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty22
Tue Mar 26 18:45:53 2024 daemon.debug ModemManager[7411]: hotplug: event reported: action=add, name=tty22, subsystem=tty
Tue Mar 26 18:45:53 2024 daemon.err ModemManager[7411]: hotplug: parent device sysfspath not found
[... cut ...]
Tue Mar 26 18:45:55 2024 daemon.info ModemManager[13349]: hotplug: no need to wait for modem at sysfs path /sys/devices/platform/fe190000.pcie/pci0004:40/0004:40:00.0/0004:41:00.0
Tue Mar 26 18:45:55 2024 daemon.debug ModemManager[13418]: hotplug: event reported: action=add, name=gre0, subsystem=net
Tue Mar 26 18:45:55 2024 daemon.err ModemManager[13418]: hotplug: parent device sysfspath not found

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：这是路由器重启后的日志，看到刚启动的时候很多错误的日志，看不明白是什么问题，请官方看看这是啥问题，目前使用倒是正常

### (2) 路由器型号和固件版本

A：A：设备型号
FriendlyElec NanoPi R6S
固件版本
iStoreOS 22.03.6 2024031514

### (3) 详细日志和/或截图

A：

jjm2473, 2024-04-07:
你的问题可真多，别用了好吧。没问题发个毛线的PR。

closed 2024-04-07 by jjm2473

pili-book, 2024-04-07:
这不是帮你们完善固件了不是。。。。发现有问题，给你们看看么

jjm2473, 2024-04-07:
你不会查下ModemManager是干嘛的？如果不是你用得上的功能，你管它干嘛。它就是日志打得多点而已。你要是实在有洁癖，那就卸载掉它。