istoreos/istoreos #1461: 手动安装好docker后，WARNING: No swap limit support

etfrommars, 2024-03-28:
使用的是 亚瑟京东云ax1800 pro ，刷了istoreos-ipq60xx-23.09.09-jdcloud_ax1800_pro-squashfs-kernel-rootfs

扩容了/overlay
安装 dockerd  docker  luci-app-dockerman
运行docker info
——————————————————————————————————————

root@iStoreOS:/# docker info

Client:
 Context:    default
 Debug Mode: false

Server:
 Containers: 0
  Running: 0
  Paused: 0
  Stopped: 0
 Images: 1
 Server Version: 20.10.25
 Storage Driver: overlay2
  Backing Filesystem: extfs
  Supports d_type: true
  Native Overlay Diff: true
  userxattr: false
 Logging Driver: json-file
 Cgroup Driver: cgroupfs
 Cgroup Version: 1
 Plugins:
  Volume: local
  Network: bridge host ipvlan macvlan null overlay
  Log: awslogs fluentd gcplogs gelf journald json-file local logentries splunk syslog
 Swarm: inactive
 Runtimes: io.containerd.runc.v2 io.containerd.runtime.v1.linux runc
 Default Runtime: runc
 Init Binary: docker-init
 containerd version: 
 runc version: 
 init version: de40ad0
 Kernel Version: 4.4.60
 Operating System: OpenWrt 19.07-SNAPSHOT
 OSType: linux
 Architecture: aarch64
 CPUs: 4
 Total Memory: 406.1MiB
 Name: iStoreOS
 ID: 273M:IQMH:Z4HY:ORBU:7H62:ZEYD:WAW4:SJR6:XZ5Q:N3EC:LFNK:UWSG
 Docker Root Dir: /opt/docker
 Debug Mode: false
 Registry: https://index.docker.io/v1/
 Labels:
 Experimental: false
 Insecure Registries:
  127.0.0.0/8
 Registry Mirrors:
  https://hub-mirror.c.163.com/
 Live Restore Enabled: false

WARNING: No swap limit support
WARNING: No kernel memory TCP limit support
WARNING: No blkio throttle.read_bps_device support
WARNING: No blkio throttle.write_bps_device support
WARNING: No blkio throttle.read_iops_device support
WARNING: No blkio throttle.write_iops_device support
WARNING: bridge-nf-call-iptables is disabled
WARNING: bridge-nf-call-ip6tables is disabled

etfrommars, 2024-03-28:
安装迅雷的时候拉取时出错。

[image]

jjm2473, 2024-04-15:
看起来内存不足

neverdead117, 2024-07-27:
https://blog.csdn.net/qq_41821301/article/details/105243294
但我自己还没搞定

jjm2473, 2024-07-29:
跟“No swap limit support”没啥关系，要关注的是“out of memory”。

jjm2473, 2024-07-29:
另外，这个机型是lean支持的，如果真是固件问题需要找他反馈。