istoreos/istoreos #1466: DNS配置与Tailscale冲突的网络问题

wldxx888, 2024-03-31:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：安装了tailscale在启用状态下，每次重启或开机后在首页DNS出会循环出现DNS检测中，然后提示“DNS配置错误”提示。虽然网络可以使用，但是严重影响网络速度，且无法检查固件更新和istore插件安装，无法修改DNS。今天偶然间在升级tailscale临时关闭服务的时候，DNS立即就恢复到正常联网状态，且升级好tailscale启用的情况下，DNS正常。因设置为每日4:00重启一次路由器，在重启后又回到上述DNS配置错误的BUG中。

### (2) 路由器型号和固件版本

A：
[image]

wldxx888, 2024-03-31:
### (3) 详细日志和/或截图

A：系统日志
Sun Mar 31 10:08:27 2024 daemon.info dnsmasq[1]: using nameserver 240e:56:4000::218#53
Sun Mar 31 10:08:27 2024 daemon.info dnsmasq[1]: using nameserver 240e:56:4000:8000::69#53
Sun Mar 31 10:08:27 2024 daemon.info dnsmasq[1]: using nameserver 240e:56:4000::218#53
Sun Mar 31 10:08:27 2024 daemon.info dnsmasq[1]: using nameserver 61.139.2.69#53
Sun Mar 31 10:08:27 2024 daemon.info dnsmasq[1]: using nameserver 218.6.200.139#53
Sun Mar 31 10:08:27 2024 daemon.info dnsmasq[1]: using nameserver 61.139.2.69#53
Sun Mar 31 10:08:27 2024 daemon.info dnsmasq[1]: using nameserver 218.6.200.139#53
Sun Mar 31 10:08:27 2024 daemon.info dnsmasq[1]: using nameserver 61.139.2.69#53
Sun Mar 31 10:08:27 2024 daemon.info dnsmasq[1]: using nameserver 218.6.200.139#53
Sun Mar 31 10:08:27 2024 daemon.info dnsmasq[1]: using only locally-known addresses for test
Sun Mar 31 10:08:27 2024 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Sun Mar 31 10:08:27 2024 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Sun Mar 31 10:08:27 2024 daemon.info dnsmasq[1]: using only locally-known addresses for local
Sun Mar 31 10:08:27 2024 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Sun Mar 31 10:08:27 2024 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Sun Mar 31 10:08:27 2024 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Sun Mar 31 10:08:27 2024 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Sun Mar 31 10:08:27 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 10 names
Sun Mar 31 10:08:27 2024 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 24 names
Sun Mar 31 10:08:27 2024 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Sun Mar 31 10:08:29 2024 user.info mwan3track[6004]: Check (ping) failed for target "www.baidu.com" on interface vwan2 (pppoe-vwan2). Current score: 5
Sun Mar 31 10:08:32 2024 user.info mwan3track[6004]: Check (ping) failed for target "114.114.114.114" on interface vwan2 (pppoe-vwan2). Current score: 5
Sun Mar 31 10:08:34 2024 daemon.err tailscaled[10311]: 2024/03/31 02:08:34 monitor: RTM_NEWROUTE: src=, dst=, gw=, outif=1, table=100
Sun Mar 31 10:08:34 2024 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Sun Mar 31 10:08:37 2024 user.info mwan3track[6003]: Check (ping) failed for target "www.baidu.com" on interface vwan1 (pppoe-vwan1). Current score: 5
Sun Mar 31 10:08:37 2024 daemon.info dnsmasq[1]: started, version 2.89 cachesize 1000
Sun Mar 31 10:08:37 2024 daemon.info dnsmasq[1]: DNS service limited to local subnets
Sun Mar 31 10:08:37 2024 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset no-nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Sun Mar 31 10:08:37 2024 daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Sun Mar 31 10:08:37 2024 daemon.info dnsmasq-dhcp[1]: DHCP, IP range 192.168.10.100 -- 192.168.10.249, lease time 12h
Sun Mar 31 10:08:37 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain 91smartyun.pt
Sun Mar 31 10:08:37 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain adobe.com
Sun Mar 31 10:08:37 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain amazonaws.com
Sun Mar 31 10:08:37 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain ampproject.org
Sun Mar 31 10:08:37 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain apple.news
Sun Mar 31 10:08:37 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain aws.amazon.com
Sun Mar 31 10:08:37 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain azureedge.net
Sun Mar 31 10:08:37 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain backpackers.com.tw
Sun Mar 31 10:08:37 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain bitfinex.com
Sun Mar 31 10:08:37 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain buzzfeed.com
Sun Mar 31 10:08:37 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain clockwise.ee
Sun Mar 31 10:08:37 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain cloudfront.net
Sun Mar 31 10:08:37 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain coindesk.com
Sun Mar 31 10:08:37 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain coinsquare.io
Sun Mar 31 10:08:37 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain cryptocompare.com
Sun Mar 31 10:08:37 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain dropboxstatic.com
Sun Mar 31 10:08:37 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain eurecom.fr
Sun Mar 31 10:08:37 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain gdax.com
Sun Mar 31 10:08:37 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain github.com
Sun Mar 31 10:08:37 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain kknews.cc
Sun Mar 31 10:08:37 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain nutaq.com
Sun Mar 31 10:08:37 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain openairinterface.org
Sun Mar 31 10:08:37 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5335 for domain skype.com
[... cut ...]
[   59.306630] overlayfs: workdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
[   59.306715] overlayfs: lowerdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
[   59.306797] overlayfs: lowerdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
[  335.209828] pppoe-wan: renamed from ppp1
[ 5832.226448] TCP: request_sock_TCP: Possible SYN flooding on port 81. Sending cookies.  Check SNMP counters.

yanjinsongyyq, 2024-04-02:
我是关闭tailscale 的magic DNS

wldxx888, 2024-04-03:
> 我正在关闭tailscale的神奇DNS

wldxx888, 2024-04-03:
> 我正在关闭 tailscale 的神奇 DNS

今天试了一下，有效

closed 2024-04-15 by jjm2473