istoreos/istoreos #1470: 最新版本装上后ping不通内网设备

shanhai2333, 2024-04-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：istoreos  ping不通内网设备，内网设备都有网，能ping通路由器，升级发现的，以为是设置防火墙出错了，今天重装，在只配置wan拨号和lan的情况下还是ping不通

### (2) 路由器型号和固件版本

A：J4125   iStoreOS 22.03.6 2024031514

### (3) 详细日志和/或截图

A：电脑ping Nas
[image]

istoreos ping Nas ，一直卡住
[image]

jjm2473, 2024-04-07:
你在lan口配置了什么？ip地址和掩码分别是什么？截图看看

shanhai2333, 2024-04-09:
> 你看看lan口配置是什么？ip地址和掩码分别是什么？截图

[image]
就改了ipv4地址其他都没改动

138161, 2024-04-11:
网络-lan-设置里面，网关--默认，因为改了ipv4地址，网关就不是原来的，比如，原来是192.168.1.x，网关就是192.168.1.1，改成192.168.3.x，网关也要改为192.168.3.1

shanhai2333, 2024-04-11:
> 网络-lan-设置里面，网关--默认，因为改了ipv4地址，网关就不是原来的，比如，原来是192.168.1.x，网关就是192.168.1.1，改成192.168.3.x，网关也要改为192.168.3.1

[image]
是默认的，其他都没改动，这个ipv4网关的地址也是自动的，我输入框点击，光标都在最左边
[image]

jjm2473, 2024-04-12:
lan口不需要配置网关，除非是旁路由，不要听其他人瞎说。

jjm2473, 2024-04-12:
@shanhai2333 那只能抓包看看了。现在只是ping nas不通？还是ping自己电脑也不通？如果ping电脑也不通，那就很方便抓包了。

shanhai2333, 2024-04-12:
> @shanhai2333 那只能抓包看看了。现在只是ping nas不通？还是ping自己电脑也不通？如果ping电脑也不通，那就很方便抓包了。

ping内网里的设备都ping不通，内网里面的设备互相能ping通

closed 2024-04-16 by shanhai2333

jjm2473, 2024-04-17:
@shanhai2333 找到原因了？

shanhai2333, 2024-05-26:
> @shanhai2333 找到原因了？
全新重装一次就行了，以为是偶然原因就没管了，今天保存配置升级最新又出现了

jjm2473, 2024-06-04:
是不是翻墙插件导致的，关掉插件，重启系统，看看还有没有问题

shanhai2333, 2024-06-06:
> 是不是翻墙插件导致的，关掉插件，重启系统，看看还有没有问题

还是一样的，用的passwall，关掉后重启还是ping不通

shanhai2333, 2024-06-11:
> 是不是翻墙插件导致的，关掉插件，重启系统，看看还有没有问题

找到问题了，之前uu插件找不到ps5，网上搜了个解决方法，防火墙里添加：”iptables -t nat -A POSTROUTING -o br-lan -j SNAT --to 192.168.13.198”。今天重装机器，慢慢试，试到这一步出现的，注释掉就能ping了