istoreos/istoreos #1473: zerotier运行提示connect: operation timed out

CarterXiang, 2024-04-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：家庭网络使用的移动宽带，istoreos安装在PVE虚拟机中，作为**旁路由**使用。istoreos中通过软件源安装了zerotier后，在terminal中启动zerotier-one，提示`connect: operation timed out`错误，此时9993端口未被任何其他进程占用。

另外家里连接istoreos的windows设备和linux设备上的zerotier客户端却可以正常使用，所以感觉不像是被运营商墙了的原因。

### (2) 路由器型号和固件版本

A： iStoreOS 22.03.6 2024031514

### (3) 详细日志和/或截图

A：
```
Starting Control Plane...
Starting V6 Control Plane...
connect: Operation timed out
connect: Operation timed out
```

jjm2473, 2024-04-07:
istoreos首页的网络状态显示什么？dns是什么

CarterXiang, 2024-04-07:
> istoreos首页的网络状态显示什么？dns是什么

网络状态正常的，没有警告和错误，istore及其连接的设备都可以正常访问外网
DNS配置的是路由器的地址

jjm2473, 2024-04-08:
其他zerotier客户端使用的是同一个DNS吗？

CarterXiang, 2024-04-08:
> 其他zerotier客户端使用的是同一个DNS吗？

其它zerotier所在机器的DNS用的istore的ip