istoreos/istoreos #1474: 调制解调器管理器协议qmi拨号断开

h88kmax, 2024-04-06:
使用调制解调器协议rm520ngl用的是qmi拨号模式，联通卡联网后15秒左右就断网，电信卡正常！联通卡切换mbim模式可以正常使用！
联通卡使用qmi模式无法上网

jjm2473, 2024-04-07:
istore里面有个新的5G拨号插件，把网络-接口里面的配置删掉，用这个新插件试试

h88kmax, 2024-04-07:
试过了！不行！我群里用h69k的也是一样联通卡qmi不行切mbim正常！你说的是商店的移动通信模组！也不行！电信卡正常

h88kmax, 2024-04-07:
> istore里面有个新的5G拨号插件，把网络-接口里面的配置删掉，用这个新插件试试

我那天是电信卡正常使用！联通卡到了换上去就不行！刚开始以为是填了APN问题！恢复出厂设置不填一样有网10秒左右就断开！然后用移动模组的，新建dhcp和dhcpv6协议 一样没有网！后来切mbim用调制解调器管理器就正常了！

jjm2473, 2024-04-07:
反正是模块的问题，如果mbim正常就用mbim。在我看来，这些模块的固件都是垃圾，厂商全都不按标准来做，可能也没什么标准。

jjm2473, 2024-04-07:
如果是FM650，h69k在开机时会自动把模块切换到mbim模式，这也是经过很多次测试才确认mbim模式可用的。至于其他模块，我没有，也懒得适配自动切换模式。

mbim是linux的标准协议，如果有问题，那就找厂商，毕竟连标准都没法遵循，只能说明厂商软件垃圾。

h88kmax, 2024-04-10:
> 如果是FM650，h69k在开机时会自动把模块切换到mbim模式，这也是经过很多次测试才确认mbim模式可用的。至于其他模块，我没有，也懒得适配自动切换模式。
> 
> mbim是linux的标准协议，如果有问题，那就找厂商，毕竟连标准都没法遵循，只能说明厂商软件垃圾。

今天降级到22.03.5 2023122916 qmi拨号正常了！看来和模块没有关系 是你的22.03.6新内核的问题啦

jjm2473, 2024-04-12:
22.03.6 跟 22.03.5 的内核就差一点小版本，而且 5G 模块相关的驱动不是rk35xx特有的，是linux原版就有，如果真有bug，那openwrt 22.03.6 和 linux 相应版本也一样。

h88kmax, 2024-04-18:
[modem0] state changed (connected -> registered)
Thu Apr 18 19:48:50 2024 daemon.info [10720]: <info>  [modem0/bearer0] connection #1 finished: duration 1741s, tx: 15261888 bytes, rx: 574788696 bytes
Thu Apr 18 19:50:43 2024 daemon.notice ddnsto[24512]: main accept error count=0 wait_cnt=0
Thu Apr 18 19:50:48 2024 daemon.notice ddnsto[13400]: check reboot failed, ret=-1
Thu Apr 18 19:51:41 2024 daemon.notice ddnsto[11459]: dial ipv4 failed ret=-102 retry=1 remote=tunnel.ddnsto.com:4445
Thu Apr 18 19:52:00 2024 daemon.notice netifd: wwan (12431): stopping network
Thu Apr 18 19:52:00 2024 daemon.notice netifd: wwan (12431): running disconnection (common)
Thu Apr 18 19:52:00 2024 daemon.notice netifd: wwan (12431): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "wwan" } (Permission denied)
Thu Apr 18 19:52:00 2024 daemon.notice netifd: wwan (12431): successfully disconnected all bearers in the modem
Thu Apr 18 19:52:00 2024 daemon.info [10720]: <info>  [modem0] state changed (registered -> disabling)
Thu Apr 18 19:52:00 2024 daemon.info [10720]: <info>  [modem0] 3GPP registration state changed (home -> unknown)
Thu Apr 18 19:52:00 2024 daemon.info [10720]: <info>  [modem0] state changed (disabling -> disabled)
Thu Apr 18 19:52:00 2024 daemon.notice netifd: wwan (12431): successfully disabled the modem
Thu Apr 18 19:52:00 2024 daemon.notice netifd: Interface 'wwan' is now down
Thu Apr 18 19:52:00 2024 daemon.notice ttyd[25014]: rops_handle_POLLIN_netlink: DELADDR
Thu Apr 18 19:52:00 2024 daemon.info avahi-daemon[10617]: Withdrawing address record for 10.31.73.122 on wwan0.
Thu Apr 18 19:52:00 2024 daemon.info avahi-daemon[10617]: Leaving mDNS multicast group on interface wwan0.IPv4 with address 10.31.73.122.
Thu Apr 18 19:52:00 2024 daemon.info avahi-daemon[10617]: Interface wwan0.IPv4 no longer relevant for mDNS.
Thu Apr 18 19:52:00 2024 daemon.notice ttyd[25014]: rops_handle_POLLIN_netlink: DELADDR
Thu Apr 18 19:52:00 2024 daemon.info avahi-daemon[10617]: Withdrawing address record for 2408:844f:d00:6d87:7d48:cd4a:c197:cf70 on wwan0.
Thu Apr 18 19:52:00 2024 daemon.info avahi-daemon[10617]: Leaving mDNS multicast group on interface wwan0.IPv6 with address 2408:844f:d00:6d87:7d48:cd4a:c197:cf70.
Thu Apr 18 19:52:00 2024 daemon.info avahi-daemon[10617]: Joining mDNS multicast group on interface wwan0.IPv6 with address fe80::1453:46ff:fe99:fa3b.
Thu Apr 18 19:52:00 2024 daemon.info avahi-daemon[10617]: Registering new address record for fe80::1453:46ff:fe99:fa3b on wwan0.*.
Thu Apr 18 19:52:00 2024 daemon.info avahi-daemon[10617]: Registering new address record for fe80::62be:b4ff:fe15:5168 on br-lan.*.
Thu Apr 18 19:52:00 2024 daemon.info avahi-daemon[10617]: Interface wwan0.IPv6 no longer relevant for mDNS.
Thu Apr 18 19:52:00 2024 daemon.info avahi-daemon[10617]: Leaving mDNS multicast group on interface wwan0.IPv6 with address fe80::1453:46ff:fe99:fa3b.
Thu Apr 18 19:52:00 2024 daemon.notice ttyd[25014]: rops_handle_POLLIN_netlink: DELADDR
Thu Apr 18 19:52:00 2024 daemon.notice netifd: Interface 'wwan' is setting up now
Thu Apr 18 19:52:00 2024 daemon.info avahi-daemon[10617]: Withdrawing address record for fe80::1453:46ff:fe99:fa3b on wwan0.
Thu Apr 18 19:52:00 2024 daemon.warn dnsmasq[1]: no servers found in /tmp/resolv.conf.d/resolv.conf.auto, will retry
Thu Apr 18 19:52:00 2024 daemon.notice netifd: wwan (12642): modem available at /org/freedesktop/ModemManager1/Modem/0
Thu Apr 18 19:52:00 2024 daemon.notice netifd: wwan (12642): starting connection with apn ''...
Thu Apr 18 19:52:00 2024 daemon.info [10720]: <info>  [modem0] simple connect started...
Thu Apr 18 19:52:00 2024 daemon.info [10720]: <info>  [modem0] simple connect state (3/8): enable
Thu Apr 18 19:52:00 2024 daemon.info [10720]: <info>  [modem0] state changed (disabled -> enabling)
Thu Apr 18 19:52:00 2024 daemon.info [10720]: <info>  [modem0] power state updated: on
Thu Apr 18 19:52:00 2024 daemon.info [10720]: <info>  [modem0] simple connect state (4/8): wait to get fully enabled
Thu Apr 18 19:52:00 2024 daemon.info [10720]: <info>  [modem0] state changed (enabling -> enabled)
Thu Apr 18 19:52:00 2024 daemon.info [10720]: <info>  [modem0] simple connect state (5/8): register
Thu Apr 18 19:52:00 2024 daemon.info [10720]: <info>  [modem0] 3GPP registration state changed (unknown -> registering)
Thu Apr 18 19:52:00 2024 daemon.info [10720]: <info>  [modem0] 3GPP registration state changed (registering -> home)
Thu Apr 18 19:52:00 2024 daemon.info [10720]: <info>  [modem0] state changed (enabled -> registered)
Thu Apr 18 19:52:00 2024 daemon.info [10720]: <info>  [modem0] simple connect state (6/8): bearer
Thu Apr 18 19:52:00 2024 daemon.info [10720]: <info>  [modem0] simple connect state (7/8): connect
Thu Apr 18 19:52:00 2024 daemon.info [10720]: <info>  [modem0] state changed (registered -> connecting)
Thu Apr 18 19:52:01 2024 daemon.info [10720]: <info>  [modem0] state changed (connecting -> connected)
Thu Apr 18 19:52:01 2024 daemon.info [10720]: <info>  [modem0] simple connect state (8/8): all done
Thu Apr 18 19:52:01 2024 daemon.notice netifd: wwan (12642): successfully connected the modem
Thu Apr 18 19:52:01 2024 daemon.notice netifd: wwan (12642): signal refresh rate is not set
Thu Apr 18 19:52:01 2024 daemon.notice netifd: wwan (12642): network operator name: CHN-UNICOM
[... cut ...]
Thu Apr 18 19:52:02 2024 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: dns.msftncsi.com
Thu Apr 18 19:52:03 2024 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: dns.msftncsi.com
Thu Apr 18 19:52:07 2024 daemon.notice ddnsto[11459]: client init ok, username=15d11147-6486-4d1c-8b85-ba269b1373b5 routerId=60beb4155167 version=3.0.4
Thu Apr 18 19:52:08 2024 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: dns.msftncsi.com
Thu Apr 18 19:52:23 2024 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: dns.msftncsi.com

h88kmax, 2024-04-18:
> 22.03.6 跟 22.03.5 的内核就差一点小版本，而且 5G 模块相关的驱动不是rk35xx特有的，是linux原版就有，如果真有bug，那openwrt 22.03.6 和 linux 相应版本也一样。

上面的是用了电信卡后 再用联通卡就掉线！