istoreos/istoreos #1475: PD虚拟机跑 isoreos 提示 interface not found

zhEdward, 2024-04-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：用来做测试，虚拟机跑码完通过 quickstart ，修改ip，虚拟机上网络设置 桥接Thunderbolt Ethernet，然后重启，192.168.6.99 无法进入web。结合标题来看感觉是 网口 没设置还是没识别到

PS：之前操作安装 LEDE固件 以及官方原版固件也是一样步骤都没问题

### (2) 路由器型号和固件版本

A：PD虚拟机，istoreos-22.03.6-2024030112-x86-64-squashfs-combined

### (3) 详细日志和/或截图

A：
[image]

[image]

jjm2473, 2024-04-07:
执行`ip link`看看，以及`lspci -nnk`

zhEdward, 2024-04-07:
由于没有ip，无法用scp，只能截屏 
```ip link```如下：
[image]
[image]

```lscpci -nnk``` 想输出txt，但报错 **lspci unable to load libkmod resources error 12**无法查看完整，然后没有滚动条，只能截图后半部分，如下：
[image]

jjm2473, 2024-04-08:
你的网卡是不是在系统第一次启动完成以后才插上去的？也就是说，哪个磁盘初次启动时是没有网卡的，不管你是虚拟网卡还是什么。

算了，直接执行`firstboot -y -r`，恢复出厂设置，让系统重新配置一遍。

closed 2024-04-08 by jjm2473

zhEdward, 2024-04-08:
> 你的网卡是不是在系统第一次启动完成以后才插上去的？也就是说，哪个磁盘初次启动时是没有网卡的，不管你是虚拟网卡还是什么。
> 
> 算了，直接执行`firstboot -y -r`，恢复出厂设置，让系统重新配置一遍。

确实。用vmc文件创建的虚拟机，导入就自动跑码，跑完我关闭才在PD给它桥接网络。恢复出厂，现已正常。感谢

persiliao, 2024-08-05:
@zhEdward 朋友，你用PD跑的iStore旁路由设备连上主路由WIFI能上网吗？