istoreos/istoreos #1476: UPnP条目变化会导致网络卡(延迟几千)一会

SwordRays, 2024-04-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：UPnP条目变化会导致网络卡一会，10~40秒左右，具体表现为延迟会变为几千毫秒。

### (2) 路由器型号和固件版本

A：FastRhino R66s + iStoreOS 22.03.6 2024031514

### (3) 详细日志和/或截图

A：[image]
可以通过如BitCom比特彗星、Syncthing之类会使用UPnP的软件复现该问题，重启比特彗星，等到右下角显示检测端口时，百分百触发路由器整个网络变卡，持续大约10~40秒左右恢复正常。

jjm2473, 2024-04-07:
ping ipv4地址也会这样吗？

SwordRays, 2024-04-07:
> ping ipv4地址也会这样吗？

IPv4也会的。
虽然没截图，我还直接ping了223.5.5.5，也是一样的，只要UPnP出现新条目，必然会导致整个网络卡一会。

jjm2473, 2024-04-07:
你确定不是因为你的这些软件占用大量网络带宽导致的？

upnp修改条目的时候，也就是添加防火墙规则而已，这能对网络产生什么影响。你看看upnp产生的条目数量有没有异常，例如有没有产生几百上千个条目。

SwordRays, 2024-04-07:
> 你确定不是因为你的这些软件占用大量网络带宽导致的？
> 
> upnp修改条目的时候，也就是添加防火墙规则而已，这能对网络产生什么影响。你看看upnp产生的条目数量有没有异常，例如有没有产生几百上千个条目。

UPnP条目多少没有影响的，哪怕UPnP列表是空的，然后打开比特彗星，它只增加一两条规则，都会出现这个现象。

至于网络带宽，我这边是千兆下行，百兆上行。测试UPnP时上下行都只有几百kbps的流量，基本不可能是流量太大影响的。

PS1：个人推测，问题可能出在UPnP条目变化后防火墙规则变化的逻辑方面……

jjm2473, 2024-04-07:
那你试试在防火墙页面随便添加个规则，看看能不能触发相同的现象

SwordRays, 2024-04-07:
> 那你试试在防火墙页面随便添加个规则，看看能不能触发相同的现象

刚刚把路由器恢复出厂设置了。然后只配以下东西：
1、修改了后台密码
2、设置了pppoe拨号
3、修改了LAN口网段为192.168.xx.1
其他啥都没动，还是会有一样的现象。
防火墙也看了，加规则，重启防火墙啥的都不影响网络延迟。

jjm2473, 2024-04-08:
那估计跟upnp没啥关系，可能是广播包会导致什么异常。你在电脑抓包看看出问题时有没有大量的重复报文。甚至你都可以把软路由关机来测试：电脑在获取ip以后，把软路由关机，然后ping内网其他设备，然后启动那些能触发异常的软件，看看ping值会不会有大波动。

SwordRays, 2024-04-09:
> 那估计跟upnp没啥关系，可能是广播包会导致什么异常。你在电脑抓包看看出问题时有没有大量的重复报文。甚至你都可以把软路由关机来测试：电脑在获取ip以后，把软路由关机，然后ping内网其他设备，然后启动那些能触发异常的软件，看看ping值会不会有大波动。

昨天又研究了一下，发现 网络 - 防火墙 - 常规设置 - 启用 FullCone-NAT 这个选项。
改成“高性能模式”就没有这个UPnP导致网络卡和CPU满载的问题，使用默认的“兼容模式”则能百分百复现。

jjm2473, 2024-04-12:
FullCone-NAT确实会查找conntrack表，但是除非你的系统创建了十几万个以上连接，不然不会影响性能。而且这跟upnp和防火墙规则都没任何关系，upnp只是让创建连接变得容易了而已。

SwordRays, 2024-04-12:
嗯，具体原理就不知道了，我只是把研究出的现象反馈出来。
对我来说，现在用高性能模式能解决问题就足够了~
至于兼容模式下的问题是不是bug，和能不能修复，就不是我能决定的啦。

jjm2473, 2024-04-12:
@SwordRays 后台的"状态"-"概览"里面可以看到连接数，看看是不是数量很大。

SwordRays, 2024-04-12:
现在高性能模式下只有3500多
等晚点测试了兼容模式后继续反馈

jjm2473, 2024-04-12:
OK，3500倒是不多

jjm2473, 2024-04-12:
哦，出问题时还可以看看内核日志，有没有“dropping packets”相关日志

SwordRays, 2024-04-12:
好的

SwordRays, 2024-04-12:
> 哦，出问题时还可以看看内核日志，有没有“dropping packets”相关日志

测试了下，
a) 使用兼容模式时：
首先，运行UPnP软件后，活动连接数从3000~6000左右快速增加到34000左右，过程约1~2分钟。在此期间可观测到：路由器CPU占用率80%以上接近满载；到阿里dns(223.5.5.5)的网络延迟增加到几千，甚至个别丢包。
然后，当达到34000左右的顶峰后，连接数开始下降，过程约2分钟。在此期间可观测到：开始下降时CPU占用立即恢复正常，网络延迟也立即恢复正常。
最终，连接数彻底降低到几千。

b) 使用高性能模式时，：
首先，运行UPnP软件后，活动连接数从3000~6000左右快速增加到64500左右，过程约1~2分钟。在此期间可观测到：路由器CPU占用率没有太大变化始终低于25%；到阿里dns的网络延迟有个别包从5ms增加不超过10ms，是正常波动。
然后，当达到64500左右的顶峰后，连接数开始下降，过程约2分钟。在此期间可观测到：CPU占用依然没有变化，网络延迟也没有变化。
最终，连接数彻底降低到几千。

PS：系统日志中并没有相关内容，内核日志中在此时前就有6段重复这样的2条日志：
[233348.251410] nf_ct_snmp_trap: dropping packet: parser failed
[233348.251410]  IN= OUT= SRC=x.x.x.x DST=y.y.y.y LEN=126 TOS=0x00 PREC=0x00 TTL=127 ID=3289 PROTO=UDP SPT=9384 DPT=162 LEN=106
但个人觉得应该跟测试无关。在几次切换模式的测试过程，该日志不会增加，始终只有之前的6段。

jjm2473, 2024-04-13:
OK，那其实就是连接数太多，“兼容模式”的Fullcone-NAT查表性能低，造成CPU飙升，进而导致丢包。“高性能模式”的Fullcone-NAT效率比较高就不会因为连接数多而导致性能下降严重。

jjm2473, 2024-04-13:
不过，你的软件为什么一打开就会创建几万个连接呢，如果是PT下载，一般也就下载或者做种的过程中需要连接其他peer，启动时创建那么多连接是干嘛

SwordRays, 2024-04-13:
> 不过，你的软件为什么一打开就会创建几万个连接呢，如果是PT下载，一般也就下载或者做种的过程中需要连接其他peer，启动时创建那么多连接是干嘛

不知道什么原因产生的连接数，测试的时候只是简单的开关 比特彗星 这个下载软件，基本没有其他使用UPnP的软件。
另外，UPnP列表中的条目始终不会超过20条，无论哪个模式都一样。

jjm2473, 2024-04-15:
嗯，连接数跟upnp条目数没必然联系。一条upnp条目就是开放一个端口，但是这个端口可以连接非常多peer。

gang1974, 2024-05-21:
我也碰到类似的情况，不过我是挂网心云，谢谢SwordRays提供的折中解决方法。

xy3xy3, 2024-08-10:
学习了，试试看，之前我也是跑网心云高丢包

jjm2473, 2024-08-13:
> 兼容模式和高性能模式的区别是什么

实现方式不一样，对一般人来说两个应该都可以。