istoreos/istoreos #1480: 上海电信IPV6光猫拨号ipv6中继方式，有ipv6地址，ping不通的临时解决方法

138161, 2024-04-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：wan和lan配置中6个全选中继，可以获得ipv6地址，但ipv6测试不过，ping -6 www.baidu,com不通，过段时间ipv6地址失效，又不通了

### (2) 路由器型号和固件版本

A：iStoreOS  [Powered by LuCI istoreos-22.03 branch (git-24.057.30647-6659faa)

### (3) 详细日志和/或截图

A：
解决的临时方法
依赖 owipcalc 包来计算子网地址： opkg install owipcalc 。
我们可以手动在 WAN 口获得 IPv6 地址后添加一条路由表，让整个子网重定向到 LAN 口，这个操作可以通过 OpenWrt 的 hotplug 机制来进行，保存以下脚本放在 /etc/hotplug.d/iface/80-reset-route6 并重启 WAN 接口即可：

#!/bin/sh

wan_dev="wan6"

[ "$HOTPLUG_TYPE" = "iface" ] || exit 0
[ "$INTERFACE" = "$wan_dev" ] || exit 0

RTMETRIC=127

. /lib/functions/network.sh

network_get_physdev lan_dev lan || exit 0

ifup_cb() {
        local _lan_dev="$1"
        local _metric="$2"

        local wan_subnet
        network_get_subnet6 wan_subnet "$wan_dev" || return
        _wan_network=$(owipcalc "${wan_subnet}" network)

        ip -6 route replace "$_wan_network" dev "$_lan_dev" metric "$_metric"
}

ifdown_cb() {
        local _lan_dev="$1"
        local _metric="$2"

        ip -6 route flush dev "$_lan_dev" metric "$_metric"
}

case "$ACTION" in
        ifup)
                ifup_cb "$lan_dev" "$RTMETRIC"
                ;;
        ifdown)
                ifdown_cb "$lan_dev" "$RTMETRIC"
                ;;
        ifupdate)
                ifdown_cb "$lan_dev" "$RTMETRIC"
                sleep 1
                ifup_cb "$lan_dev" "$RTMETRIC"
                ;;
        *)
                ;;
esac

exit 0

第零步，安装软件包
0.1 系统--软件包--更新列表
      过滤器输入：owipcalc
     安装
0.2 编辑 80-reset-route6文件，，内容见上面，拷贝到 /etc/hotplug.d/iface/80-reset-route6
0.3 赋予执行权限 777
chmod +X /etc/hotplug.d/iface/80-reset-route6

第一步，先搞定wan6口
1、网络--接口--wan6--编辑
1.1、常规设置
     协议：DHCPv6客户端
         设备：eth0
         开机自动运行：选择
         请求IPv6地址：try
         请求指定长度的IPv6前缀：自动
       
1.2、高级设置
         强制链路：不选
         请求DHCP时发送的客户端ID：空
         使用默认网关：选择
         自动获取DNS服务器：选择
         DNS 权重：空
     使用网关跃点：空
     覆盖 IPv4 路由表：未指定
     覆盖 IPv6 路由表：未指定
     IPv6 源路由：选择
     委托 IPv6 前缀：选择
     IPv6 分配长度：已禁用
     IPv6 前缀过滤器：-- 请选择 --
     IPv6 后缀：::1
     IPv6 优先级：空
     
                          
1.3、防火墙设置
     创建和分配防火墙区域：wan和wan6
     
1.4、DHCP服务器--常规设置
     忽略此接口：不选
     
1.5、DHCP服务器--IPv6设置
     指定的主接口：选择
     RA 服务：中继模式
     DHCPv6 服务：中继模式
     NDP 代理：中继模式
     学习路由：选择
     

第二步，搞定lan口
2、网络--接口--lan--编辑
2.1、常规设置
     协议：静态地址
     设备：br-lan
     开机自动运行：选择
     IPv4 地址：192.168.100.1
     IPv4 子网掩码：255.255.255.0
     IPv4 网关：灰色无法更改
     IPv4 广播地址：灰色无法更改
     IPv6 地址：空
     IPv6 网关：空
     IPv6 路由前缀：空
     
2.2、高级设置
     强制链路：选择
     使用默认网关：不选择
     使用自定义的 DNS 服务器：空
     DNS 搜索域名：空
     DNS 权重：空
     使用网关跃点：空
     覆盖 IPv4 路由表：未指定
     覆盖 IPv6 路由表：未指定
     委托 IPv6 前缀：选择
     IPv6 分配长度：禁用
     IPv6 前缀过滤器：-- 请选择 --
     IPv6 后缀：::1
     IPv6 优先级：空
     
                 
2.3、防火墙设置
     创建和分配防火墙区域：lan
     
2.4、DHCP服务器--常规设置
     忽略此接口：不选
     启动：100
     客户数：150
     租期：infinite
     
2.5、DHCP服务器--高级设置
     动态 DHCP：选择
     强制：不选
     IPv4 子网掩码：灰色无法更改
     DHCP 选项：空
     
2.6、DHCP服务器--IPv6设置
     指定的主接口：不选
     RA 服务：中继模式
     DHCPv6 服务：中继模式
     NDP 代理：中继模式
     学习路由：选择
     NDP 代理从属设备：不选
     

第三步，网络--接口--全局网络选项
     IPv6 ULA 前缀：空
     数据包引导：选择
     

第四步，网络--DHCP/DNS--高级设置
     过滤 IPv6 AAAA 记录：不选
     过滤 IPv4 A 记录：不选
-----------------------------------------------

软路由重启（系统--重启）

IPV6联网测试
https://testipv6.cn/
ping -6 www.baidu.com

第六步，其他问题
6.1A、如果获得了ipv6地址，缺无法ping通，大概率是以下三个地方
网络--接口--全局网络选项，IPv6 ULA 前缀：空
网络--接口--wan6，IPv6 分配长度：已禁用
网络--接口--lan，IPv6 分配长度：已禁用
原因是纯 relay 模式下客户端不会获取到“IPv6 ULA 前缀”下的地址/路由/邻居信息，当客户端尝试回复从路由器发出的 ULA-prefix 下地址的 neighbor solicitation 时，匹配不到这个 ULA 地址的路由只好放弃。

6.1B 参考此文章，主要是NDP消息的更新机制
odhcpd 中继模式原理、局限以及解决方案
https://blog.icpz.dev/articles/notes/odhcpd-relay-mode-discuss/

6.2、软路由iStore，22.03.6版的防火墙配置，是一个通用配置，不适合国内的IPv6环境，icmpv6的消息被防火墙阻挡，需要根据情况打开。待查
6.3、网络拓扑：电信光猫拨号-->OpenWrt X86电脑（iStore）-->AP。
软件是 iStoreOS 22.03.6 2024012613。3150主板，两个网口，日常CPU负载20%以下，8G内存剩下7G以上空闲。
6.4、iStoreOS 页面设置在网络防火墙一节似乎没有很完美，明明页面设置了，后台文件没变化，经查，zone域规则不能自行修改，要加在用户规则内

6.5、两个网关问题
        iStore首页--终端--输入root和密码，查询路由器的默认 IPv6 网关
        ip -6 route | grep default 得到：
        default from 240e:38a:xxx:xxx::/64 via fe80::133:7cc9:xxx dev eth0 proto static metric 512 pref medium  #外网到光猫lan侧
        default from 240e:38a:xxx:xxx::/64 via fe80::1 dev eth0 proto static metric 640 pref medium
        出现此情况，在本模式（中继SLAAC）下忽略，在IPV6 NAT模式下需要修改若干内容

6.5、openwrt 需要的 ipv6 包：
        odhcp6c                  2022-08-05-7d21e8d8-18，DHCPv6 客户端
        odhcpd-ipv6only      2023-01-02-4a673e1c-2， ipv6版本
         luci-proto-ipv6        git-21.148.48881-79947af    git-23.355.78888-e047387，从 luci Web 界面配置 IPv6
         ip6tables-mod-nat      1.8.7-7
         ip6tables-zz-legacy    1.8.7-7
         kmod-ip6tables           5.10.201-1  并非必须，提供IPv6防火墙
         kmod-nf-ipt6               5.10.201-1
         kmod-ipt-nat6            提供IPv6的NAT支持
        netifd                          OpenWrt Network Interface Configuration Daemon
         owipcalc                    计算子网地址 

6.6、文件 /etc/config/dhcp 如下
config dhcp 'lan'
        option interface 'lan' #端口名称
        option start '100'    #端口号起始--结束
        option limit '150'
        option dhcpv4 'server'
        option leasetime '12h'  # infinite 租期，v4v6均有效
        option ra 'relay'
        option ndp 'relay'        #中继运行模式
        option dhcpv6 'relay'

config dhcp 'wan'
        option interface 'wan'
        option ignore '1'

config dhcp 'wan6'
        option interface 'wan6'
        option master '1'
        option ra 'relay'
        option ndp 'relay'
        option dhcpv6 'relay'

------------------------------------------------------

6.7、passwall插件，DNS设置，过滤代理域名 IPv6，要勾选，不对IPV6解析，使得IPV6直接连通，不经过passwall。

138161, 2024-04-08:
2.2、高级设置
使用默认网关：选择
好像对插件的兼容更好

138161, 2024-04-09:
经过测试，防火墙还需要增加一条 lan 到 wan 的规则
入站  UDP 来自 lan 端口 546
到 此设备 端口  547

138161, 2024-04-09:
这下稳如老狗了，分析原因是电信部门只管光猫下直接的设备，不管二级路由器下的设备，于是，他们就用绑定代替分配地址，现象是没有ipv6 PD前缀，只有一个ipv6地址（二级路由器wan侧），而软路由的的dhcp6C能提供中继功能，但如何维护路由器的路由表和处理DHCP到期续约，人家没给出，所以就需要处理这两方面的内容，首先是wan和lan口的配置，涉及ULA问题，其次是路由表的维护，告诉路由器，下面的PC在哪里，最后是放行下面PC的续约申请，涉及到PC的546端口到路由器的547端口

138161, 2024-04-10:
# 解决DHCPv6续约的icmpv6通讯，lan侧到路由器本身。此消息经过中继后再传到上级光猫。
#  过程：Solicit---Advertise---Request---Reply
#  PC端口546，发送请求Solicit至，路由器本身[ff02::1:2]:547端口
#  路由器本身546端口，中继到，光猫547端口
#  光猫547端口，回应一个Advertise给，路由器546端口
#  路由器本身547端口，中继到，PC端口546
#  PC端口546，回应一个Request给路由器547端口
#  路由器本身546端口，中继到，光猫547端口
#  光猫547端口，回应一个Reply给，路由器546端口结束。 
ip6tables -A zone_lan_input -p udp  --sport 546 --dport 547   -j ACCEPT

# 此条配置istore缺省规则已有，解决上级光猫到路由器本身，DHCPv6续约的icmpv6通讯
ip6tables -A zone_wan_input -p udp  --dport 546   -j ACCEPT

138161, 2024-04-11:
[image]

liulangvip, 2024-06-03:
我居然看完了，试试看

huahua-er, 2024-08-29:
感觉isoreos的ipv6有个坑。就是pppoe高级设置里面“获取ipv6地址”的自动手动选项，居然指的是wan_6口的建立方式，手动就是要自己建立wan_6接口。一开始直接快捷拨号，不知道怎么删他建立的wan_6,后来多播的时候自己建的wan_6获取不到ip。
话说我想问自动建立的wan_6是灰色的，是没有在防火墙的wan区域吗？

jjm2473, 2025-04-03:
@huahua-er 不需要管 wan_6 ，这是pppoe拨号以后，自动创建的虚拟的一个ipv6接口，对防火墙来说，实际的接口跟wan一样，不需要管 wan_6 的防火墙配置。