istoreos/istoreos #1481: 通过iStore安装的qBittorrent无法安装/升级搜索插件

Haruhi-Suzumiya, 2024-04-08:
## 通过iStore安装的qBittorrent无法安装/升级搜索插件

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述
### (1) 具体问题

A：
4月6日通过iStore安装了qBittorrent软件，当时能够刷新搜索插件，也可以安装新插件
4月7日上午发现qBittorrent中的搜索插件消失了，点击刷新无效，我尝试了如下方法：
1. 点击安装插件，输入Github提供的插件路径，点击OK无反应
2. 将插件的.py文件放入.../date/nova3/engine文件夹内，重启qBittorrent无效
3. 卸载qBittorrent重新安装并重试上述操作，均无效

4月7日在luci中打开了log
点击“刷新插件”按钮时，log打印了Python版本，并报告几个插件“过时”并开始更新，之后出现“不支持插件xxx”的信息（xxx为几个预设插件的名称，WebUI列表中没有显示这几个插件），详情可以看图片

4月8日上午更新了固件版本到最新（2024年3月15日版本）并重装qBittorrent，无效

### (2) 路由器型号和固件版本

A：
路由器型号： 电犀牛R68S（4G）
固件版本：2022年8月版本（问题发生时）、2024年3月15日版本（现在）

### (3) 详细日志和/或截图

A：点击WebUI“刷新插件”按钮时，log文件的记录：

(N) 2024-04-08T02:32:14 - WebAPI 登录成功。IP：::ffff:117.80.147.234
(I) 2024-04-08T02:32:18 - 检测到 Python，可执行文件名称：'python3'，版本：3.9.16
(I) 2024-04-08T02:32:22 - 插件 "eztv" 已过时，正在更新至 1.14 版本
(I) 2024-04-08T02:32:22 - 插件 "jackett" 已过时，正在更新至 4.0 版本
(I) 2024-04-08T02:32:22 - 插件 "limetorrents" 已过时，正在更新至 4.7 版本
(I) 2024-04-08T02:32:22 - 插件 "piratebay" 已过时，正在更新至 3.2 版本
(I) 2024-04-08T02:32:22 - 插件 "solidtorrents" 已过时，正在更新至 2.1 版本
(I) 2024-04-08T02:32:22 - 插件 "torlock" 已过时，正在更新至 2.22 版本
(I) 2024-04-08T02:32:22 - 插件 "torrentproject" 已过时，正在更新至 1.2 版本
(I) 2024-04-08T02:32:22 - 插件 "torrentscsv" 已过时，正在更新至 1.2 版本
(I) 2024-04-08T02:32:22 - 正在更新 8 插件
(I) 2024-04-08T02:32:22 - 正在更新插件 torrentproject
(I) 2024-04-08T02:32:22 - 正在更新插件 eztv
(I) 2024-04-08T02:32:22 - 正在更新插件 torlock
(I) 2024-04-08T02:32:22 - 正在更新插件 piratebay
(I) 2024-04-08T02:32:22 - 正在更新插件 solidtorrents
(I) 2024-04-08T02:32:22 - 正在更新插件 torrentscsv
(I) 2024-04-08T02:32:22 - 正在更新插件 limetorrents
(I) 2024-04-08T02:32:22 - 正在更新插件 jackett
(I) 2024-04-08T02:32:22 - 不支持插件 torrentproject。
(I) 2024-04-08T02:32:23 - 不支持插件 torlock。
(I) 2024-04-08T02:32:23 - 不支持插件 eztv。
(I) 2024-04-08T02:32:23 - 不支持插件 torrentscsv。
(I) 2024-04-08T02:32:23 - 不支持插件 solidtorrents。
(I) 2024-04-08T02:32:24 - 不支持插件 piratebay。
(I) 2024-04-08T02:32:24 - 不支持插件 limetorrents。
(I) 2024-04-08T02:32:24 - 不支持插件 jackett。

138161, 2024-04-29:
碰到过，没解决，而且运行几天后貌似卡死，重启后流量马上来了，后来改用docker版
qBittorrent-qBittorrentEE
4.6.4-4.6.4.10  (amd64;arm64v8;arm32v7) 集成Trackers自动更新
https://hub.docker.com/r/johngong/qbittorrent

带搜索功能，稳定，不会卡死

138161, 2024-04-29:
https://cf.trackerslist.com/all.txt

docker pull johngong/qbittorrent
docker create \
    --name=qbittorrent \
    -e QB_WEBUI_PORT=8989 \
    -e QB_EE_BIN=true \
    -e UID=1000 \
    -e GID=1000 \
    -e UMASK=022 \
    -e TZ=Asia/Shanghai
    --net=host \
    -e QB_TRACKERS_UPDATE_AUTO=true
    -p 6884:6884 \
    -p 6884:6884/udp \
    -p 8989:8989 \
    -v /mnt/sata2-4/Configs:/config \
    -v /mnt/sata2-4:/Downloads \
    --restart unless-stopped \
    johngong/qbittorrent:latest

/config和/downloads在weiUI中不要更改，指相对目录变量

编辑--日志内查看用户名admin和密码xxxxx(每次启动不一样)

138161, 2024-04-29:
可在此菜单更改： 'Tools' menu -> 'Options...' -> 'Web UI' tab -> 'Authentication'
webUI 白名单  输入IP地址：0.0.0.0/0
端口改为 6884
防火墙自定义规则
# qBittorrent Docker
ip6tables -A INPUT -p tcp    --dport 6884  -j ACCEPT
ip6tables -A INPUT -p udp  --dport 6884  -j ACCEPT

138161, 2024-04-29:
不过，我看还是不要用了，直接istore的迅雷，免费用户下载速度都比qb快，付费用户更快，一年也就130，就当一年的电费，开了台下载机

138161, 2024-04-29:
我比较过，刚出的剧集，qt和迅雷（都是docker版），qb几个小时都是0，迅雷几十k，出了几天的种子，迅雷至少快3倍，付费的话，快6倍以上，如果是早就出来的种子，只能迅雷下到，qb得等到人家有种子的机器上线，有的等了

138161, 2024-04-29:
而且，qb的DHT数量，在中国境内也就370左右，太少了，不值得

Haruhi-Suzumiya, 2024-05-01:
> 碰到过，没解决，而且运行几天后貌似卡死，重启后流量马上来了，后来改用docker版 qBittorrent-qBittorrentEE 4.6.4-4.6.4.10 (amd64;arm64v8;arm32v7) 集成Trackers自动更新 https://hub.docker.com/r/johngong/qbittorrent
> 
> 带搜索功能，稳定，不会卡死

多谢兄弟热情回复