istoreos/istoreos #1486: 编译镜像生成错误

YanxingLiu, 2024-04-10:
## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译M28K的istoreos固件，编译成功未报错，但是生成的镜像大小只有16M，并且无法被解压。

### (2) 路由器型号和固件版本

A：芒果派M28K

### (3) 详细日志和/或截图

A：
编译成功截图
[image]

编译日志
```
Parallel mksquashfs: Using 12 processors
Creating 4.0 filesystem on /root/projects/istoreos/build_dir/target-aarch64_cortex-a53_musl/linux-rockchip_rk35xx/root.squashfs, block size 262144.
Pseudo file "/dev" exists in source filesystem "/root/projects/istoreos/build_dir/target-aarch64_cortex-a53_musl/root-rockchip/dev".
Ignoring, exclude it (-e/-ef) to override.
[===========================================================================================================================================================================================================================================\] 1048/1048 100%

Exportable Squashfs 4.0 filesystem, xz compressed, data block size 262144
        compressed data, compressed metadata, compressed fragments,
        no xattrs, compressed ids
        duplicates are removed
Filesystem size 5171.27 Kbytes (5.05 Mbytes)
        32.72% of uncompressed filesystem size (15803.71 Kbytes)
Inode table size 9772 bytes (9.54 Kbytes)
        20.86% of uncompressed inode table size (46843 bytes)
Directory table size 13640 bytes (13.32 Kbytes)
        45.96% of uncompressed directory table size (29676 bytes)
Number of duplicate files found 161
Number of inodes 1407
Number of files 1040
Number of fragments 41
Number of symbolic links  244
Number of device nodes 1
Number of fifo nodes 0
Number of socket nodes 0
Number of directories 122
Number of ids (unique uids + gids) 1
Number of uids 1
        root (0)
Number of gids 1
        root (0)
/root/projects/istoreos/staging_dir/host/bin/make_ext4fs -L rootfs -l 1073741824 -b 4096 -m 0 -J -T 1712554030 /root/projects/istoreos/build_dir/target-aarch64_cortex-a53_musl/linux-rockchip_rk35xx/root.ext4 /root/projects/istoreos/build_dir/target-aarch64_cortex-a53_musl/root-rockchip/
Creating filesystem with parameters:
    Size: 1073741824
    Block size: 4096
    Blocks per group: 32768
    Inodes per group: 8192
    Inode size: 256
    Journal blocks: 0
    Label: rootfs
    Blocks: 262144
    Block groups: 8
    Reserved blocks: 0
    Reserved block group size: 63
Created filesystem with 1416/65536 inodes and 9346/262144 blocks
rm -f /root/projects/istoreos/build_dir/target-aarch64_cortex-a53_musl/linux-rockchip_rk35xx/mangopi_m28k-kernel.bin
[... 335 lines cut ...]
Generating index for package ./libnghttp2-14_1.44.0-1_aarch64_cortex-a53.ipk
Generating index for package ./libzstd_1.5.2-1_aarch64_cortex-a53.ipk
Generating index for package ./tar_1.34-1_aarch64_cortex-a53.ipk
Generating index for package ./xz-utils_5.4.1-1_aarch64_cortex-a53.ipk
Generating index for package ./xz_5.4.1-1_aarch64_cortex-a53.ipk
Generating index for package ./luci-app-store_0.1.14-3_all.ipk
Generating index for package ./luci-lib-taskd_1.0.18_all.ipk
Generating index for package ./luci-lib-xterm_4.18.0_all.ipk
Generating index for package ./taskd_1.0.3-1_all.ipk
Generating index for package ./luci-i18n-argon-zh-cn_git-23.088.28463-56b29d4_all.ipk
Generating index for package ./luci-theme-argon_2.2.10.10_all.ipk
Signing package index...
make[2]: Leaving directory '/root/projects/istoreos'
export MAKEFLAGS= ;make -w -r json_overview_image_info
make[2]: Entering directory '/root/projects/istoreos'
WORK_DIR=/root/projects/istoreos/build_dir/target-aarch64_cortex-a53_musl/json_info_files /root/projects/istoreos/scripts/json_overview_image_info.py /root/projects/istoreos/bin/targets/rockchip/rk35xx/profiles.json 
make[2]: Leaving directory '/root/projects/istoreos'
export MAKEFLAGS= ;make -w -r checksum
make[2]: Entering directory '/root/projects/istoreos'
make[2]: Leaving directory '/root/projects/istoreos'
make[1]: Leaving directory '/root/projects/istoreos'
```

编译结果（从图里可以看出镜像只有16MB大小）
[image]

jjm2473, 2024-04-12:
固件应该没问题，只是你没选择软件包，所以这么小。你可以使用h28k的config和feed来选择默认的软件包 https://fw.koolcenter.com/iStoreOS/h28k/

jjm2473, 2024-04-12:
注意，ext4镜像是没用的，istoreos只支持squashfs镜像启动，有些功能也依赖squashfs。

closed 2024-04-12 by jjm2473

YanxingLiu, 2024-04-13:
> 固件应该没问题，只是你没选择软件包，所以这么小。你可以使用h28k的config和feed来选择默认的软件包 https://fw.koolcenter.com/iStoreOS/h28k/

好的，我试试，但是这个编译出来的固件也没法被解压，解压的时候报文件不完整。

jjm2473, 2024-04-13:
不需要tar解压，这都不是一个tar包，只是gzip压缩了而已。而且一般情况下也不需要解压啊，除非你要dd写入sd卡。如果要解压，那么使用命令`gunzip -k abc.img.gz`

jjm2473, 2024-04-13:
你以前真的有解压过其他固件？你自己看看正常固件怎么解压的。

YanxingLiu, 2024-04-13:
> 你以前真的有解压过其他固件？你自己看看正常固件怎么解压的。

好的，不好意思打扰了，感谢您