istoreos/istoreos #1494: 光猫拨号--软路由--华硕68U（开ipv6 passthrough）问题

138161, 2024-04-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：以下两种方式能正常使用，中继模式，slaac获得ipv6，可续约
     光猫拨号--软路由--PC
     光猫拨号--华硕68U（开ipv6 passthrough）--PC
现在使用这个方式：光猫拨号--软路由--华硕68U（开ipv6 passthrough）--PC
出现情况：华硕68U关闭ipv6，一切正常，开ipv6 passthrough，PC有ipv6地址，能ping6通，就是网页无法访问

问题：是哪里需要设置？初步认为，开了ipv6，可能优先走ipv6，TCP协议的流量好像被挡住了，icmpv6协议正常，在华硕68U的路由器上，直接测试，能连ipv6，比如默认的QQ等网站，现在怀疑两个方面，一是防火墙哪个地方没开，二是需要NAT，有这方面经验的能否告知，谢谢。

### (2) 路由器型号和固件版本

A：istore 22.3

### (3) 详细日志和/或截图

A：

jjm2473, 2024-04-13:
可能只是因为默认过滤了ipv6 dns而已，在后台“网络”-“DNS/DHCP”的高级设置里，取消勾选“过滤 AAAA记录”。还不行就看看置顶的ipv6 issue，看能不能解决。

138161, 2024-04-15:
我思考了几天，猜测可能是这样的，其实，这就是有ipv6地址，缺不通的情况，问题还是出在NDP消息的传递，NDP就是告诉路由表，电脑在哪里，路由表只有知道了下面电脑的位置情况，才能让TCP流量正确到达目的地。
光猫--iStore--路由器--电脑，这其中，iStore打开了547端口，用来接收光猫546端口发出的NDP消息，但这个消息是进入iStore内部，并没有再传到下一级路由器，于是，该路由器就没办法知道这个通路该如何走，只有在iStore的openwrt内核中实现这样的一个功能，就是将收到的NDP消息再通过546端口转发到Lan侧，由下级路由器的547端口来接收，那么路由器就知道了TCP流量往哪里走了，所以，这大概是无解的。
那么，有没有解决办法？有的，就是路由器使用NAT方式，该方式下，路由器Lan侧的RA和DHCP是服务器模式，NDP是禁用的。实际用华硕68U的IPV6之Native模式，就是NAT模式，是可以解决此问题。