istoreos/istoreos #1496: pve 8 mt7922 直通 2024031514 驱动存在 网络中 没有无线设置入口

mr00ff, 2024-04-13:
尊敬的开发者 早/中/晚都好
如题 还需要提供啥
虚拟化: pve
网卡型号: mt7922
固件版本 iStoreOS 22.03.6 2024031514

```shell
dmesg|grep -i wifi

lspci
```

[image]

[image]

closed 2024-04-18 by mr00ff

SkiffBoy, 2024-05-01:
大佬，最后怎么解决的？

1137129606, 2025-01-07:
**相同问题mt7921k x86实体机无法识别**

[image]
[image]
[image]

tonbit, 2025-06-04:
[image]