istoreos/istoreos #1498: iStoreOS的docker中只剩配置

bboyjim, 2024-04-15:
## 1.关于你要提交的问题
昨天更新了固件22.03.6 2024031514，iStoreOS的docker中只剩配置了，而且首页的DOCKER配置也出现这个提示。尝试重启docker，重新安装都不行

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志和/或截图
[image]

[image]

bboyjim, 2024-04-15:
[image]
我解决了，因为docker配置被还原了。所以docker目录要自己手动改一下

jjm2473, 2024-04-15:
怎么会无缘无故还原？你升级时没保留配置？

closed 2024-04-22 by jjm2473

smhjw, 2024-06-28:
> [image]
> 我解决了，因为docker配置被还原了。所以docker目录要自己手动改一下

改了以后我怎么保存不了呢

jjm2473, 2024-07-17:
检查下是不是根目录满了导致无法写入

ljiner, 2025-02-06:
我这个改完还是不显示