istoreos/istoreos #1505: Jellyfin 色调映射问题

decayjoy, 2024-04-17:
## 1.关于你要提交的问题
iStore内通过docker安装的Jellyfin无法启用基于OpenCL的色调映射

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：无法启用基于intel opencl的色调映射，不开启色调映射时可以正常硬解HDR10，杜比视界颜色异常，VPP色调映射可以正常运行，但同样不支持杜比视界，但是开启opencl的色调映射，任何HDR直接无法播放，按Jellyfin的说法基于opencl的色调映射方案应该是可以支持杜比视界的，网上搜了很久，找到唯一一条Nyanmisaka大神的回复，似乎是Docker宿主机驱动的问题，不知道是否有解决方案。

### (2) 路由器型号和固件版本

A：畅网微控8505软路由，PVE 8.1.10虚拟化，SRIOV核显直通，已开启i915-oot驱动，固件版本：iStoreOS 22.03.6 2024031514

### (3) 详细日志和/或截图

A：JellyFin日志，显示是FFMPEG问题，但是没有任何有意义的错误信息。
[image]
B：FFMPEG日志
[Log.txt](https://github.com/istoreos/istoreos/files/15008190/Log.txt)
C：Nyanmisaka大神的回复，另一位用户2年前在 Fedora 上安装docker版的Jellyfin，遇到类似问题，似乎说Intel的opencl只认Ubuntu，不知道移植驱动是否是这个问题。
[image]

jjm2473, 2024-04-22:
不知道，最好别自己选驱动，现在i915-oot是自动启用的，不要手动选择。如果你能问到具体的内核配置，才有参考价值，不然谁知道ubuntu哪个配置跟这个问题有关。

decayjoy, 2024-04-23:
> 不知道，最好别自己选驱动，现在i915-oot是自动启用的，不要手动选择。如果你能问到具体的内核配置，才有参考价值，不然谁知道ubuntu哪个配置跟这个问题有关。

感谢答复，我也没有乱装驱动，就是istore自带的i915-oot，默认没有启用，找不到/dev/dri才去手动开启的，开了后能正常找到显卡了，后面的Ubuntu的说法，只是实在开不了色调映射，唯一找到这么一个说法，并不具有特别的参考意义。不知道有没有人遇到类似的问题后来解决了的。

jjm2473, 2024-04-23:
那可能i915-oot对8505的核显支持不好吧，毕竟不是官方驱动。你执行下`lspci -nnk`看看。